Kinovarobotics/kinova-ros #35: Which joints are continuous?

davetcoleman, 2015-03-03:
From the README.md, it seems like only the last 3 wrist joints are continuous. But in the URDF [here](https://github.com/Kinovarobotics/jaco-ros/blob/master/jaco_model/urdf/jaco_common.xacro#L65) it also makes joint 1 continuous:

```
<xacro:jaco_joint joint_name="${joint_1}" type="continuous" parent="${link_base}" child="${link_1}" joint_axis_xyz="${joint_1_axis_xyz}" joint_origin_xyz="${joint_1_origin_xyz}" joint_origin_rpy="${joint_1_origin_rpy}" joint_lower_limit="${joint_1_lower_limit}" joint_upper_limit="${joint_1_upper_limit}"/>
```

Is this a bug?

I'm using Jaco 2, is there a difference?

Also, are there reasons to set fake limits? I see the comment in the README.md:

> You may want to limit to +/-6.28 rad in software to prevent the joints from rotating excessively.

francoisferland, 2015-03-09:
You are right, this is an issue in the "Master" version of the URDF. It has already been fixed in the newer branches that should be merged soon.

davetcoleman, 2015-03-09:
Which branches are those?

francoisferland, 2015-03-09:
Both fferland-newapi and fferland-experimental have the changes. I forgot to mention that Joint 1 can be considered to be continuous, only Joint 2 and 3 are truly revolute with proper limits.

wahid97, 2017-04-26:
This may not be relevant to this post, but I would greatly appreciate your support since I am stuck with my Kinova robot. It simply lost the USB connection and now the joystick green light flashes continuously. I couldn't able to update the USB driver in my computer because it doesn't appear in the device manager! Is there any option to do the factory reset? It was working fine yesterday!

Thanks in advance.

Wahid

jeff-o, 2017-04-26:
The flashing green lights mean the arm needs to be homed. Have you tried
returning it to home using the joystick?

wahid97, 2017-04-26:
> 297427947>,

Thanks Jeff for your reply. No, the joystick doesn't work and to move the
robot to its home position we need to have the solid green light. Also, if
the robot is not in the right position when we start, it suppose to show
the solid red light, right? I believe it fails to synchronize with the
internal board and I need to do the factory resetting. I couldn't able to
use the SDK gui file to do this!

Regards,
Wahid

jeff-o, 2017-04-26:
> Phone: 519-513-2416 <(519)%20513-2416>
> 297432274>,

That's truly unfortunate! I'm not aware of any way to do a factory reset on
the arm, so you may need to contact Kinova support for help on this one.
Good luck!

wahid97, 2017-04-26:
> issues/35#issuecomment-
> 297436028>,

Thanks so much Jeff, I will.

closed 2017-09-11 by martine1406

Alchemist77, 2018-01-24:
To wahid97,

Did you solve the problem? I faced the same situation.

martine1406, 2018-01-24:
Hi

If the green light does not stop blinking, your arm's program may have been corrupted in some way. You will have to reprogram the arm. Can you still see the arm connected through USB? If so, you can reprogram the arm using the Development Center. If not, you will have to reprogram the arm in bootloader mode. Please contact support@kinova.ca with your arm's serial number. They will be able to provide you the right firmware program file for your robot and assist you if you have any question on how to reprogram the arm

All the best,

Martine

GianlucaCerilli, 2018-08-07:
Hi,

I have the same problem too. @wahid97 and @Alchemist77 have you faced the problem?
When I switch on the arm the green light flashes continuously and it doesn´t stop. So the robot does not move and if I connected it to the pc, it is not visible.
How can I do? I have also contacted the support but they are not answering, do you know if there is also another solution @martine1406 ?

Thank you very much,
Gianluca

wahid97, 2018-08-07:
> HI,
>
> There are several options to fix it. I do not remember the solution, but
> the cause of the problem is communication. First, you have to wait for
> Kenova support or call them directly. After you contact Kinova you can do,
> 1. update software (newest version)
> 2. open the arm to check the electronic lines between each motor and
> controller are connected well (be careful!!)
> 3. Otherwise, you have to send the arm to Kinova in Canada. (It is last
> option..).
>
> It is my supposition. The Kinova arm is weak and sensitive so follow the
> Kinova support advice.
>
> Thanks.
>
> Jaeseok

Hi Giancula,

Yes
I do remember
.
This is most likely due to the corrupted firmware which you may need to
flash
in bootloader mode
(use a
wire to connect two pins).

please contact
Support@KinovaRobotics.com <Support@KinovaRobotics.com>

T
hey will provide the detail instructions. There might be differences in
serial number/firmware version/etc than mine.

I hope you get their response very soon.

Thanks.

Kind regards,
Wahid

nsaif-kinova, 2018-08-07:
Hello Gianluca,
You should reprogram your arm to solve this problem, I see your email is answered now, I apologize for the delay caused by the time difference.
If you have any other issue please don't hesitate to contact us.
Sincerely,