Kinovarobotics/kinova-ros #36: Code review of fferland-newapi

davetcoleman, 2015-03-09:
I know this branch isn't ready, but I need my Jaco 2 working now so I studied your changes. If you don't mind:

There is a misspelling of "RefresDevicesList" in ``include/kinova/Kinova.API.UsbCommandLayerUbuntu.h``

In ``kinova_driver/src/jaco_arm.cpp`` there appears to be a regression where it says
```
joint_state.position[1] = (jaco_angles.joint2 - 270) * (PI / 180);
```
shouldn't the 270 be the variable ``j6o`` like it was previously?

Also, how stable is this and the experimental branch? How many bumps should I expect if I start using it now?

francoisferland, 2015-03-09:
Yes it should. Actually, this was developed before the j6o variable in the main branch, and only tested on the Mico. Thanks for catching it.

davetcoleman, 2015-03-09:
https://github.com/Kinovarobotics/jaco-ros/pull/38

Should I renamed RefresDevicesList also?

francoisferland, 2015-03-09:
Sure, thanks in advance!

closed 2017-09-11 by martine1406