Kinovarobotics/kinova-ros #46: Add simplified collision and inertial elements to the URDF model.

peci1, 2015-05-05:
Kinova should provide a simplified collision model for e.g. collision checking or robot body removal from point clouds. Please, try to provide the model in terms of primitive shapes, not meshes (which are slow in computations). Or at least provide simplified convex hulls of the meshes.

As a starting point, you can cherry-pick my work for Jaco arm: https://github.com/Kinovarobotics/jaco-ros/commit/14c006aab60e19b2a07ad6c8f43cc57220643269 (I won't create a PR because we have lots of different stuff changed in our working branch).

francoisferland, 2015-05-05:
Thanks again for the pull request. We are currently slightly changing the structure of our URDF macros for the next release, but I will integrate your work and adapt it for our other models.

francoisferland, 2015-05-05:
Yes, it is part of our plans. Actually, it is already available with in the fferland-experimental branch, along with a set of launch files and controllers for Gazebo:

https://github.com/Kinovarobotics/jaco-ros/tree/fferland-experimental

However, I would not recommend using that branch on a real robot.

peci1, 2015-05-05:
Great, thanks!

closed 2017-09-11 by martine1406