Kinovarobotics/kinova-ros #54: [Question] What is the correct transformation matrices of MICO?

mu-777, 2015-09-18:
Hello,

I need the transformation matrices of MICO to control it.
Now I'm using the matrices referring to the comments on ``jaco_arm_kinematics.cpp``.
However, tfs from ``jaco_tf_updater.cpp`` (which uses ``jaco_arm_kinematics.cpp``, you may know) is different from tfs from MICO's urdf.

Which is the same as the real MICO configulations?
How can I know the transformation matrices of MICO?

thanks.

mklingen, 2016-02-26:
I'm just bumping this because I have also noticed that the kinematics defined in the URDF are completely different than those defined in `jaco_arm_kinematics.cpp`. The hardcoded kinematics in `jaco_arm_kinematics.cpp` are much closer to correct than the kinematics in the URDF. Sometimes, the tool frame can be off by as much as 10 cm(!). This is very bad news for anyone using the mico arm driver with ROS. I'm going to take a close look at this in the coming weeks and try to come up with a solution.

mklingen, 2016-02-26:
Pinging @francoisferland or anyone else who worked on the URDF.

francoisferland, 2016-02-26:
Hi,

It has been a while since I worked on the URDF and I no longer have access to the physical robots, but it sounds like a mix-up between the latest URDF version and the old version of the driver. I think the latest, correct version of the driver is still in the "fferland_newapi" branch, not on the master one. If you look at jaco_arm.cpp in that branch, the J5 offset error you mention in #58 should be fixed.

Hope this helps!

mklingen, 2016-02-26:
@francoisferland thank you very much. I have pulled from `fferland_newapi`. While it does fix the PI/2 offset of J6, it does not solve all the problems with the URDF. Here are a couple of screenshots of the tool frame reported by the driver as compared to the URDF. The tool frame is supposed to be aligned with the axis of J6, and located between the fingers. 

[image]
[image]

mklingen, 2016-02-26:
It looks like most of the problem is with J2 and J3, as you can see in this image. 

[image]

[image]

J1 is supposed to make a nearly right angle with the base here, and J2 is a nearly right angle with link 1. The joint angles reported by the firmware also make sense for this configuration (~270 degrees and ~90 degrees).

francoisferland, 2016-02-26:
You are right, there seem to be an offset on J2-3. I don't remember seeing this before.

My first guest was that there is also an API offset setting shared between the driver and the hardware controller. That offset is meant to adjust the zero point for cartesian commands. The tool topic is coming straight from the low level API, and (I think) includes this offset. However, it might not be attached to the correct URDF frame, or that frame does not have the proper offset. For that, you could always measure the inverse transform and re-publish it as a correction TF frame, but it wouldn't solve what you see in the second picture.

Another way to test it would be a completely straight pose (pointing to the ceiling). You could then try adjusting that offset in the angle conversion code. Otherwise I would suggest reaching Kinova support directly. As I'm no longer affiliated with them I can't really help beyond that.

mklingen, 2016-02-26:
Thank you very much for helping me understand this issue. I have noticed that there are parameters for joint offsets in the `jaco_arm_kinematics` module. Do you know if these parameters are supposed to be calibrated for each robot, or if they are fixed?

Looking at the robot configuration when its straight up, I can definitely see that the error is just with J2 it seems.

[image]

francoisferland, 2016-02-26:
As far as I know, they should be fixed from the factory.

mklingen, 2016-02-26:
I fixed it!! It all comes down to this line:

https://github.com/Kinovarobotics/kinova-ros/blob/551a13503912ad3818e88e16cd7a69a21c7cdde3/jaco_driver/src/jaco_arm.cpp#L334

Why is the offset randomly set to 260 degrees instead of 270 for mico arms!? Crazy!

francoisferland, 2016-02-26:
I vaguely remember this being an issue at one point, but I'm surprised it's still the case... Maybe something else changed since last time I looked at it. Anyway, I'm happy you fixed your issue!

François

closed 2017-09-11 by martine1406