Kinovarobotics/kinova-ros #61: roslaunch kinova_bringup fails

AliShafiekhani, 2016-07-29:
I tried to launch robot with following command :
roslaunch kinova_bringup kinova_robot.launch kinova_robotType:=j2s7s300 use_urdf:=false
but it returns error :
[ INFO] [1469825146.149697563]: kinova_robotType is j2s7s300.
Please specify the kinematic of robots other than jaco and mico!
[j2s7s300_driver-1] process has died [pid 2725, exit code -6, cmd /home/administrator/vinobot_ws/devel/lib/kinova_driver/kinova_arm_driver j2s7s300 __name:=j2s7s300_driver __log:=/home/administrator/.ros/log/694e5d8c-55cb-11e6-bc33-003018c2a7c0/j2s7s300_driver-1.log].
log file: /home/administrator/.ros/log/694e5d8c-55cb-11e6-bc33-003018c2a7c0/j2s7s300_driver-1*.log

LongfeiProjects, 2016-08-09:
Hello Ali, as we discussed via email. Only the DOF of the arm is counted, but the DOF of fingers is not. Therefore, please replace "7" with the DOF of your arm. 
If everything is fine, I would close this issue. If you still get problems, please let me know.

AliShafiekhani, 2016-08-09:
Hey,

I tried 6 DOF as well as the default and some other possibilities, non of them worked. So I think there should be another reason for that!!!

Regards,

Ali Shafiekhani
President, MU Robotics Organization

Graduate Research Assistant

ViGIR - Vision-Guided and Intelligent Robotics Lab<http://vigir.missouri.edu/>
18, Engineering Building West
Dept. of Electrical and Computer Engineering
University of Missouri-Columbia

LongfeiProjects, 2016-08-10:
Hi Ali, it is possible that something else is going wrong. Please try the following and give me more feedback if the problem is still existing.  The first priority would be ensure that you can install ROS stack successfully.

1. "#include <KinovaTypes.h>" in kinova_comm.cpp is unnecessary, since "KinovaTypes.h" is already included in kinova_comm.h with the right path "kinova/KinovaTypes.h". Therefore, please comment or delete this line in kinova_comm.cpp.
If you made another copy of "KinovaTypes.h" under "include" folder, you can delete it as well. This might be the cause of "redefinition"

2. If you edited other places in kinova_comm.cpp, other than the include on the top, please let me know as well.

3. j2s7s300 is not the right regarding to your robot. You may provide the error message when using "j2n6a300" for example. Please try this after you going through the two steps above.

AliShafiekhani, 2016-08-12:
Hi,

That #include line was the only thing I changed.
I commented the #include KinovaTypes.h as you said and it compiled fine but still same error!

below is the error message using j2n6a300 like you said:

process[j2n6a300_state_publisher-2]: started with pid [5953]
[j2n6a300_driver-1] process has died [pid 5941, exit code -6, cmd /home/administrator/vinobot_ws/devel/lib/kinova_driver/kinova_arm_driver j2n6a300 __name:=j2n6a300_driver __log:=/home/administrator/.ros/log/dcc43f06-5f40-11e6-a5a0-003018c2a7c0/j2n6a300_driver-1.log].
log file: /home/administrator/.ros/log/dcc43f06-5f40-11e6-a5a0-003018c2a7c0/j2n6a300_driver-1*.log

Regards,

Ali Shafiekhani
President, MU Robotics Organization

Graduate Research Assistant

ViGIR - Vision-Guided and Intelligent Robotics Lab<http://vigir.missouri.edu/>
18, Engineering Building West
Dept. of Electrical and Computer Engineering
University of Missouri-Columbia

  1.  "#include " in kinova_comm.cpp is unnecessary, since "KinovaTypes.h" is already included in kinova_comm.h with the right path "kinova/KinovaTypes.h". Therefore, please comment or delete this line in kinova_comm.cpp.

LongfeiProjects, 2016-08-16:
Hello Ali, 

This is indeed a strange error that you are not able to launch kinova driver node. Would you tell me more information:

1. as mentioned in my previous comment:
In fact, "#include " refers to the location "/usr/include/KinovaTypes.h". Please check if you have files under this folder. Please install Kinova SDK if you haven't done it yet. You can find them here "https://drive.google.com/a/kinova.ca/file/d/0B5d8FVDq3A-XM3NoV1lEWktyd0k/view". 

2. Can you drive the jaco v2 arm with Kinova SDK ? 
After setup and run Kinova SDK, you should be able to control the robot via "Virtual Joystick". This function does not depend on ROS. This could help to clear certain potential causes.

3. Please try the test below, and let me know the result.
In terminal 1: roslaunch kinova_description display_kinova_robot.launch gui:=true 
In terminal 2: rosrun rviz rviz
This test does not need kinova driver and does not need to link with real robot. It provides you an interface to control robot model in Rviz. 

I am looking forward to your feedback.

AliShafiekhani, 2016-08-18:
Hi,

Yeah it is really weird. I have the jaco-ros package (old version) working all right. Here are tested you asked:

1, 2- There was no KinovaTypes.h in /usr/include/ directory so I installed the Kinova SDK and try to run the DevelopmentCenter and use Virtual Joystick but I got core dumped! but I can move the robot to home using examples.

3- Trying the roslaunch I got following error:

started roslaunch server http://cpr-umoc-01:37702/

SUMMARY
========

PARAMETERS
 * /robot_description
 * /rosdistro
 * /rosversion
 * /use_gui

NODES
  /
    joint_state_publisher (joint_state_publisher/joint_state_publisher)
    robot_state_publisher (robot_state_publisher/state_publisher)

ROS_MASTER_URI=http://localhost:11311

core service [/rosout] found

ERROR: cannot launch node of type [joint_state_publisher/joint_state_publisher]: joint_state_publisher
ROS path [0]=/opt/ros/hydro/share/ros
ROS path [1]=/home/administrator/vinobot_ws/src
ROS path [2]=/opt/ros/hydro/share
ROS path [3]=/opt/ros/hydro/stacks
process[robot_state_publisher-2]: started with pid [6893]

Thanks for helping.

Regards,

Ali Shafiekhani
President, MU Robotics Organization

Graduate Research Assistant

ViGIR - Vision-Guided and Intelligent Robotics Lab<http://vigir.missouri.edu/>
18, Engineering Building West
Dept. of Electrical and Computer Engineering
University of Missouri-Columbia

LongfeiProjects, 2016-08-18:
Thanks Ali, the information is very helpful. Though right now I am not sure about the solution, but we have something worth to try.

 1  _I have the jaco-ros package (old version) working all right._ 
Please check with support@kinova.ca for the proper firmware update, if you have not done so. The new package come along with new APIs which should be available in the robot firmware as well.

 2 _There was no KinovaTypes.h in /usr/include/ directory so I installed the Kinova SDK and try to run the DevelopmentCenter and use Virtual Joystick but I got core dumped! but I can move the robot to home using examples._
The Kinova SDK test should be passed before moving on diagnose ROS problem. Kinova SDK is using Kinova API directly, and it is essential to prove that robot is in good communication condition. My highest suspicion is still the firmware. If you still have SDK problem after updating firmware, please check with support@kinova.ca, they are very experienced in this.

 3 _Trying the roslaunch I got following error:
.. ERROR: cannot launch node of type [joint_state_publisher/joint_state_publisher]:_ joint_state_publisher
joint_state_publisher is a ROS package which should be automatically installed while you install ROS. I provide some output regarding to this error, and please check on your side and compare the output.
$ **roscd joint_state_publisher/** 
opt/ros/indigo/share/joint_state_publisher $ **echo $ROS_PACKAGE_PATH**
/home/lzhao/catkin_ws/src:/opt/ros/indigo/share:/opt/ros/indigo/stacks

closed 2017-09-11 by martine1406