Kinovarobotics/kinova-ros #62: Cartesian Velocity and Position Control not Working

PascalEgli, 2016-08-04:
We are using the mico arm with 6 joints and two fingers and firmware version 5.2.1.22 using the latest kinova-ros driver. The joint position and velocity control is working fine, however the cartesian position and control is not working. Also, mico_home service is not working. We were using the command line tools as mentioned in the readme.

We tested the kinova SDK (in Windows) and we were able to control the cartesian velocity using the keyboard. However this only works once the robot is sent to home position using -H on the keyboard.

Do you have any ideas on how we can debug this issue?

LongfeiProjects, 2016-08-09:
Hello Pascal, 

Right after the power up, did you set the robot to its Home position before any other command? This is a normal procedure also when you are using joystick to control robot.
If you already did that but still get problem with Cartesian control, could you please provide me error messages if there is any?

-- 
Longfei

Jorge-C, 2016-08-12:
I'm not sure if this is related, but we're trying to use the cartesian position control, as implemented in `pose_action_client.py`, for a Jaco arm (robot type `j2n6a300`). After we send a bunch of commands (~3), the robot stops executing the commands. The action does not time out, so it somehow thinks it's successful. Finger control keeps working, so it doesn't look like one of the known issues [here](http://wiki.ros.org/jaco_ros). We home the robot after powering it up and launching kinova_bringup kinova_robot.launch, before sending any other commands.

Any thoughts? Thanks!

LongfeiProjects, 2016-08-16:
Hello Jorge-C, I checked the problem, and uploaded a branch for hotfix. Please try with hotfix_position_control branch, and let me know if this solve your problem.

Jorge-C, 2016-08-17:
Thanks @LongfeiProjects for the quick response. We tested that branch and we still have the same problem. BTW, that branch is 12 commits behind master, is that OK? Thanks!

LongfeiProjects, 2016-08-18:
Hi Jorge-C, I just modified that branch. Now the hotfix branch stems from master. I modified threshold and added some debug tags. Those debug message may help you to probe the problem as well. If you still have this problem, please let me know.

Jorge-C, 2016-08-30:
Thanks @LongfeiProjects!

We still see the problem. Here's some typical information printed using the latest changes:
```
***-----------------------***
void kinova::KinovaPoseActionServer::actionCallback(const ArmPoseGoalConstPtr&):  target X -0.0125032; Y 0.396702; Z -0.0161243; ThetaX 3.14159; ThetaY 0; ThetaZ 0
***-----------------------***

[DEBUG] [1472578602.677539219]: void kinova::KinovaPoseActionServer::actionCallback(const ArmPoseGoalConstPtr&): current_pose X -0.486066; Y -0.053926; Z 0.0227115; ThetaX -3.0608; ThetaY -0.0125398; ThetaZ -0.151381
[DEBUG] [1472578602.786295796]: void kinova::KinovaPoseActionServer::actionCallback(const ArmPoseGoalConstPtr&): current_pose X -0.486066; Y -0.053926; Z 0.0227115; ThetaX -3.0608; ThetaY -0.0125398; ThetaZ -0.151381
[DEBUG] [1472578602.896357090]: void kinova::KinovaPoseActionServer::actionCallback(const ArmPoseGoalConstPtr&): current_pose X -0.486066; Y -0.053926; Z 0.0227115; ThetaX -3.0608; ThetaY -0.0125398; ThetaZ -0.151381
[DEBUG] [1472578603.005519269]: void kinova::KinovaPoseActionServer::actionCallback(const ArmPoseGoalConstPtr&): current_pose X -0.486066; Y -0.053926; Z 0.0227115; ThetaX -3.0608; ThetaY -0.0125398; ThetaZ -0.151381
[DEBUG] [1472578603.119348973]: void kinova::KinovaPoseActionServer::actionCallback(const ArmPoseGoalConstPtr&): current_pose X -0.486066; Y -0.053926; Z 0.0227115; ThetaX -3.0608; ThetaY -0.0125398; ThetaZ -0.151381
[DEBUG] [1472578603.228452977]: void kinova::KinovaPoseActionServer::actionCallback(const ArmPoseGoalConstPtr&): current_pose X -0.486066; Y -0.053926; Z 0.0227115; ThetaX -3.0608; ThetaY -0.0125398; ThetaZ -0.151381
[DEBUG] [1472578603.337303771]: void kinova::KinovaPoseActionServer::actionCallback(const ArmPoseGoalConstPtr&): current_pose X -0.486066; Y -0.053926; Z 0.0227115; ThetaX -3.0608; ThetaY -0.0125398; ThetaZ -0.151381
[DEBUG] [1472578603.446320869]: void kinova::KinovaPoseActionServer::actionCallback(const ArmPoseGoalConstPtr&): current_pose X -0.486066; Y -0.053926; Z 0.0227115; ThetaX -3.0608; ThetaY -0.0125398; ThetaZ -0.151381
[DEBUG] [1472578603.555350197]: void kinova::KinovaPoseActionServer::actionCallback(const ArmPoseGoalConstPtr&): current_pose X -0.486066; Y -0.053926; Z 0.0227115; ThetaX -3.0608; ThetaY -0.0125398; ThetaZ -0.151381
[DEBUG] [1472578603.663493561]: void kinova::KinovaPoseActionServer::actionCallback(const ArmPoseGoalConstPtr&): current_pose X -0.486066; Y -0.053926; Z 0.0227115; ThetaX -3.0608; ThetaY -0.0125398; ThetaZ -0.151381
[DEBUG] [1472578603.663587978]: void kinova::KinovaPoseActionServer::actionCallback(const ArmPoseGoalConstPtr&): stall_interval_seconds_
[DEBUG] [1472578603.722310667]: Setting the current goal as canceled
[DEBUG] [1472578603.722634328]: Setting status to canceled on goal, id: /pp_node-25-1472578602.659, stamp: 1472578602.66
[DEBUG] [1472578603.722772167]: Publishing result for goal with id: /pp_node-25-1472578602.659 and stamp: 1472578602.66
[DEBUG] [1472578604.230789956]: The action server has received a new goal request
[DEBUG] [1472578604.230904649]: A new goal has been recieved by the single goal action server
[DEBUG] [1472578604.231005049]: Accepting a new goal
[DEBUG] [1472578604.231061044]: Accepting goal, id: /pp_node-26-1472578604.230, stamp: 1472578604.23
[DEBUG] [1472578604.249312014]: Publishing feedback for goal, id: /pp_node-26-1472578604.230, stamp: 1472578604.23
[DEBUG] [1472578604.249411204]: Publishing feedback for goal with id: /pp_node-26-1472578604.230 and stamp: 1472578604.23
[DEBUG] [1472578604.352971956]: Publishing feedback for goal, id: /pp_node-26-1472578604.230, stamp: 1472578604.23
[DEBUG] [1472578604.353131660]: Publishing feedback for goal with id: /pp_node-26-1472578604.230 and stamp: 1472578604.23
[DEBUG] [1472578604.455090427]: Publishing feedback for goal, id: /pp_node-26-1472578604.230, stamp: 1472578604.23
[DEBUG] [1472578604.455163334]: Publishing feedback for goal with id: /pp_node-26-1472578604.230 and stamp: 1472578604.23
[DEBUG] [1472578604.557093630]: Publishing feedback for goal, id: /pp_node-26-1472578604.230, stamp: 1472578604.23
[DEBUG] [1472578604.557199793]: Publishing feedback for goal with id: /pp_node-26-1472578604.230 and stamp: 1472578604.23
[DEBUG] [1472578604.660116382]: Publishing feedback for goal, id: /pp_node-26-1472578604.230, stamp: 1472578604.23
[DEBUG] [1472578604.660235953]: Publishing feedback for goal with id: /pp_node-26-1472578604.230 and stamp: 1472578604.23
[DEBUG] [1472578604.763187354]: Publishing feedback for goal, id: /pp_node-26-1472578604.230, stamp: 1472578604.23
[DEBUG] [1472578604.763405203]: Publishing feedback for goal with id: /pp_node-26-1472578604.230 and stamp: 1472578604.23
[DEBUG] [1472578604.866193098]: Publishing feedback for goal, id: /pp_node-26-1472578604.230, stamp: 1472578604.23
[DEBUG] [1472578604.866379039]: Publishing feedback for goal with id: /pp_node-26-1472578604.230 and stamp: 1472578604.23
[DEBUG] [1472578604.969069999]: Publishing feedback for goal, id: /pp_node-26-1472578604.230, stamp: 1472578604.23
[DEBUG] [1472578604.969168267]: Publishing feedback for goal with id: /pp_node-26-1472578604.230 and stamp: 1472578604.23
[DEBUG] [1472578605.071947529]: Publishing feedback for goal, id: /pp_node-26-1472578604.230, stamp: 1472578604.23
[DEBUG] [1472578605.072053552]: Publishing feedback for goal with id: /pp_node-26-1472578604.230 and stamp: 1472578604.23
[... cut ...]
[DEBUG] [1472578606.381938209]: Setting status to canceled on goal, id: /pp_node-26-1472578604.230, stamp: 1472578604.23
[DEBUG] [1472578606.381987982]: Publishing result for goal with id: /pp_node-26-1472578604.230 and stamp: 1472578604.23
[DEBUG] [1472578606.909043523]: The action server has received a new goal request
[DEBUG] [1472578606.909256306]: A new goal has been recieved by the single goal action server
[DEBUG] [1472578606.909419935]: Accepting a new goal
[DEBUG] [1472578606.909479428]: Accepting goal, id: /pp_node-27-1472578606.907, stamp: 1472578606.91
[DEBUG] [1472578606.911936641]: 
```

LongfeiProjects, 2016-09-02:
Hi Jorge-C, did you sent a new command before the previous one is processed? If that is the case, actionlib do have a "preempt" feature, which allow new command over-write the previous one. If that's not the case, could you tell me how is your action goal send?

siddarth31, 2016-09-02:
Hi Longfei Zhao,

I am replying on behalf of Pascal. So, we tried to use the service call to move the arm to the home position first, as you suggested. Here are the commands:
roslaunch kinova_bringup kinova_robot.launch kinova_robotType:=m1n6s200 use_urdf:=true
rosservice call /m1n6s200_driver/in/home_arm "{}" 

But the arm does not move. Here is the output we get in terminal:
homearm_result: KINOVA ARM HAS BEEN RETURNED HOME
The arm is ready to use.
[ INFO] [1472851157.450370101]: Homing the arm
[ WARN] [1472851175.451989620]: Homing arm timer out! If the arm is not in home position yet, please re-run home arm.

we looked into the code. In the **void KinovaComm::homeArm(void)** function the command **kinova_api_.sendJoystickCommand(mycommand)** use the kinova joystick message to send the arm to home position.  We do not have a Kinova joystick with our MICO. In order to use the api code is it necessary to have the joystick connected to the robot?  If not, how do we sent the arm to home position using the API?

Also, this is tested on the hotfix_position_control branch as well but the home service does not work. 

Thanks!

Siddarth

LongfeiProjects, 2016-09-06:
Hello Siddarth,

Though I cannot reproduce this problem. There are some things we can check together. 

1. I can confirm that we do not need to connect the joystick to robot to use the api functions. I disconnected the joystick and reboot the robot. The same command drives the robot back to Home.

2. The joystick command is actually "virtual joystick", which you can find in Kinova SDK. Please install it from Kinova website (http://www.kinovarobotics.com/service-robotics/products/software/). After run the "Development Center", you can observe robot state via "Monitoring" and control robot via "Virtual Joystick". Please check if robot go Home when you press "Home" button located at the right-bottom of VirtualJoystick/Cartesian tab.

3. A previous version of Home is defined as following. Please replace the current one with following to see if it works.

void KinovaComm::homeArm(void)
{
    boost::recursive_mutex::scoped_lock lock(api_mutex_);

    if (isStopped())
    {
        ROS_INFO("Arm is stopped, cannot home");
        return;
    }
    else if (isHomed())
    {
        ROS_INFO("Arm is already in \"home\" position");
        return;
    }

    stopAPI();
    ros::Duration(1.0).sleep();
    startAPI();

    ROS_INFO("Homing the arm");
    int result = kinova_api_.moveHome();
    if (result != NO_ERROR_KINOVA)
    {
        throw KinovaCommException("Move home failed", result);
    }
}

======================

Though I don't think this could be a cause, I list the environment of my set-up. I was testing with a Jaco v2, on Ubuntu 14.04, ROS indigo.

siddarth31, 2016-09-07:
Hi Longfei Zhao,

The number 3 above fixed the issue. 

1) The problem was with the virtual joystick as the function void KinovaComm::homeArm(void) in the current api populates a joystick message and then use it to home the arm. 
The alternate function definition calls the moveHome() directly and it works.  

2) The Cartesian velocities publisher works once the arm is sent to home. 

Thanks a lot! 

Siddarth

LongfeiProjects, 2016-09-07:
I am very happy that everything is working now. It is my pleasure to help. 
So I would close this issue for now.

closed 2016-09-07 by LongfeiProjects