Kinovarobotics/kinova-ros #65: Make Error

AnandRamakrishnan09, 2016-08-26:
During catkin_make an error in kinova_comm.cpp occurs due to the header file: 
**#include <KinovaTypes.h>**
I guess Changing it into:
**#include "KinovaTypes.h"** solves the problem for me.

mlauret, 2016-08-29:
Hello,

I'm not from Kinova but I think I can help :)

You probably write about the line `#include <KinovaTypes.h>` , which tell the compiler to look in the default include path (`/usr/include` and `<catkin_ws>/kinova_driver/include` according to the CMakelists.txt)

A clean method would be to install the [kinova SDK](http://www.kinovarobotics.com/service-robotics/support/), which install the file `KinovaTypes.h` in `/usr/include/KinovaTypes.h` and that would solve this error.

A cleaner method would be to change `#include <KinovaTypes.h>` to `#include <kinova/KinovaTypes.h>` or `#include "kinova/KinovaTypes.h"` since this is where the file is located in the ROS Stack (`<catkin_ws>/kinova_driver/include/kinova/KinovaTypes.h`) and how the other include are added (look at kinova_api.h : `#include "kinova/Kinova.API.UsbCommandLayerUbuntu.h"` )

That pretty weird, but replacing `#include <KinovaTypes.h>` to `#include "KinovaTypes.h"`  didn't solve the problem for me.

LongfeiProjects, 2016-08-29:
Thanks @mlauret , that is a very clear explanation. 

#include "KinovaTypes.h" is indeed looking for a path which exists when kinova SDK is installed. This should not be necessary, but highly recommended due to SDK's power on controlling/observing robot outside of ROS.

You can delete this line, because  #include <kinova/KinovaTypes.h> is already there in kinova_comm.h

I would correct this soon.

AnandRamakrishnan09, 2016-08-29:
@mlauret, @LongfeiProjects Thanks for the replies makes it clear. @mlauret I had to copy the `KinovaTypes.h` header file into the `/kinova_driver/src` to make it work :), forgot to mention that I guess.

closed 2016-09-06 by LongfeiProjects