Kinovarobotics/kinova-ros #67: Jaco fails to hold joint angles when fingers are controlled

kmpatel9, 2016-09-06:
Hello,
I am using Jaco arm for pick and place task using 3 finger gripper.
My code is in ROS. I set the noint angles first using jaco_angle_action server, then try to control finger positions using jaco_fingers_action server.
These two action servers are sent the goals in same python node. Finger positions are changed by the client periodically in a while loop.
I observe that, both of the tasks, that are, moving the fingers and holding up the arm position fail to happan simultaneously.
The arm is seen to be moving under the influence of gravity when fingers are being controlled.
The arm keeps going down in jerks, finally ends up on the floor.

Any pointers on how to work it around are most appreciated.
Also a quick response would be highly useful as publishing a paper with approaching deadline is blocked by this issue.

Thanking in advance,
Kunal

LongfeiProjects, 2016-09-06:
Hello,
I just replied you on the forum. However, you are very welcome to bring the topic here. Please keep me updated.

--
Longfei

kmpatel9, 2016-09-06:
Thank you Longfei,
I also sent the reply to your message on forum. I can add it here as well whichever place you prefer.

LongfeiProjects, 2016-09-06:
Hello Kunal,

The Kinova forum is a place for various problems and here is dedicated for ROS issues. Since I am mostly here, let's continue the our topic in this place. 

Error code 1015 means "Cannot find the requested device". This may due to your "resetting firmware". There is nothing to worry about that. You can get contact with support@kinova.ca, they are very experienced to handle the firmware problems.

Regarding to the falling problem, before going deeper debug, could you change one function name in the "kinova_comm.cpp":
In KinovaComm::setFingerPositions(), please try to change "result = kinova_api_.getAngularPosition(joint_angles);"
to  "result = kinova_api_.getAngularCommand(joint_angles);" for a test.

Please let me know if this brings any improvement.

kmpatel9, 2016-09-06:
Okay!

I will be able to try the fixes you suggested, only after I fix the firmware issu.

I shall get back as soon as I do so.

I really appreciate your prompt support Longfei!

Thank you,

Kunal

Error code 1015 means "Cannot find the requested device". This may due to your "resetting firmware". There is nothing to worry about that. You can get contact with support@kinova.ca<mailto:support@kinova.ca>, they are very experienced to handle the firmware problems.

kmpatel9, 2016-09-10:
Hello Longfei,

I got the firmware up an running. The issue was I programmed later version of hexfile to the arm than it was supposed to work with. Hence the bootloader was failing to exit bootload state. Solution was to program older hexfiles that I received from Kinova support ( Thanks be to them) in a sequence of upgrading. So That worked and I am all set to test the fixes.

Now, you mentioned kinova_comm.cpp. I only see jaco_comm.cpp in my repo. The namespace used for the functions is also different in that file as "JacoComm::". Further, the function  "getAngularCommand()" is not found any where, and hence catkin_make fails to build.

Let me know what I am missing here.

Regards,

Kunal

LongfeiProjects, 2016-09-12:
Hi Kunal,

What is your current firmware version? Are you using a jaco 2nd generation arm?
kinova_comm and getAngularCommand() are available in our latest ROS release, which coming along with new APIs. That's why you didn't find them in the "jaco-ros".

kmpatel9, 2016-09-12:
Longfei, I think the arm is first generation , firmware jaco_5_1_0_0002.hex. Also I was on another branch of the repository.
Now I switched to Master branch and your suggestion makes sense.
After switching branch, the change you suggested is already present.
I shall try the combined control requests and get back to you.

kmpatel9, 2016-09-13:
Hi Longfei,

The issue is resolved.

Arm is working as expected and correctly.

The problem was I was using branch "jaco-ros-master" for my entire work till now.( When I started, the master branch was failing to compile with catkin_make).

Switching to most recent master branch did the work.

Thank you very much for the prompt support you have provided us!!

Thanks you,

Kunal Patel

________________________________
From: Patel, Kunal Madhukar
Sent: Monday, September 12, 2016 10:57:01 AM
To: Kinovarobotics/kinova-ros
Subject: Re: [Kinovarobotics/kinova-ros] Jaco fails to hold joint angles when fingers are controlled (#67)

Longfei, I think the arm is first generation , firmware jaco_5_1_0_0002.hex. Also I was on another branch of the repository.
Now I switched to Master branch and your suggestion makes sense.
After switching branch, the change you suggested is already present.
I shall try the combined control requests and get back to you.

closed 2017-09-11 by martine1406