Kinovarobotics/kinova-ros #68: Drinking Mode Enabled (default)

siddarth31, 2016-09-08:
Hello,

We are using the mico arm with the kinova api. When we send Cartesian velocity using the SendAdvancedTrajectory call  the robot acts in the drinking mode (a velocity in +Y direction makes the arm wrist to rotate and it will go near the drinking position). The ActualSystemStatus for the DrinkingMode is 0 (enabled by default).  We do not have a Kinova joystick. 
How can we disable the DrinkingMode and achieve a normal Cartesian behavior for the velocity control using the api? 

Thanks!

Siddarth

LongfeiProjects, 2016-09-09:
Hi Sid,

If you are suspecting the drink mode, please try to set the following parameters to zero:
ClientConfigurations.DrinkingDistance (x offset for drink mode)
ClientConfigurations.DrinkingHeight (y offset for drink mode)
ClientConfigurations.DrinkingLenght (z offset for drink mode)

However, I get confirmed with our software expert, who created the "KinovaTypes.h": actually 0 means NOT active, and the comment in "KinovaTypes.h" was an error.

If you try the example from README, do you still have this problem?
rostopic pub -r 100 /m1n4a200_driver/in/cartesian_velocity kinova_msgs/PoseVelocity "{twist_linear_x: 0.0, twist_linear_y: 0.0, twist_linear_z: 0.0, twist_angular_x: 0.0, twist_angular_y: 0.0, twist_angular_z: 10.0}"

Please be noted that if robot is close to its limit, it may rotate to extend itself.

Best regards!
-----
Longfei

siddarth31, 2016-09-23:
Hi Longfei,

We tested the ClientConfigurations.DrinkingDistance, ClientConfigurations.DrinkingHeight and ClientConfigurations.DrinkingLenght and they are all set to zero. 

When we send twist_linear_x value the arm moves right but the wrist also rotates.
Also, when we send  twist_linear_y value the arm moves backwards but the wrist rotates as it moves back.
...like as if it is going for a drinking pose. 

This is same for the api as well as the publish command on termninal. 

In order to show this behavior we have uploaded a video on below youtube link.
https://www.youtube.com/watch?v=c1gA8AugfYg
(the robot wrist rotates when moving to right/left or going back)

We have also posted a video of our own teleoperation implementation (that is not using Kinova api) for comparison. 
https://www.youtube.com/watch?v=-OoKxfo9tCE

The reason why we uploaded both is to show that the robot is able to execute the translation without the wrist rotation with our implementation. 

Is the robot in drinking mode or there is something with the cartesian velocity in api that we can change to correct the teleoperation behavior?

Siddarth

LongfeiProjects, 2016-09-27:
Hello Siddarth,

I am sorry for the delayed response. I am transferred to other projects so that I won't pass by here as much as before.  But my colleague Sid (soli@kinova.ca) will continue on supporting all the ROS issues. I will remind him to check about this thread.

Thanks for providing video to us. Would you please let me know answers for following question?

1.  (The second video) a video of our own teleoperation implementation (that is not using Kinova api) for comparison.  
 - The second video seems pretty good. If you are not using Kinova api, how you control the motion of the robot. Is it a speed control or position control? Sharing a snippet of code is appropriated.

2. (The first video) In order to show this behavior we have uploaded a video on below youtube link.
 - The motion command and behaviour in this video seems totally different with the second video. If you command the robot to go certain pose which is not achievable, robot will rotate itself to get close to goal. I think this may happen in the video. Could you please give the same command as in the second video to have a better comparison. In addition, the camera in the second video is very stable and it helps to observe.

3. If you are using twist_linear_x, I suppose you are using Cartesian velocity control. The Kinova joystick is using velocity control as well (if in Cartesian control, Cartesian velocity). Could you drive the robot follow the same path (path in second video), and see if the rotation problem occurs.

4. The twist_linear_x belongs to PoseVelocity message, which finally passed through the api function: "sendAdvanceTrajectory". You can try to change the api to "sendBasicTrajectory" in "kinova_comm.cpp":

void KinovaComm::setCartesianVelocities(const CartesianInfo &velocities)
{
...
int result = kinova_api_.sendBasicTrajectory(kinova_velocity);
...
}

siddarth31, 2016-09-27:
Hi Longfei,

Here are the answers: 

1) In the second video we are computing the Joint velocity from the commanded Cartesian velocity. These joint velocities are then sent to the robot directly using the api. 

2) The motion commands are exactly same in the first and the second video (move left/right or move back)...they are not shown in the same order in the videos. 

3) We do not have the Kinova Joystick. But yes, we used the api to send the Cartesian velocities. 
The wrist rotates even with pure translation command when the cartesian control is used in the api.
It may be because of the Drinking mode and if so how can be correct it?

4) I will try the "sendBasicTrajectory" function and will report any updates.

Thanks!

Siddarth

siddarth31, 2016-09-28:
Update: We tested the sendBasicTrajectory function with the api...same behavior.
The wrist rotates (we are just giving the twist_linear_x in the command).

What should be the next steps to debug this?

Siddarth

soli7, 2016-10-22:
Hi Siddharth,

If you still have this issue, try this - In the API setFrameType to 0.
frameType	The frame type. 0 = fixed frame, 1 = rotating frame.

This is not related to the drinking mode.

siddarth31, 2016-11-07:
Hi Soli,

Changing the frame type fixed it...Thanks!

Siddarth

closed 2016-11-07 by siddarth31