Kinovarobotics/kinova-ros #73: JACO version 1 URDF model missing?

fredrikbaberg, 2016-09-28:
I need to write code that works with both JACO and JACO2, however when I tried to launch with kinova_robotType:=j1n6a300 I get an error message that the file 'j1n6a300_standalone.xacro' does not exist. I looked in the folder, and there are no URDF files for JACO version 1?

How can the new API be used with both versions of the arm - with URDF model?

soli7, 2016-10-03:
Hi fredrikbaberg,

Jaco 1 and Jaco 2 have almost identical geometry. I recommend using the j2n6a300 (for j1n6a300) for now. We will add an accurate URDF for Jaco 1 in the next release. If you find any issues, let us know. Thanks.

~Sid

fredrikbaberg, 2016-10-05:
Thank you, then we will try the j2na300 model for now.

closed 2016-10-05 by fredrikbaberg