Kinovarobotics/kinova-ros #75: j2n6a300_state_publisher publishing to the wrong namespace?

fredrikbaberg, 2016-10-06:
I've embedded "kinova_bringup/launch/kinova_robot.launch" in a namespace (e.g. "kinova_ns"), which leads to that almost all topics has the correct prefix. The exception is the topic "/j2n6a300_driver/out/joint_state" (subscribed to by "j2n6a300_state_publisher").

I don't think this causes troubles, but it does not look as nice.

fredrikbaberg, 2016-10-10:
Turns out the problem is slightly larger than I thought: By including the launchfile in a namespace (<group ns="kinova_ns">) the TF tree breaks. That means it seems I can't use namespace with arm-related code for now.

closed 2017-09-11 by martine1406

fredrikbaberg, 2018-03-01:
It would have been nice if this had been solved, the same issue is still present.

martine1406, 2018-03-01:
Hi @fredrikbaberg 

In the current version of the ros package, there is no more j2n6a300... you should use j2n6s300 prefix if you are working on a jaco2 arm. This is why we did not fix the issue since we do not allow using j2n6a300 anymore. Do you get the same namespace problem for j2n6s300?

fredrikbaberg, 2018-03-01:
Hi @martine1406, the same issue is there with j2n6s300. See pull request #76.