Kinovarobotics/kinova-ros #77: Finger position does not update in RViz

fredrikbaberg, 2016-10-10:
The finger positions of the robot model does not update in RViz.

andre-nguyen, 2016-10-25:
Same problem here, did you manage to establish the root cause?

fredrikbaberg, 2016-10-25:
No, I didn't, but haven't had time to look more at it.

andre-nguyen, 2016-10-26:
I'm not sure I have time to inverstigate myself but I noticed the tf for the fingers were floating while I was investigating another issue.

[image]

pomerlef, 2016-10-28:
Same here. It's not related to Rviz, the tf for the fingers are not published.

The other links are published here: https://github.com/Kinovarobotics/kinova-ros/blob/master/kinova_driver/src/kinova_arm_kinematics.cpp#L214

But I couldn't find where any of the `<id>_link_finger_X` would be published as a tf.

ksenglee, 2016-10-29:
If the issue is similar to what I had encountered, the finger positions can be shown in rviz by modifying kinova_arm.cpp; specifically:
void KinovaArm::publishJointAngles ( void ) {
 ....
 if(finger_number_==2)
    {
        joint_state.position[joint_total_number_-2] = 0; // these need to be changed accordingly
        joint_state.position[joint_total_number_-1] = 0;
    }
    else if(finger_number_==3)
    {
        joint_state.position[joint_total_number_-3] = 0;
        joint_state.position[joint_total_number_-2] = 0;
        joint_state.position[joint_total_number_-1] = 0;
    } 
}

soli7, 2016-11-03:
Thanks for reporting the issue. Currently finger positions are not published as a part of joint_states but as a separate topic and that is why RViz does not show it. It will be fixed in the next release. 
If you would like fix it for you now. 

Change kinova_arm.cpp [here](https://github.com/Kinovarobotics/kinova-ros/blob/24b856ad273b071bbd909ea0e7e2c2ee397b0e1d/kinova_driver/src/kinova_arm.cpp#L341)
 `if(finger_number_==2)
    {
        joint_state.position[joint_total_number_-2] = fingers.Finger1/6800*80*M_PI/180;
        joint_state.position[joint_total_number_-1] = fingers.Finger2/6800*80*M_PI/180;
    }
    else if(finger_number_==3)
    {
        joint_state.position[joint_total_number_-3] = fingers.Finger1/6800*80*M_PI/180;
        joint_state.position[joint_total_number_-2] = fingers.Finger2/6800*80*M_PI/180;
        joint_state.position[joint_total_number_-1] = fingers.Finger3/6800*80*M_PI/180;
    }`

pomerlef, 2016-11-03:
Thanks, the patch cover partially the problem. The topic `<prefix>/out/joint_state` does now update the finger values, but nothing catches those and send them as on a `tf` broadcast.

The only `tf` broadcaster I could find is [here](https://github.com/Kinovarobotics/kinova-ros/blob/master/kinova_driver/src/kinova_arm_kinematics.cpp#L214), but it only loops through the `tf` for the arm joints, not the fingers. It doesn't look trivial how to add those based on the JointState message.

soli7, 2016-11-03:
The robot state publisher launched in the launch file in kinova_bringup subscribes to joint_states and publishes the tf. RViz shows finger positions for me.

pomerlef, 2016-11-04:
Alright, @soli7 thanks for the clarification. I did not realized that a `robot_state_publisher` was launched. I'll investigate what's going on starting from there. I'm using the launch file with the robot type `j2n6a300`.

Btw, thanks for providing the driver with the 3D model and the interactive control. It's appreciated.

closed 2017-09-11 by martine1406