Kinovarobotics/kinova-ros #88: Kinova launch question

yangyulounk, 2017-02-27:
I am using the Kinova-ROS package for a project. After installation the package from github, I tried to bring up the model in rviz. However, when launching the driver, the error messages appear repeatly:    

[ INFO] [1488228894.606556572]: Initializing Kinova API (header version: 50200, library version: 5.2.0) [ERROR] [1488228894.611927608]: KinovaCommException: Could not initialize Kinova API (return code: 1015)    

I am using Ubuntu 14.04 and ROS indigo, and I have installed the Kinova SDK with Kinova API. Do anyone recognized the problem? I would really appreciate your help.

soli7, 2017-02-28:
Ok the error 1015 means that the robot is not found. Make sure the robot is connected and that you copied the udev rules file. If you have the SDK can you connect to the robot using the DeveopmentCenter?

yangyulounk, 2017-05-02:
I am working on the arm simulation model in Gazebo. It is glad to see that you guys released the Gazebo support in beta. However, I have issues integrating the API. When launching kinova_control package to use setCartesianPosition to control the arm model, the program gives me error messages that it requires real robot arm informations. Is there way to directly use the inverse kinematics functions without the real robot present? Thanks.

martine1406, 2017-05-04:
Hi. I do not get how you use the kinova_control package to use setCartesianPosition. To control the gazebo model, you have to publish on '/(your robot)/effort_joint_trajectory_controller/command' (where your robot is j2n6s300 for e.g.). The gazebo model will not answer to Cartesian command directly, only angular commands. You need an inverse kinematics solver to be able to command the robot in Cartesian space. MoveIt could be a good resource for that.

closed 2017-05-08 by yangyulounk