Kinovarobotics/kinova-ros #89: urdf missing for j2n6s200

rongoldman-tri, 2017-03-15:
Hi. I'm working with a Jaco arm with a 2-finger hand (j2n6s200). The urdf files for it are not included in the kinova_ros/kinova_description/urdf folder. Attached are what I think the urdf files should be based on the latest versions for the j2n6s300 arm from the kinova-ros-beta. Can you please add them to the urdf folder? Thanks.

-- Ron --

p.s. Had to add a .txt suffix to each file so that I could attach them to the issue.

[j2n6s200.xacro.txt](https://github.com/Kinovarobotics/kinova-ros/files/845943/j2n6s200.xacro.txt)
[j2n6s200_standalone.xacro.txt](https://github.com/Kinovarobotics/kinova-ros/files/845944/j2n6s200_standalone.xacro.txt)

soli7, 2017-03-16:
Hi Ron, 

I have added the files to the beta branch. If you find any issues let me know. 

Best,
Sid

rongoldman-tri, 2017-03-16:
Sid,

Thanks very much! I'll be glad to share any other issues as we work with putting a Jaco arm on a mobile base to make a mobile home robot.

-- Ron --

rongoldman-tri, 2017-03-16:
Sid,

Just tried using the new files from github and ran into a problem:

There is still a reference to the 3-fingered arm at the end of the file j2n6s200_standalone.xacro:

  <xacro:j2n6s300  base_parent="${robot_root}"/>

changing it to:

  <xacro:j2n6s200  base_parent="${robot_root}"/>

makes things work.

Thanks for your help on this.

-- Ron --

soli7, 2017-03-16:
You are right! Fixed. 

Thanks for your help!

rongoldman-tri, 2017-03-16:
Everything now works great! Thanks.

closed 2017-03-16 by rongoldman-tri