Kinovarobotics/kinova-ros #90: Gazebo and Rviz simultaneously

pietroastolfi, 2017-03-30:
Is it already implemented a configuration which allows to communicate Rviz and Gazebo in order to obtain the same movement on gazebo while planning on Rviz?

martine1406, 2017-05-04:
yes. sorry about the delay in answering your question. you can launch 
roslauch kinova_gazebo robot_launch.launch  kinova_robotType:=(your robot)     (e.g. j2n6s300)
and then
roslaunch (your robot)_moveit_config (your robot)_gazebo_demo.launch

closed 2017-09-05 by martine1406

Tejas-Shastha, 2018-10-10:
This suggestion was useful in that I was able to have control over the gazebo sim using rviz. However, the interactive markers do not appear in rviz for controlling the gripper pose. Any suggestions? 
These markers do appear when I launch roslaunch j2n6s300_moveit_config j2n6s300_virtual_robot_demo.launch.

martine1406, 2018-10-10:
The interactive markers appear on my end when I launch:
`roslaunch j2n6s300_moveit_config j2n6s300_gazebo_demo.launch`
then
`roslaunch j2n6s300_moveit_config j2n6s300_virtual_robot_demo.launch`