Kinovarobotics/kinova-ros #94: Detecting singularities

pouyaAB, 2017-05-25:
Is there any way to know that the arm is stuck in one of the singularities? I know that two red lights will turn on when it stuck there but is there anyway to extract such information from the ROS?

Kinova, 2017-05-30:
Hi pouyaAB

Unfortunately, there is no way to check whether the real robot is stuck in singularity. Normally, the arm is programmed to avoid its own singularities though. Do you have problems with the arm getting stuck in singularity? Are you trying to control the arm in singularity or using the real robot?

Thanks

pouyaAB, 2017-05-31:
I wrote a controller for the arm with PS4 controller, I was trying to improve on that. When the arm is trying to avoid the singularities it moves much slower (and sometimes it stuck and need to go to the home) than in normal situations, I was trying to solve that issue.

martine1406, 2017-06-03:
Hi pouyaAB

Well if you want, I can provide you with a small technical sheet that tells you what are the singularities in the Jaco (or Mico) arm. Which arm are you using? At least it could be a start. You could check whether the arm is close to a singular config and react accordingly.

pouyaAB, 2017-06-04:
@martine1406 Yes that would be great. I am using the Mico arm.

martine1406, 2017-06-05:
Hi. Here is the file. Mico and Jaco 2 have the same singular configurations
[Jaco arm singularities.docx](https://github.com/Kinovarobotics/kinova-ros/files/1052189/Jaco.arm.singularities.docx)

closed 2017-09-05 by martine1406