Kinovarobotics/kinova-ros #95: FIngers don't move after command/roslaunch fails at specific situation

xxxxxzx, 2017-06-02:
I try to run the command "rosrun kinova_demo fingers_action_client.py m1n4s200 percent -- 20 20
" to move the fingers. 
Even though the screen shows:
"obtained current finger position 
Sending finger position ...
Finger position sent!"

The fingers cannot move. However, the joint part is good. Then, only after I manually close the fingers by joystick, no matter the values i type in the rosrun command, the fingers will directly open to the maximum. Is there a bug or did I miss something?

Also the above is under the situation when I roslaunch as use_"urdf = false". As soon as I try to change the urdf to "true" for roslaunch, the process will die. Is there anything associated with the problem I meet above? My ros environment is kinetic since i'm using ubuntu 16.04.

[image]

martine1406, 2017-06-03:
Hi

This ros package was developed for Ubuntu Indigo (14.04) and was never tested on Ubuntu kinetic. This might be the source of the problem. Do you have access to a computer with Ubuntu 14.04?

xxxxxzx, 2017-06-06:
Sorry I don't have Ubuntu 14.04, but things work well on my other friend's computer (16.04). Before the "roslaunch part", I've sourced setup.bash and setup.sh, is there anything I missed?

when I call "roslaunch kinova_bringup kinova_robot.launch kinova_robotType:=m1n4s200 use_urdf:=true"
The process will die and after I change it to false, it works. 

[image]

But as shown in the pic, there is not things like m1n4s200_finger_1 and m1n4s200_finger_2

Moreover, the process will still die after a few operations accidentally and I can only close all the terminals and redo things to bring it up again.

please help me, thanks :sob:

xxxxxzx, 2017-06-06:
And actually for the most times both use_urdf = true and = false lead to a dead process

martine1406, 2017-06-06:
Hi. It will be hard for me to test your specific case as I have Ubuntu 14.04 on my computer. Could you try to remove all other packages from your catkin workspace and check if you still get the same errors? I understand you have a 4dof mico arm with a 2 finger gripper. Could you confirm that information please? I'll check if I can repeat your problem on my computer tomorrow. 
Thanks

xxxxxzx, 2017-06-06:
I got this computer only for Kinova so there is no other package in the
catkin-ws. And yes that's the type. Thanks for help

martine1406, 2017-06-06:
Hi. I am sorry but I was not able to reproduce your problem on my computer. When you test **the same robot** with your friend's computer with Ubuntu 16.04, does everything work well? If so, you might have something on your computer that creates a confliect or something. Please let me know.

xxxxxzx, 2017-06-15:
ahhh now I am trying to use a ubuntu 14.04 laptop with indigo to install the kinova package, but there are even errors like these after I enter "catkin_make".

make[2]: *** [kinova-ros/kinova_moveit/inverse_kinematics_plugins/ikfast/j2s7s300_ikfast/CMakeFiles/j2s7s300_robot_arm_moveit_ikfast_plugin.dir/src/j2s7s300_robot_arm_ikfast_moveit_plugin.cpp.o] Error 1
make[1]: *** [kinova-ros/kinova_moveit/inverse_kinematics_plugins/ikfast/j2s7s300_ikfast/CMakeFiles/j2s7s300_robot_arm_moveit_ikfast_plugin.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
make[2]: *** [kinova-ros/kinova_moveit/inverse_kinematics_plugins/ikfast/j2n6s300_ikfast/CMakeFiles/j2n6s300_arm_moveit_ikfast_plugin.dir/src/j2n6s300_arm_ikfast_moveit_plugin.cpp.o] Error 1
make[1]: *** [kinova-ros/kinova_moveit/inverse_kinematics_plugins/ikfast/j2n6s300_ikfast/CMakeFiles/j2n6s300_arm_moveit_ikfast_plugin.dir/all] Error 2
make: *** [all] Error 2
Invoking "make -j2 -l2" failed

martine1406, 2017-06-15:
Hi

you just cloned the kinova-ros package into your catkin workspace and run catkin_make and got these errors? Have you also sudo apt-get install ros-indigo-moveit and sudo apt-get install ros-indigo-trac-ik? If not, please see if installing these two packages solve your issue

xxxxxzx, 2017-06-16:
there are errors still

[  1%] Building CXX object kinova-ros/kinova_moveit/inverse_kinematics_plugins/ikfast/j2n6s300_ikfast/CMakeFiles/j2n6s300_arm_moveit_ikfast_plugin.dir/src/j2n6s300_arm_ikfast_moveit_plugin.cpp.o
[  1%] Built target j2s7s300_robot_arm_moveit_ikfast_plugin
Scanning dependencies of target m1n6s300_mico_arm_moveit_ikfast_plugin
[  1%] Building CXX object kinova-ros/kinova_moveit/inverse_kinematics_plugins/ikfast/m1n6s300_ikfast/CMakeFiles/m1n6s300_mico_arm_moveit_ikfast_plugin.dir/src/m1n6s300_mico_arm_ikfast_moveit_plugin.cpp.o
c++: internal compiler error: Killed (program cc1plus)
Please submit a full bug report,
with preprocessed source if appropriate.
See <file:///usr/share/doc/gcc-4.8/README.Bugs> for instructions.
make[2]: *** [kinova-ros/kinova_moveit/inverse_kinematics_plugins/ikfast/j2n6s300_ikfast/CMakeFiles/j2n6s300_arm_moveit_ikfast_plugin.dir/src/j2n6s300_arm_ikfast_moveit_plugin.cpp.o] Error 4
make[1]: *** [kinova-ros/kinova_moveit/inverse_kinematics_plugins/ikfast/j2n6s300_ikfast/CMakeFiles/j2n6s300_arm_moveit_ikfast_plugin.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
Linking CXX shared library /home/user/catkin_ws/devel/lib/libm1n6s300_mico_arm_moveit_ikfast_plugin.so
[  1%] Built target m1n6s300_mico_arm_moveit_ikfast_plugin
make: *** [all] Error 2
Invoking "make -j2 -l2" failed

xxxxxzx, 2017-06-16:
It seems working, but when I enter the commands:

cd ~/catkin_ws/src/kinova_ros
sudo cp kinova_driver/udev/10-kinova-arm.rules /etc/udev/rules.d/

do I need to manually create a directory of "/etc/udev/rules.d/" and what is the repository?

Since after I do this and call the roslaunch command as before, there is still error:

process[m1n4s200_driver-2]: started with pid [91833]
process[m1n4s200_tf_updater-3]: started with pid [91842]
[ INFO] [1497574245.321690130]: kinova_robotType is m1n4s200.
[ INFO] [1497574245.387124571]: Initializing Kinova USB API (header version: 50300, library version: 5.2.0)
[ INFO] [1497574245.387965093]: kinova_robotType is m1n4s200.
[ERROR] [1497574245.388764350]: KinovaCommException: Could not initialize Kinova API (return code: 1015)

[m1n4s200_driver-2] process has died [pid 91833, exit code -11, cmd /home/user/catkin_ws/devel/lib/kinova_driver/kinova_arm_driver m1n4s200 __name:=m1n4s200_driver __log:=/home/user/.ros/log/d43f7c5c-522d-11e7-a052-000c296f9fda/m1n4s200_driver-2.log].
log file: /home/user/.ros/log/d43f7c5c-522d-11e7-a052-000c296f9fda/m1n4s200_driver-2*.log

martine1406, 2017-06-16:
Hi

Is the file 10-kinova-arm.rules now appearing in your /etc/udev/rules.d/ directory? Error 1015 means that the connection with the robot cannot be established. Is your robot connected USB? Can you try downloading Kinova SDK from our website. http://www.kinovarobotics.com/fr/robotique-innovation/produits/logiciel/. Take Kinova SDK for MICO2 and follow the install instructions found here (http://kinovarobotics.com/wp-content/uploads/2015/03/Kinova-SDK-User-Guide-1.0.6.pdf). Afterward, open the Dev Center and check if you can connect with the robot; its serial number should appear in the top right corner.

xxxxxzx, 2017-06-19:
alright I download the SDK and there is serial number appearing at the top right corner, but unfortunately, when use the Virtual joystick in DevCenter, for the buttons of the Gripper, "Open 2 fingers" and "Close 2 fingers“ still respond nothing, and the Angular control works well

martine1406, 2017-06-20:
You have to open the gripper completely before you can close it. This is an
initialisation procedure. Can you confirm that it still does not work?

-- 

*Martine Blouin, ing. jr.*

Robotic control specialist - project manager (R&D)

*mblouin@kinova. <alecours@kinovarobotics.com>ca*

T: 514-277-3777 # 4034

F: 514-635-6577

4333, rue Grande-Allée
Boisbriand, QC, Canada J7H 1M7

KinovaRobotics.com

Avis sur la confidentialité et avertissement relatif à la Loi sur l'accès
aux documents et sur la protection des renseignements personnels.
L'information transmise par ce courriel est de nature privilégiée et
confidentielle. Elle est destinée à l'usage exclusif du destinataire
ci-dessus. Si vous n'êtes pas le destinataire visé, vous êtes par la
présente avisé qu'il est strictement interdit d'utiliser cette information,
de la copier, de la distribuer ou la diffuser. Si cette communication vous
a été transmise par erreur, veuillez la détruire et nous en aviser
immédiatement par courriel info@kinovarobotics.com

xxxxxzx, 2017-06-26:
ok Development center works now, but those command scripts still don't work.
[image]
[image]
[image]
[image]
[image]

xxxxxzx, 2017-06-26:
Here is when I try to open the finger to 100:
[image]
[image]

No matter what number I entered, the finger will always open to 100.
[image]
[image]

Kinova, 2017-06-28:
Hi. If the bringup does not work, then you cannot control the fingers. Do you succeed to connect to the arm? Do you robot's fingers open? The 'entering destruction' appears because the JointTrajectoryController destructor is called. JointTrajectoryController constructor is called by kinova_driver. When launching kinova_bringup node, are you running other nodes at the same time? When do you get the output shown in this image 

[image]

ybisk, 2017-06-28:
Hi

Thank you for all your replies.  I am using the same device as @xxxxxzx and I have the following replication.  I have performed this inside VirtualBox so I can send you the drive image if you'd like.  This is a completely blank and clean install of Ubuntu 14.04.  

Thanks for your help!

# Install ubuntu 14.04 desktop amd64
``` bash
# http://wiki.ros.org/kinetic/Installation/Ubuntu
$ sudo sh -c 'echo "deb http://packages.ros.org/ros/ubuntu $(lsb_release -sc) main" > /etc/apt/sources.list.d/ros-latest.list'
$ sudo apt-key adv --keyserver hkp://ha.pool.sks-keyservers.net:80 --recv-key 421C365BD9FF1F717815A3895523BAEEB01FA116
$ sudo apt-get install ros-indigo-desktop-full

$ sudo apt-get install git vim

$ unzip KINOVA SDK MICO2.zip
$ cd KINOVA SDK MICO2/Ubuntu/64 bits
$ ./installSDK64.sh     # All values checked for install

$ sudo rosdep init
$ rosdep update
$ echo "source /opt/ros/indigo/setup.bash" >> ~/.bashrc
$ source ~/.bashrc
$ sudo apt-get install python-rosinstall

$ cd ~/
$ mkdir catkin_ws
$ mkdir catkin_ws/src
$ cd ~/catkin_ws/src
$ git clone https://github.com/Kinovarobotics/kinova-ros.git kinova-ros
$ cd ~/catkin_ws
$ catkin_make

# -- failed, requires moveit --
$ sudo apt-get install ros-indigo-moveit
$ catkin_make
# -- built successfully -- 

$ sudo cp srd/kinova-ros/kinova_dirver/udev/10-kinova-arm.rules /etc/udev/rules.d/
$ source devel/setup.sh
$ source devel/setup.bash
$ roslaunch kinova_bringup kinova_robot.launch kinova_robotType:=m1n4s200
... started core service [/rosout]
... kinova_robotType is m1n4s200
... Initializing Kinova USB API
[ERROR] ... KinovaCommException: Could not initialize Kinova API (return code: 1015)   
# -- This is line 155 of src/kinova_comm.cpp --

$ sudo apt-get install input-utils
$ sudo shutdown -r now 
# -- change virtualbox usb to include 3.0 --
# -- Reboot for USB connection inside virtual box -- 
$ source catkin_ws/devel/setup.sh
$ source catkin_ws/devel/setup.bash
$ roslaunch kinova_bringup kinova_robot.launch kinova_robotType:=m1n4s200
... The arm is ready to use
... m1n4s200_joint_1 m1n4s200_joint_2 m1n4s200_joint_3 m1n4s200_joint_4

# -- New Tab --
$ cd ~/catkin_ws/src/kinova-ros/kinva_demo/nodes/kinova_demo
$ source catkin_ws/devel/setup.sh
$ source catkin_ws/devel/setup.bash
$ rosrun kinova_demo joints_action_client.py -v -r m1n4s200 degree -- 10 10 10 10    # Success
... joint values sent in radian are:
...

$ rosrun kinova_demo fingers_action_client.py m1n4s200 percent -- 100 100            # No change, already open
... obtained current finger position
... Sending finger position ...
... Finger position sent!
$ rosrun kinova_demo fingers_action_client.py m1n4s200 percent -- 10 10              # No change
... obtained current finger position
... Sending finger position ...
... Finger position sent!
```

After this, to test the hardware I
``` bash
$ cd /opt/kinova/GUI
$ ./DevelopmentCenter.sh
```
 Here the virtual joystick opens and closes the fingers no problem.  Please let me know if there is a debugging step I can do to help locate the problem.  Thanks!

ybisk, 2017-06-28:
I'm going to follow-up here with the output of bringup during the commands pasted previously:

``` bash
[DEBUG] [1498676493.037489712]: The action server has received a new goal request
[DEBUG] [1498676493.037891826]: A new goal has been recieved by the single goal action server
[DEBUG] [1498676493.038175832]: Accepting a new goal
[DEBUG] [1498676493.038317640]: Accepting goal, id: /m1n4s200_gripper_workout-1-1498676493.037, stamp: 1498676493.04
[DEBUG] [1498676493.563896895]: Setting the current goal as succeeded
[DEBUG] [1498676493.564089634]: Setting status to succeeded on goal, id: /m1n4s200_gripper_workout-1-1498676493.037, stamp: 1498676493.04
[DEBUG] [1498676493.564533696]: Publishing result for goal with id: /m1n4s200_gripper_workout-1-1498676493.037 and stamp: 1498676493.04
[ WARN] [1498676493.564890590]: void kinova::KinovaAnglesActionServer::actionCallback(const ArmJointAnglesGoalConstPtr&): LINE 137, setSucceeded
[DEBUG] [1498676501.265010478]: The action server has received a new goal request
[DEBUG] [1498676501.265342785]: A new goal has been recieved by the single goal action server
[DEBUG] [1498676501.265582422]: Accepting a new goal
[DEBUG] [1498676501.265682370]: Accepting goal, id: /m1n4s200_gripper_workout-1-1498676501.265, stamp: 1498676501.26
[DEBUG] [1498676502.213888913]: Setting the current goal as succeeded
[DEBUG] [1498676502.214088249]: Setting status to succeeded on goal, id: /m1n4s200_gripper_workout-1-1498676501.265, stamp: 1498676501.26
[DEBUG] [1498676502.214507315]: Publishing result for goal with id: /m1n4s200_gripper_workout-1-1498676501.265 and stamp: 1498676501.26
[ WARN] [1498676502.214818363]: void kinova::KinovaAnglesActionServer::actionCallback(const ArmJointAnglesGoalConstPtr&): LINE 137, setSucceeded
[DEBUG] [1498676569.905216599]: The action server has received a new goal request
[DEBUG] [1498676569.905567950]: A new goal has been recieved by the single goal action server
[DEBUG] [1498676569.905958592]: Accepting a new goal
[DEBUG] [1498676569.906124104]: Accepting goal, id: /m1n4s200_gripper_workout-1-1498676569.905, stamp: 1498676569.90
[DEBUG] [1498676570.495813874]: Setting the current goal as aborted
[DEBUG] [1498676570.495987420]: Setting status to aborted on goal, id: /m1n4s200_gripper_workout-1-1498676569.905, stamp: 1498676569.90
[DEBUG] [1498676570.496311489]: Publishing result for goal with id: /m1n4s200_gripper_workout-1-1498676569.905 and stamp: 1498676569.90
[DEBUG] [1498676570.497134336]: void kinova::KinovaFingersActionServer::actionCallback(const SetFingersPositionGoalConstPtr&): LINE 165, Trajectory command failed
[DEBUG] [1498676575.300428727]: The action server has received a new goal request
[DEBUG] [1498676575.300904436]: A new goal has been recieved by the single goal action server
[DEBUG] [1498676575.301210903]: Accepting a new goal
[DEBUG] [1498676575.301315099]: Accepting goal, id: /m1n4s200_gripper_workout-1-1498676575.300, stamp: 1498676575.30
[DEBUG] [1498676575.893776399]: Setting the current goal as aborted
[DEBUG] [1498676575.893943476]: Setting status to aborted on goal, id: /m1n4s200_gripper_workout-1-1498676575.300, stamp: 1498676575.30
[DEBUG] [1498676575.894383118]: Publishing result for goal with id: /m1n4s200_gripper_workout-1-1498676575.300 and stamp: 1498676575.30
[DEBUG] [1498676575.894724575]: void kinova::KinovaFingersActionServer::actionCallback(const SetFingersPositionGoalConstPtr&): LINE 165, Trajectory command failed
[ WARN] [1498677187.578858573]: destruction entered!
[FATAL] [1498677187.585281598]: ASSERTION FAILED
        file = /opt/ros/indigo/include/ros/publisher.h
        line = 102
        cond = false
        message =
[FATAL] [1498677187.585309890]: Call to publish() on an invalid Publisher
[FATAL] [1498677187.585324344]:

[m1n4s200_driver-2] process has died [pid 2261, exit code -5, cmd /home/demo/catkin_ws/devel/lib/kinova_driver/kinova_arm_driver m1n4s200 __name:=m1n4s200_driver __log:=/home/demo/.ros/log/93b0c60e-5c33-11e7-ad37-0800271f96b9/m1n4s200_driver-2.log].
log file: /home/demo/.ros/log/93b0c60e-5c33-11e7-ad37-0800271f96b9/m1n4s200_driver-2*.log
```

The log files are attached here:
[logs.tar.gz](https://github.com/Kinovarobotics/kinova-ros/files/1109993/logs.tar.gz)

Thanks

martine1406, 2017-06-29:
Hi. Just to let you know, I was very busy in the last few days but I'll take the time to investigate more this issue next Monday. Stay tuned and thank you for your feedback

Kinova, 2017-07-03:
Hi
Okay so I investigated the issue, and it turns out that there was an error in kinova_joint_trajectory_controller.cpp. For a 4 dof robot, traj_feedback_msg_ fields were resized to 4, but in commandCB, at least 6 fields of traj_command_points_ are read by default. I corrected this file in a new commit. 

For the finger motion problem, I found a problem with the sendAdvanceTrajectory API function for a 4 dof. I patched the problem for the moment by calling at the sendBasicTrajectory function instead in setFingerPositions (file kinova_comm.cpp). I tested the fix with different mico 4 dof firmware version. I have created a new commit for this correction too.  

So in total, I have created two new commits. 
Let me know if it works for you
Martine

ybisk, 2017-07-03:
Thanks!  You can close this issue. 
I can confirm this works in Ubuntu 14.04 and 16.04 with indigo and kinetic ROS, respectively.

In 16.04 I did have to edit 

```bash
catkin_ws/src/CMakeLists.txt
```
to add the line
```bash
add_compile_options(-std=c++11)
```

closed 2017-07-11 by Kinova