Kinovarobotics/kinova-ros #98: catkin_make error

victor29z, 2017-07-19:
When I invoking catkin_make to build this package, there is an error as shown here. The platform is  ROS Indigo with 64 bit Ubuntu 14.04.

Building CXX object kinova-ros/kinova_driver/CMakeFiles/kinova_driver.dir/src/kinova_arm.cpp.o
In file included from /opt/ros/indigo/include/ros/ros.h:45:0,
                 from /home/turtlebot/catkin_ws/src/kinova-ros/kinova_driver/include/kinova_driver/kinova_arm.h:13,
                 from /home/turtlebot/catkin_ws/src/kinova-ros/kinova_driver/src/kinova_arm.cpp:9:
/opt/ros/indigo/include/ros/node_handle.h: In instantiation of ‘void ros::NodeHandle::param(const string&, T&, const T&) const [with T = float; std::string = std::basic_string<char>]’:
/home/turtlebot/catkin_ws/src/kinova-ros/kinova_driver/src/kinova_arm.cpp:264:82:   required from here
**/opt/ros/indigo/include/ros/node_handle.h:1916:7: error: no matching function for call to ‘ros::NodeHandle::getParam(const string&, float&) const’**
       if (getParam(param_name, param_val))
       ^
/opt/ros/indigo/include/ros/node_handle.h:1916:7: note: candidates are:
/opt/ros/indigo/include/ros/node_handle.h:1482:8: note: bool ros::NodeHandle::getParam(const string&, std::string&) const
   bool getParam(const std::string& key, std::string& s) const;
        ^
/opt/ros/indigo/include/ros/node_handle.h:1482:8: note:   no known conversion for argument 2 from ‘float’ to ‘std::string& {aka std::basic_string<char>&}’
/opt/ros/indigo/include/ros/node_handle.h:1493:8: note: bool ros::NodeHandle::getParam(const string&, double&) const
   bool getParam(const std::string& key, double& d) const;
        ^
/opt/ros/indigo/include/ros/node_handle.h:1493:8: note:   no known conversion for argument 2 from ‘float’ to ‘double&’
/opt/ros/indigo/include/ros/node_handle.h:1504:8: note: bool ros::NodeHandle::getParam(const string&, int&) const
   bool getParam(const std::string& key, int& i) const;
        ^
/opt/ros/indigo/include/ros/node_handle.h:1504:8: note:   no known conversion for argument 2 from ‘float’ to ‘int&’
/opt/ros/indigo/include/ros/node_handle.h:1515:8: note: bool ros::NodeHandle::getParam(const string&, bool&) const
   bool getParam(const std::string& key, bool& b) const;
        ^
/opt/ros/indigo/include/ros/node_handle.h:1515:8: note:   no known conversion for argument 2 from ‘float’ to ‘bool&’
/opt/ros/indigo/include/ros/node_handle.h:1526:8: note: bool ros::NodeHandle::getParam(const string&, XmlRpc::XmlRpcValue&) const
   bool getParam(const std::string& key, XmlRpc::XmlRpcValue& v) const;
        ^
/opt/ros/indigo/include/ros/node_handle.h:1526:8: note:   no known conversion for argument 2 from ‘float’ to ‘XmlRpc::XmlRpcValue&’
/opt/ros/indigo/include/ros/node_handle.h:1538:8: note: bool ros::NodeHandle::getParam(const string&, std::vector<std::basic_string<char> >&) const
   bool getParam(const std::string& key, std::vector<std::string>& vec) const;
        ^
/opt/ros/indigo/include/ros/node_handle.h:1538:8: note:   no known conversion for argument 2 from ‘float’ to ‘std::vector<std::basic_string<char> >&’
/opt/ros/indigo/include/ros/node_handle.h:1549:8: note: bool ros::NodeHandle::getParam(const string&, std::vector<double>&) const
   bool getParam(const std::string& key, std::vector<double>& vec) const;
        ^
/opt/ros/indigo/include/ros/node_handle.h:1549:8: note:   no known conversion for argument 2 from ‘float’ to ‘std::vector<double>&’
/opt/ros/indigo/include/ros/node_handle.h:1560:8: note: bool ros::NodeHandle::getParam(const string&, std::vector<float>&) const
   bool getParam(const std::string& key, std::vector<float>& vec) const;
        ^
/opt/ros/indigo/include/ros/node_handle.h:1560:8: note:   no known conversion for argument 2 from ‘float’ to ‘std::vector<float>&’
/opt/ros/indigo/include/ros/node_handle.h:1571:8: note: bool ros::NodeHandle::getParam(const string&, std::vector<int>&) const
   bool getParam(const std::string& key, std::vector<int>& vec) const;
        ^
/opt/ros/indigo/include/ros/node_handle.h:1571:8: note:   no known conversion for argument 2 from ‘float’ to ‘std::vector<int>&’
/opt/ros/indigo/include/ros/node_handle.h:1582:8: note: bool ros::NodeHandle::getParam(const string&, std::vector<bool>&) const
   bool getParam(const std::string& key, std::vector<bool>& vec) const;
        ^
/opt/ros/indigo/include/ros/node_handle.h:1582:8: note:   no known conversion for argument 2 from ‘float’ to ‘std::vector<bool>&’
/opt/ros/indigo/include/ros/node_handle.h:1594:8: note: bool ros::NodeHandle::getParam(const string&, std::map<std::basic_string<char>, std::basic_string<char> >&) const
   bool getParam(const std::string& key, std::map<std::string, std::string>& map) const;
        ^
/opt/ros/indigo/include/ros/node_handle.h:1594:8: note:   no known conversion for argument 2 from ‘float’ to ‘std::map<std::basic_string<char>, std::basic_string<char> >&’
/opt/ros/indigo/include/ros/node_handle.h:1605:8: note: bool ros::NodeHandle::getParam(const string&, std::map<std::basic_string<char>, double>&) const
   bool getParam(const std::string& key, std::map<std::string, double>& map) const;
        ^
/opt/ros/indigo/include/ros/node_handle.h:1605:8: note:   no known conversion for argument 2 from ‘float’ to ‘std::map<std::basic_string<char>, double>&’
/opt/ros/indigo/include/ros/node_handle.h:1616:8: note: bool ros::NodeHandle::getParam(const string&, std::map<std::basic_string<char>, float>&) const
   bool getParam(const std::string& key, std::map<std::string, float>& map) const;
        ^
/opt/ros/indigo/include/ros/node_handle.h:1616:8: note:   no known conversion for argument 2 from ‘float’ to ‘std::map<std::basic_string<char>, float>&’
/opt/ros/indigo/include/ros/node_handle.h:1627:8: note: bool ros::NodeHandle::getParam(const string&, std::map<std::basic_string<char>, int>&) const
   bool getParam(const std::string& key, std::map<std::string, int>& map) const;
        ^
/opt/ros/indigo/include/ros/node_handle.h:1627:8: note:   no known conversion for argument 2 from ‘float’ to ‘std::map<std::basic_string<char>, int>&’
/opt/ros/indigo/include/ros/node_handle.h:1638:8: note: bool ros::NodeHandle::getParam(const string&, std::map<std::basic_string<char>, bool>&) const
   bool getParam(const std::string& key, std::map<std::string, bool>& map) const;
        ^
/opt/ros/indigo/include/ros/node_handle.h:1638:8: note:   no known conversion for argument 2 from ‘float’ to ‘std::map<std::basic_string<char>, bool>&’
**make[2]: *** [kinova-ros/kinova_driver/CMakeFiles/kinova_driver.dir/src/kinova_arm.cpp.o] Error 1
make[1]: *** [kinova-ros/kinova_driver/CMakeFiles/kinova_driver.dir/all] Error 2
make: *** [all] Error 2
Invoking "make -j8 -l8" failed**

martine1406, 2017-08-03:
Hi victor29z. sorry for the delayed response. did you find an answer to your problem? The error seems to come from standard roscpp Nodehandles (ros /opt/ros/indigo/include/ros/node_handle.h). Something wrong with your particular ros intallation? I never got this error on my side, but if it persists, I'll investigate

nehagarg, 2017-08-24:
i am getting the same error. Did you find any solution?

martine1406, 2017-08-28:
I did not investigate since I did not get any feedback. As I said, it seems to be an error with roscpp Nodehandles (i.e. not specific to Kinova package?). Maybe search on Google for this specific error? e.g. this thread: https://answers.ros.org/question/253797/no-matching-function-for-call-to-rosnodehandleparam/

nehagarg, 2017-08-29:
It was solved by reinstalling ros indigo. Thanks!

closed 2017-08-29 by martine1406