Kinovarobotics/kinova-ros #101: Error using kinova_ros wti Ubuntu 16.04

martine1406, 2017-09-11:
From <s.alvarezl@alumnos.upm.es>

Dear all,

I have been using the robotic arm Jaco2 with ROS Indigo in Ubuntu 14.04 as recomended, but now we want to upgrade to ROS Kinetic in Ubuntu 16.04 in order to use the newest versions of other software we need to use.

This might have caused problems. After cloning the packages into the ROS workspace the catkin_make is not working. There are some messages as the shown below:

-------------

In file included from /opt/ros/kinetic/include/moveit/macros/class_forward.h:40:0,
                 from /opt/ros/kinetic/include/moveit/kinematics_base/kinematics_base.h:42,
                 from /home/sebastian/catkin_ws/src/kinova-ros/kinova_moveit/inverse_kinematics_plugins/ikfast/j2s7s300_ikfast/src/j2s7s300_robot_arm_ikfast_moveit_plugin.cpp:49:
/opt/ros/kinetic/include/moveit/macros/declare_ptr.h:52:16: error: ‘shared_ptr’ in namespace ‘std’ does not name a template type
   typedef std::shared_ptr<Type> Name##Ptr;

-------------

And at the end of the process shows the following:

-------------

Makefile:138: recipe for target 'all' failed
make: *** [all] Error 2
Invoking "make -j4 -l4" failed

-------------

I would like to know if you or someone else have tried to use the kinova_ros package in ROS Kinetic and if you can help us to solve this problem

Thank you.
Regards.

martine1406, 2017-09-11:
Hi

Although this package was developed for Ubuntu 14.04, I heard of people who made it work also on 16.04. The solution to your problem seems to enable C++11 in your CMakeLists.txt? see https://github.com/ros-planning/moveit/issues/462

martine1406, 2017-09-12:
I would say you have to edit

catkin_ws/src/CMakeLists.txt

to add the line

add_compile_options(-std=c++11)

sealvarezlu, 2017-09-14:
Now it works, thank you.

At first I edited:  catkin_ws/src/CMakeLists.txt  
But it didn't work.

Then I edited all the CMakeLists contained into the Moveit packages inside kinova_ros packages, adding the line 

add_compile_options(-std=c++11)

catkin_make
And done!

Thank you.

closed 2017-09-15 by martine1406