Kinovarobotics/kinova-ros #102: ROS Jaco2 package compiling errors and temporary solution

martine1406, 2017-09-15:
Hi, all, 

Thanks for sharing the update. Here is a quick update of the work I did with ROS package on the Jaco2 we received. I tried to install Kinova ROS package following this site but received errors during compiling and I found some tricks to pass them.
https://github.com/Kinovarobotics/kinova-ros

Here is the system I was using. 
Up board (Intel Atom x5-z8350 1.44GHZx4 4GRAM, http://www.up-board.org/up/comparison-up-versus-edisongalileo-joule/)
Ubuntu 16.04 LTS 64bit
ROS Kinetic (http://wiki.ros.org/kinetic/Installation/Ubuntu)

The error messages after git clone and catkin_make are in the attached file. As most errors are due to the shared_ptr from moveit, I added "add_compile_options(-std=c++11)" in the CMakeList to enable the C++ 11 and it seemed work fine. There might be better ways but this is just my quick temporary solution. I can get the tool_pose topic and call the home service with the USB port. I am still testing other functions and will try the ethernet port. 

Thanks,

Cheng-Shiu (Joshua) Chung

martine1406, 2017-09-15:
#101

martine1406, 2017-09-15:
Similar to issue #101

Hatcher, 2017-09-30:
Have you been able to use the gazebo simulator after building? I am getting the error:
`Error [Param.cc:451] Unable to set value [-nan -nan -nan] for key[size]
gzclient: /build/ogre-1.9-mqY1wq/ogre-1.9-1.9.0+dfsg1/OgreMain/src/OgreNode.cpp:630: virtual void Ogre::Node::setScale(const Ogre::Vector3&): Assertion `!inScale.isNaN() && "Invalid vector supplied as parameter"' failed.
`

Any thoughts on how to fix it?

martine1406, 2017-10-02:
Hi.

You can have a look at this issue and solution
https://github.com/ros-simulation/gazebo_ros_pkgs/issues/488. Are you using
gazebo 7 by any chance? These nan errors are caused by dimension zero
collision geometry. This package was developed for gazebo 2 initially.

-- 

*Martine Blouin, ing.*

Innovation Field Application Specialist

*mblouin@kinova. <alecours@kinovarobotics.com>ca*

T: 514-277-3777 # 4034

F: 514-635-6577

4333, rue Grande-Allée
Boisbriand, QC, Canada J7H 1M7

KinovaRobotics.com

Avis sur la confidentialité et avertissement relatif à la Loi sur l'accès
aux documents et sur la protection des renseignements personnels.
L'information transmise par ce courriel est de nature privilégiée et
confidentielle. Elle est destinée à l'usage exclusif du destinataire
ci-dessus. Si vous n'êtes pas le destinataire visé, vous êtes par la
présente avisé qu'il est strictement interdit d'utiliser cette information,
de la copier, de la distribuer ou la diffuser. Si cette communication vous
a été transmise par erreur, veuillez la détruire et nous en aviser
immédiatement par courriel info@kinovarobotics.com

closed 2018-03-15 by martine1406