Kinovarobotics/kinova-ros #103: Spawning jaco2 in Gazebo: Error [Param.cc:452] Unable to set value [-nan -nan -nan] for key[size]

tttor, 2017-10-10:
# running:
roslaunch kinova_gazebo robot_launch.launch

# got:
Error [Param.cc:452] Unable to set value [-nan -nan -nan] for key[size]
gzclient: /build/ogre-1.9-mqY1wq/ogre-1.9-1.9.0+dfsg1/OgreMain/src/OgreNode.cpp:630: virtual void Ogre::Node::setScale(const Ogre::Vector3&): Assertion `!inScale.isNaN() && "Invalid vector supplied as parameter"' failed.
Aborted (core dumped)
[gazebo_gui-3] process has died [pid 6104, exit code 134, cmd /opt/ros/kinetic/lib/gazebo_ros/gzclient __name:=gazebo_gui __log:=/home/tor/.ros/log/2fb0b64c-ad54-11e7-a213-34689501c30b/gazebo_gui-3.log].
log file: /home/tor/.ros/log/2fb0b64c-ad54-11e7-a213-34689501c30b/gazebo_gui-3*.log
^C[gazebo-2] killing on exit
[gazebo-2] escalating to SIGTERM

# diagnose
It seems that something is wrong in the xacro/urdf,
in this case, the default: j2n6s300_standalone.xacro

# env
ROS Kinetic, Xenial
checking out from master
last commit 1bb9c190110b07d548234a77f38b0ffd87b6d3a4

tttor, 2017-10-10:
As suggested in #102 and [here](https://github.com/ros-simulation/gazebo_ros_pkgs/issues/488)
Have modified collision geometry in kinova_common.xacro and j2n6s300_standalone.xacro
```
    <collision>
      <origin xyz="0 0 0" rpy="0 0 0" />
      <geometry>
        <box size = "0.1 0.1 0.1"/>
      </geometry>
    </collision
```
BUT still does not work

tttor, 2017-10-10:
Solved by modifying these 2 files:
1) kinova_common.xacro
`    <xacro:macro name="kinova_virtual_link" params="link_name">
        <link name="${link_name}">
            <visual>
                <geometry>
                    <box size = "0.00001 0.00001 0.00001"/>
                </geometry>
            </visual>
            <collision>
                <geometry>
                    <box size = "0.00001 0.00001 0.00001"/>
                </geometry>
            </collision>
        </link>
    </xacro:macro>`

2) j2n6s300_standalone.xacro
`  <link name="root">
    <visual>
      <origin xyz="0 0 0" rpy="0 0 0" />
      <geometry>
        <box size = "0.00001 0.00001 0.00001"/>
      </geometry>
    
    </visual>

    <collision>
      <origin xyz="0 0 0" rpy="0 0 0" />
      <geometry>
        <box size = "0.00001 0.00001 0.00001"/>
      </geometry>
    </collision>
  </link>`

martine1406, 2017-10-23:
thanks for your update. very useful for our ROS community. Do you still have an issue?

tttor, 2017-10-23:
Hi,
No, this is not an issue anymore for me as the above solution works.
But, does the official kinova-ros team think so? or they have another solution?
The related files still have 0s in geometry tag, as in https://github.com/Kinovarobotics/kinova-ros/blob/master/kinova_description/urdf/kinova_common.xacro
Thank you.