Kinovarobotics/kinova-ros #110: Moveit support for set_joint_value_target

sjhansen3, 2018-01-15:
I am looking to use moveit to plan in joint space for the Jaco j2s7s300. Is the set_joint_value_target() function in python supported for this arm?

Here is a simplification of code I am trying to execute:
`
#!/usr/bin/env python

import sys
import copy
import rospy
import moveit_commander
import moveit_msgs.msg
import geometry_msgs.msg

moveit_commander.roscpp_initialize(sys.argv)
rospy.init_node('move_group_python_interface',
                anonymous=True)

robot = moveit_commander.RobotCommander()
scene = moveit_commander.PlanningSceneInterface()
group = moveit_commander.MoveGroupCommander("arm")
display_trajectory_publisher = rospy.Publisher(
                                    '/move_group/display_planned_path',
                                    moveit_msgs.msg.DisplayTrajectory)

current_config = [4.71, 2.84, 0.0, 0.75, 4.62, 4.48, 4.88, 0.0, 0.0, 0.0]
print("current config {}".format(current_config))

group.set_joint_value_target(current_config)
print("planning2")

plan2 = group.plan()
print("displaying2")
rospy.sleep(5)

rospy.spin()
`

I get this error:
`Traceback (most recent call last):
  moveit_interface.py", line 41, in <module>
    group.set_joint_value_target(current_config)
  moveit_commander/move_group.py", line 216, in set_joint_value_target
    raise MoveItCommanderException("Error setting joint target. Is the target within bounds?")
moveit_commander.exception.MoveItCommanderException: Error setting joint target. Is the target within bounds?`

This config is valid - I am able to plan to the same config if I plan using a pose target i.e.
`group.set_pose_target(pose_target)`

martine1406, 2018-01-26:
Hi sjhansen3

Sorry for the delay in answering you. I can't see why set_joint_value_target() would not be supported. You are sending 10 values (7 for the joints, and three for the fingers I assume). But I am not sure it is the right way to do it. Have you tried the way it is done on the page in the section Planning to a joint-space goal (http://docs.ros.org/indigo/api/pr2_moveit_tutorials/html/planning/scripts/doc/move_group_python_interface_tutorial.html)?

You can see the documentation for set_joint_value_target on this page: http://docs.ros.org/jade/api/moveit_commander/html/classmoveit__commander_1_1move__group_1_1MoveGroupCommander.html#a55db2d061bbf73d05b9a06df7f31ea39

It is said that arg1 from set_joint_value_target could be dict (in you case it is not), list (maybe it is), JointState (it is not in your case), string (but you would need an arg2 then), or Pose or PoseStamped (it is not in your case). It is said: the list should specify all the variable values
for the group. The group here is arm and am has 7 joints... not sure you can specify 10 entries. 

In any case, I am not 100% what is the error here, but I am pretty confident you can use set_joint_value_target to set joint configuration targets. I hope the resources I suggested can help you figure out what is the issue here. 

Good luck!
Martine

martine1406, 2018-03-15:
HI @sjhansen3 

Have you found the solution to this issue?

sjhansen3, 2018-03-15:
I have not worked on it - I ended up just sending pose commands. If you like you can close the issue and I will re open if it becomes relevant again.

valmik, 2018-04-07:
What's happening is that the joint bounds are failing when the set_joint_value_target is called, even though they shouldn't be failing. In the C++ implementation, it sets it anyways and doesn't throw an exception, so it works fine even though bound checking fails. To make it work in Python, just catch the exception and it should work fine.

My suspicion is that there's a problem translating the urdf into the robot_model that moveit uses. I'll play around it a bit over the next couple weeks and see if I find the problem.

martine1406, 2018-04-09:
Hi @sjhansen3 and @valmik 

I'll keep the issue open for a few more weeks, to leave @valmik the time to investigate the issue. Thank you both for your contribution. Then, if we cannot find a solution, I'll close the issue for now. 
Thanks again
Martine

sjhansen3, 2018-04-09:
I dont think this issue should be closed. I believe what @valmik is saying is that there is a bug in the code. Python is throwing errors when it shouldn't. Perhaps this is an issue with the URDF or some other setting in moveit. @martine1406 Could you look into this on your end? Are you able to plan to joint configurations?

martine1406, 2018-04-10:
Hi @sjhansen3 

We are able to send joint targets as you can see in the [pick and place example ](https://github.com/Kinovarobotics/kinova-ros/blob/master/kinova_moveit/kinova_arm_moveit_demo/src/pick_place.cpp). For example, please see [setJointValueTarget](https://github.com/Kinovarobotics/kinova-ros/blob/master/kinova_moveit/kinova_arm_moveit_demo/src/pick_place.cpp#L724 ).

@valmik says:  joint bounds are failing when the set_joint_value_target is called, **even though they shouldn't be failing. ** In the C++ implementation, **it sets it anyways and doesn't throw an exception**, so it works fine even though bound checking fails. To make it work in Python, just catch the exception and it should work fine.

Can you try catching the exception for now if you want to continue working with python? If the problem is indeed in the way the urdf is translated into the robot_model that moveit uses, this could involve a bug in moveit as well. see https://github.com/ros-planning/moveit_tutorials/issues/62

Thanks
Martine

closed 2018-06-04 by martine1406