Kinovarobotics/kinova-ros #113: Cartesian control fails under certain conditions

ikostrikov, 2018-01-26:
Hi!

I have a problem with j2n6s300. Under certain conditions cartesian control stops working.

Specifically, if the end effector fails to reach a certain location and gripper commands are called afterwards, later calls of functions for cartesian control do not work.

I attached a small code snippet to reproduce the problem.

```python
import roslib
roslib.load_manifest('kinova_demo')
import rospy

from robot_control_modules import *

prefix = 'j2n6s300_'
nbJoints = 6
interactive = True
duration_sec = 5

if __name__ == '__main__':
    rospy.init_node('test_action_servers')

    position = [1, -1, 0.5]
    quaternion = tf.transformations.quaternion_from_euler(
        90 * 3.1415 / 180, 0, 0, 'rxyz')
    result = cartesian_pose_client(position, quaternion, prefix)

    # Works when the line below is commented
    result = gripper_client([6800, 6800, 6800], prefix)

    #set cartesian pose 2
    position = [0.0, -0.5, 0.2]
    quaternion = tf.transformations.quaternion_from_euler(
        180 * 3.1415 / 180, 0, 0, 'rxyz')
    result = cartesian_pose_client(position, quaternion, prefix)
```

martine1406, 2018-02-02:
Hi ikostrikov 

OK.... well your first point in unreachable right (outside of the robot range)? What happens if you put a first point that is reachable by the robot? How do you expect the robot to react exactly?

I'll conduct a few tests later on with your code snippet. I'll keep you posted if I have more insight

Thanks
Martine

martine1406, 2018-02-02:
Hi ikostrikov

I just tested on my computer and your example code runs fine on my machine. The Jaco2 arm executes both goals (although it is never actually reach the first goal). I am sorry, but I cannot reproduce your issue on my side :-(

What I suggest is you check what is going on in the terminal where you launched kinova_bringup kinova_robot.launch. Any particular error? 

Here is the output from my terminal:

(before the first target pose was sent, then accepted, then I get all the feedback form the action server)
593; Y -0.599395; Z 0.287938; ThetaX 2.72944; ThetaY 0.0337076; ThetaZ -0.0795684
[DEBUG] [1517590597.851930811]: void kinova::KinovaPoseActionServer::actionCallback(const ArmPoseGoalConstPtr&): current_pose X 0.41287; Y -0.598626; Z 0.288614; ThetaX 2.7176; ThetaY 0.0380505; ThetaZ -0.0878567
[DEBUG] [1517590597.965550319]: void kinova::KinovaPoseActionServer::actionCallback(const ArmPoseGoalConstPtr&): current_pose X 0.417971; Y -0.597833; Z 0.289124; ThetaX 2.7065; ThetaY 0.0424142; ThetaZ -0.0962577
[DEBUG] [1517590598.074507458]: void kinova::KinovaPoseActionServer::actionCallback(const ArmPoseGoalConstPtr&): current_pose X 0.423113; Y -0.59717; Z 0.289413; ThetaX 2.69516; ThetaY 0.0471381; ThetaZ -0.104241
[DEBUG] [1517590598.182502811]: void kinova::KinovaPoseActionServer::actionCallback(const ArmPoseGoalConstPtr&): current_pose X 0.428162; Y -0.596496; Z 0.289919; ThetaX 2.68406; ThetaY 0.0521607; ThetaZ -0.111892
[DEBUG] [1517590598.290526966]: void kinova::KinovaPoseActionServer::actionCallback(const ArmPoseGoalConstPtr&): current_pose X 0.432893; Y -0.595987; Z 0.29038; ThetaX 2.67295; ThetaY 0.0568156; ThetaZ -0.119807
[DEBUG] [1517590598.398523505]: void kinova::KinovaPoseActionServer::actionCallback(const ArmPoseGoalConstPtr&): current_pose X 0.437226; Y -0.595523; Z 0.290795; ThetaX 2.66272; ThetaY 0.0615767; ThetaZ -0.127096
[DEBUG] [1517590598.507494750]: void kinova::KinovaPoseActionServer::actionCallback(const ArmPoseGoalConstPtr&): current_pose X 0.441793; Y -0.595155; Z 0.291127; ThetaX 2.65137; ThetaY 0.0664284; ThetaZ -0.135184
[DEBUG] [1517590598.616794202]: void kinova::KinovaPoseActionServer::actionCallback(const ArmPoseGoalConstPtr&): current_pose X 0.445867; Y -0.594712; Z 0.291378; ThetaX 2.64206; ThetaY 0.0711197; ThetaZ -0.14215
[DEBUG] [1517590598.725524383]: void kinova::KinovaPoseActionServer::actionCallback(const ArmPoseGoalConstPtr&): current_pose X 0.450331; Y -0.594392; Z 0.292192; ThetaX 2.63057; ThetaY 0.0768321; ThetaZ -0.149172
[DEBUG] [1517590598.834584254]: void kinova::KinovaPoseActionServer::actionCallback(const ArmPoseGoalConstPtr&): current_pose X 0.454323; Y -0.594227; Z 0.292615; ThetaX 2.61986; ThetaY 0.0815488; ThetaZ -0.155758
[DEBUG] [1517590598.943899104]: void kinova::KinovaPoseActionServer::actionCallback(const ArmPoseGoalConstPtr&): current_pose X 0.458453; Y -0.594005; Z 0.293009; ThetaX 2.60931; ThetaY 0.0871699; ThetaZ -0.162831
[DEBUG] [1517590599.053437478]: void kinova::KinovaPoseActionServer::actionCallback(const ArmPoseGoalConstPtr&): current_pose X 0.462427; Y -0.593829; Z 0.293659; ThetaX 2.59833; ThetaY 0.0925052; ThetaZ -0.170051
[DEBUG] [1517590599.162477524]: void kinova::KinovaPoseActionServer::actionCallback(const ArmPoseGoalConstPtr&): current_pose X 0.466378; Y -0.593637; Z 0.293855; ThetaX 2.58857; ThetaY 0.0982466; ThetaZ -0.176941
[DEBUG] [1517590599.270476998]: void kinova::KinovaPoseActionServer::actionCallback(const ArmPoseGoalConstPtr&): current_pose X 0.469676; Y -0.593802; Z 0.294374; ThetaX 2.57803; ThetaY 0.102525; ThetaZ -0.183474
[DEBUG] [1517590599.378515931]: void kinova::KinovaPoseActionServer::actionCallback(const ArmPoseGoalConstPtr&): current_pose X 0.473189; Y -0.593806; Z 0.294645; ThetaX 2.56866; ThetaY 0.108304; ThetaZ -0.188749
[DEBUG] [1517590599.486562047]: void kinova::KinovaPoseActionServer::actionCallback(const ArmPoseGoalConstPtr&): current_pose X 0.476331; Y -0.593887; Z 0.295135; ThetaX 2.55922; ThetaY 0.112909; ThetaZ -0.193969
[DEBUG] [1517590599.595495440]: void kinova::KinovaPoseActionServer::actionCallback(const ArmPoseGoalConstPtr&): current_pose X 0.479991; Y -0.593975; Z 0.296131; ThetaX 2.54813; ThetaY 0.119579; ThetaZ -0.198904
[DEBUG] [1517590599.704821625]: void kinova::KinovaPoseActionServer::actionCallback(const ArmPoseGoalConstPtr&): current_pose X 0.483259; Y -0.594128; Z 0.296578; ThetaX 2.53817; ThetaY 0.124696; ThetaZ -0.205239
[DEBUG] [1517590599.813798638]: void kinova::KinovaPoseActionServer::actionCallback(const ArmPoseGoalConstPtr&): current_pose X 0.486555; Y -0.594294; Z 0.297219; ThetaX 2.52793; ThetaY 0.13031; ThetaZ -0.211054
[DEBUG] [1517590599.922555897]: void kinova::KinovaPoseActionServer::actionCallback(const ArmPoseGoalConstPtr&): current_pose X 0.489907; Y -0.594794; Z 0.298332; ThetaX 2.51576; ThetaY 0.137166; ThetaZ -0.214703
[DEBUG] [1517590600.031855641]: void kinova::KinovaPoseActionServer::actionCallback(const ArmPoseGoalConstPtr&): current_pose X 0.492738; Y -0.595039; Z 0.298951; ThetaX 2.50628; ThetaY 0.141954; ThetaZ -0.220091
[DEBUG] [1517590600.140794777]: void kinova::KinovaPoseActionServer::actionCallback(const ArmPoseGoalConstPtr&): current_pose X 0.495354; Y -0.595239; Z 0.29871; ThetaX 2.49892; ThetaY 0.14623; ThetaZ -0.225089
[DEBUG] [1517590600.252563195]: void kinova::KinovaPoseActionServer::actionCallback(const ArmPoseGoalConstPtr&): current_pose X 0.498232; Y -0.595901; Z 0.299759; ThetaX 2.48742; ThetaY 0.152288; Thet [... cut ...]
[DEBUG] [1517590626.112611592]: void kinova::KinovaPoseActionServer::actionCallback(const ArmPoseGoalConstPtr&): arm_comm_.isCloseToOther
[DEBUG] [1517590626.112687731]: Setting the current goal as succeeded
[DEBUG] [1517590626.112770635]: Setting status to succeeded on goal, id: /test_action_servers-3-1517590613.936, stamp: 1517590613.94
[DEBUG] [1517590626.112856745]: Publishing result for goal with id: /test_action_servers-3-1517590613.936 and stamp: 1517590613.94
**[ WARN] [1517590626.113028555]: void kinova::KinovaPoseActionServer::actionCallback(const ArmPoseGoalConstPtr&): LINE 175, setSucceeded**

ikostrikov, 2018-02-05:
Hi!

I expect the arm to move independently of gripper commands. I attached a code snippet where it first moves to a reachable position. 

By the way, what is the recommended way to find unreachable positions?

I updated the code snippet to reproduce the problem. Apparently it also depends on the initial position.

```python
import roslib
roslib.load_manifest('kinova_demo')
import rospy

from robot_control_modules import *

prefix = 'j2n6s300_'
nbJoints = 6
interactive = True
duration_sec = 5

if __name__ == '__main__':
    rospy.init_node('test_action_servers')
    quaternion = tf.transformations.quaternion_from_euler(
        3.1415, 0, 0, 'rxyz')

    position = [0.0, -0.5, 0.2]
    result = cartesian_pose_client(position, quaternion, prefix)
    result = gripper_client([0, 0, 0], prefix)

    position = [0.0, -0.5, 0.01]
    result = cartesian_pose_client(position, quaternion, prefix)

    # Works when the line below is commented
    result = gripper_client([6800, 6800, 6800], prefix)

    position = [0.0, -0.5, 0.2]
    # Arm doesn't execute the line below
    result = cartesian_pose_client(position, quaternion, prefix)

    result = gripper_client([0, 0, 0], prefix)
```

```
[ INFO] [1517865186.962918540]: kinova_robotType is j2n6s300.
[ INFO] [1517865186.973153120]: Initializing Kinova USB API (header version: 50300, library version: 5.2.0)
[ INFO] [1517865187.132320921]: Found 1 device(s), using device at index 0 (model: Jaco V2 Service, serial number: PJ00900006027686-0 , code version: 393481, code revision: 6)
[ INFO] [1517865187.204289192]: Initializing fingers...this will take a few seconds and the fingers should open completely
[ INFO] [1517865187.351086807]: The arm is ready to use.
j2n6s300_joint_1 j2n6s300_joint_2 j2n6s300_joint_3 j2n6s300_joint_4 j2n6s300_joint_5 j2n6s300_joint_6
[DEBUG] [1517865221.571504360]: The action server has received a new goal request
[DEBUG] [1517865221.571678186]: A new goal has been recieved by the single goal action server
[DEBUG] [1517865221.571897511]: Accepting a new goal
[DEBUG] [1517865221.571977884]: Accepting goal, id: /test_action_servers-1-1517865221.552, stamp: 1517865221.55
[DEBUG] [1517865221.574194178]:

***-----------------------***
void kinova::KinovaPoseActionServer::actionCallback(const ArmPoseGoalConstPtr&):  target X 0; Y -0.5; Z 0.2; ThetaX 3.1415; ThetaY 0; ThetaZ 0
***-----------------------***

[DEBUG] [1517865221.594160247]: void kinova::KinovaPoseActionServer::actionCallback(const ArmPoseGoalConstPtr&): current_pose X 3.7279e-05; Y -0.496648; Z 0.193207; ThetaX -3.12372; ThetaY -0.00514646; ThetaZ -0.0089985
[DEBUG] [1517865221.594196995]: void kinova::KinovaPoseActionServer::actionCallback(const ArmPoseGoalConstPtr&): arm_comm_.isCloseToOther
[DEBUG] [1517865221.594227402]: Setting the current goal as succeeded
[DEBUG] [1517865221.594281570]: Setting status to succeeded on goal, id: /test_action_servers-1-1517865221.552, stamp: 1517865221.55
[DEBUG] [1517865221.594324510]: Publishing result for goal with id: /test_action_servers-1-1517865221.552 and stamp: 1517865221.55
[ WARN] [1517865221.594524923]: void kinova::KinovaPoseActionServer::actionCallback(const ArmPoseGoalConstPtr&): LINE 175, setSucceeded
[DEBUG] [1517865221.903395443]: The action server has received a new goal request
[DEBUG] [1517865221.903575227]: A new goal has been recieved by the single goal action server
[DEBUG] [1517865221.904496060]: Accepting a new goal
[DEBUG] [1517865221.904575227]: Accepting goal, id: /test_action_servers-2-1517865221.902, stamp: 1517865221.90
[DEBUG] [1517865221.916187566]: Setting the current goal as succeeded
[... 48 lines cut ...]
***-----------------------***

[DEBUG] [1517865225.733188359]: void kinova::KinovaPoseActionServer::actionCallback(const ArmPoseGoalConstPtr&): current_pose X -0.00126967; Y -0.494569; Z 0.00761275; ThetaX -3.1255; ThetaY -0.000129294; ThetaZ 0.00758842
[DEBUG] [1517865225.841182184]: void kinova::KinovaPoseActionServer::actionCallback(const ArmPoseGoalConstPtr&): current_pose X -0.00127012; Y -0.49457; Z 0.00761239; ThetaX -3.1255; ThetaY -0.000131644; ThetaZ 0.00758187
[DEBUG] [1517865225.949190765]: void kinova::KinovaPoseActionServer::actionCallback(const ArmPoseGoalConstPtr&): current_pose X -0.0012702; Y -0.494571; Z 0.00761181; ThetaX -3.12551; ThetaY -0.000132285; ThetaZ 0.00757863
[DEBUG] [1517865226.057195716]: void kinova::KinovaPoseActionServer::actionCallback(const ArmPoseGoalConstPtr&): current_pose X -0.00126949; Y -0.494571; Z 0.00761119; ThetaX -3.12551; ThetaY -0.000133578; ThetaZ 0.00757177
[DEBUG] [1517865226.167195120]: void kinova::KinovaPoseActionServer::actionCallback(const ArmPoseGoalConstPtr&): current_pose X -0.00126978; Y -0.494571; Z 0.00761057; ThetaX -3.12551; ThetaY -0.00013495; ThetaZ 0.00756973
[DEBUG] [1517865226.280196412]: void kinova::KinovaPoseActionServer::actionCallback(const ArmPoseGoalConstPtr&): current_pose X -0.00126986; Y -0.494572; Z 0.00761091; ThetaX -3.12551; ThetaY -0.000135568; ThetaZ 0.00756661
[DEBUG] [1517865226.388188044]: void kinova::KinovaPoseActionServer::actionCallback(const ArmPoseGoalConstPtr&): current_pose X -0.00127017; Y -0.494572; Z 0.00761005; ThetaX -3.12551; ThetaY -0.000136995; ThetaZ 0.00756452
[DEBUG] [1517865226.496175488]: void kinova::KinovaPoseActionServer::actionCallback(const ArmPoseGoalConstPtr&): current_pose X -0.00127008; Y -0.494573; Z 0.00761025; ThetaX -3.12551; ThetaY -0.000136819; ThetaZ 0.00756169
[... cut ...]
[DEBUG] [1517865226.912778757]: The action server has received a new goal request
[DEBUG] [1517865226.912842810]: A new goal has been recieved by the single goal action server
[DEBUG] [1517865226.912976803]: Accepting a new goal
[DEBUG] [1517865226.913047049]: Accepting goal, id: /test_action_servers-6-1517865226.912, stamp: 1517865226.91
[DEBUG] [1517865228.353168508]: Setting the current goal as succeeded
[DEBUG] [1517865228.353219391]: Setting status to succeeded on goal, id: /test_action_servers-6-1517865226.912, stamp: 1517865226.91
```

martine1406, 2018-02-06:
Hi ikostrikov 

I get your problem: a command that is unreachable by the gripper's action server will impact the arm's action server. The reason for this apparent connection between both action servers is that the finger action sets the arm to angular mode. Your second pose goal ([0.0, -0.5, 0.01]) brings the 5th joint near a singularity position. When you switch to angular mode to execute the finger command, then you are not able to switch back the arm to Cartesian mode. 

As I agree with you that this is a weird behaviour. I brought a small fix in the kinova_comm setFingerPositions function to detect the current control mode (Cartesian or angular), and then send a finger command in this mode. Now your example is working.

Thanks for sharing your feedback
Let me know if you experience other problems

Martine

closed 2018-02-06 by martine1406