Kinovarobotics/kinova-ros #115: Low accuracy of pose execution

sjhansen3, 2018-02-07:
I am experiencing positional accuracy of +- 0.015m on average for executed poses using move it. This seems like an unacceptable amount of accuracy for the Jaco. I am using the J2s7s300.

Here is my test:
I use move it to plan to a pose. I compare the FK of the final pose in the plan to the FK of the encoder readings from the robot. The difference between these two positions is 0.015m after 10 tests moving from A->B and back to A.

Here are the launch files that I use in case others would like to try and replicate the issue:
1. `roslaunch kinova_bringup kinova_robot.launch kinova_robotType:=j2s7s300 /joint_states:=/j2s7s300_driver/out/joint_state`
2. `roslaunch j2s7s300_moveit_config j2s7s300_demo.launch /joint_states:=/j2s7s300_driver/out/joint_state`
3. I then run a script which plans and executes using moveit pose commands.

I was told by Kinova to expect ~5mm accuracy and ~1mm precision - So I'm guessing I must have some tuning or configurations to adjust perhaps?

Where would you suggest looking next to adjust tolerances or tuning. Is this a normal amount of error to expect for the Jaco? How much error do others experience?

sjhansen3, 2018-02-07:
Ideally Im looking for a place downstream of the moveit execution. Perhaps a kinova trajectory followingtolerance I can adjust to improve the accuracy.

sjhansen3, 2018-02-07:
Here is a link to a discussion about the UR driver which is what made me think this is probably a kinova driver issue. 

https://groups.google.com/forum/#!topic/moveit-users/w5_Y8Q46kK8

sjhansen3, 2018-02-08:
I thought I would add a little more information. 

1. Here is the out from `rostopic echo /j2s7s300/follow_joint_trajectory/goal` I dont understand why path tolerance and goal tolerance are set to empty lists and 0. Perhaps this is a bug? Perhaps I am not setting the variables correctly?

$ rostopic echo /j2s7s300/follow_joint_trajectory/goal
header: 
  seq: 29
  stamp: 
    secs: 1518045211
    nsecs: 957230539
  frame_id: ''
goal_id: 
  stamp: 
    secs: 1518045211
    nsecs: 957231244
  id: /move_group-30-1518045211.957231244
goal: 
  trajectory: 
    header: 
      seq: 0
      stamp: 
        secs: 0
        nsecs:         0
      frame_id: /world
    joint_names: ['j2s7s300_joint_1', 'j2s7s300_joint_2', 'j2s7s300_joint_3', 'j2s7s300_joint_4', 'j2s7s300_joint_5', 'j2s7s300_joint_6', 'j2s7s300_joint_7']
    points: 
      - 
        positions: [-2.357549506554978, 2.6375641172619413, 2.5487367759473294, 1.2006024210459416, -1.9696260053229828, 4.052361788463812, -0.6103549765331557]
        velocities: [0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0]
        accelerations: [0.0909106145673609, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0]
        effort: []
        time_from_start: 
          secs: 0
          nsecs:         0
      - 
        positions: [-2.342220072296859, 2.7031670890331765, 2.5504550285488325, 1.2506990996827951, -1.866634311119342, 3.8841079252180464, -0.572741276598979]
        velocities: [0.02914259176716774, 0.1247169721237104, 0.0032665480849017685, 0.0952381562047237, 0.19579619502239176, -0.319865271428598, 0.07150692475515621]
        accelerations: [0.01034638301768056, 0.0442777901398185, 0.0011597100869450239, 0.033812038746044576, 0.06951277509188798, -0.11356054529036695, 0.02538683030816484]
        effort: []
        time_from_start: 
          secs: 0
          nsecs: 580725323
      - 
        positions: [-2.3268906380387393, 2.7687700608044112, 2.552173281150336, 1.3007957783196489, -1.7636426169157011, 3.715854061972281, -0.5351275766648022]
        velocities: [0.031888135504530915, 0.1364666443729345, 0.003574291840465765, 0.10421060880657101, 0.21424229004846612, -0.35, 0.0782436416198794]
        accelerations: [-1.9197509346639657e-15, 1.9053167171101013e-15, -1.921555211858199e-15, -9.526583585550507e-16, -9.815267936627796e-16, 0.0, -4.907633968313898e-16]
        effort: []
        time_from_start: 
          secs: 1
          nsecs:  61450647
      - 
        positions: [-2.31156120378062, 2.8343730325756464, 2.5538915337518393, 1.3508924569565024, -1.6606509227120605, 3.5476001987265153, -0.4975138767306254]
        velocities: [0.031888135504530415, 0.1364666443729348, 0.0035742918404657606, 0.10421060880657065, 0.21424229004846584, -0.35, 0.07824364161987929]
        accelerations: [-1.5877639309250825e-16, -6.928424425854905e-16, 1.9026103013187493e-15, -5.485002670468466e-16, -1.7321061064637263e-16, 0.0, 8.660530532318632e-17]
        effort: []
        time_from_start: 
          secs: 1
          nsecs: 542175970
      - 
        positions: [-2.296231769522501, 2.8999760043468816, 2.555609786353343, 1.400989135593356, -1.5576592285084196, 3.3793463354807494, -0.45990017679644857]
        velocities: [0.03188813550453088, 0.13646664437293432, 0.0035742918404657606, 0.10421060880657089, 0.2142422900484661, -0.35, 0.07824364161987929]
        accelerations: [2.0785273277564717e-15, -1.2124742745246085e-15, -1.9026103013187493e-15, 1.5011586256018961e-15, 1.1547374043091509e-15, 0.0, -8.660530532318632e-17]
        effort: []
        time_from_start: 
          secs: 2
          nsecs:  22901294
      - 
        positions: [-2.2809023352643814, 2.9655789761181164, 2.557328038954846, 1.4510858142302097, -1.4546675343047788, 3.211092472234984, -0.42228647686227183]
        velocities: [0.031888135504530915, 0.1364666443729345, 0.003574291840465303, 0.10421060880657101, 0.21424229004846637, -0.35, 0.07824364161987934]
        accelerations: [-1.9197509346639657e-15, 1.9053167171101013e-15, 0.0, -9.526583585550507e-16, 0.0, 0.0, 2.598159159695593e-16]
[... 88 lines cut ...]
        effort: []
        time_from_start: 
          secs: 7
          nsecs: 791605176
      - 
        positions: [-2.0969491241669496, 3.7528146373729374, 2.577947070172886, 2.0522459578724526, -0.21876720386108905, 1.1920461132857962, 0.02907792234784956]
        velocities: [0.02914259176716769, 0.12471697212371025, 0.003266548084901762, 0.09523815620472324, 0.1957961950223914, -0.319865271428598, 0.07150692475515606]
        accelerations: [-0.010346383017680455, -0.04427779013982119, -0.001159710086945011, -0.03381203874604171, -0.06951277509188718, 0.11356054529036663, -0.025386830308163747]
        effort: []
        time_from_start: 
          secs: 8
          nsecs: 272330500
      - 
        positions: [-2.0816196899088304, 3.818417609144172, 2.5796653227743893, 2.1023426365093063, -0.1157755096574482, 1.0237922500400305, 0.06669162228202641]
        velocities: [0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0]
        accelerations: [-0.09091061456736069, -0.3890558764756342, -0.010190030326910706, -0.29709640721070757, -0.6107882429142861, 0.9978230020396045, -0.22306658677616864]
        effort: []
        time_from_start: 
          secs: 8
          nsecs: 853055824
  path_tolerance: []
  goal_tolerance: []
  goal_time_tolerance: 
    secs: 0
    nsecs:         0

martine1406, 2018-02-08:
Hi sjhansen3

Absolute positional accuracy of the Jaco arm (including the 7 dof arm) can be of the centimeter level. I am not too sure how you define position (you expect 1 mm?) and accuracy (you expect 5 mm?). Internal tests done here at Kinova showed an average absolute precision in Cartesian mode of about 1.5 cm and a repeatability of about 4-5 mm.

That said, when using MoveIt, there is another aspect that might contribute to the imprecision you witness with the robot.  If you look carefully, you can see that kinova_driver/src/kinova_joint_trajectory_controller.cpp sends velocity commands to the robot (instead of position commands), because we do not want the robot to stop at every point like it does when you send an angular position trajectory.

The problem is that MoveIt sends a bunch of trajectory points with velocity, acceleration and some form of time stamp (secs+nsecs), but we just take the list of velocity commands and send them to the robot, that will treat these commands at 100Hz. But the trajectory coming from MoveIt is not necessarily sampled at 100 Hz (from your echoed readings, it indeed appears not to be sampled at that rate). Hence, the robot will move in the right direction, but not necessarily to the exact positions commanded by Moveit.

If you are ready to have the robot stop at every angular position point, then you can send position points instead of velocity points and the robot's position should be much more accurate. For a trajectory planned in MoveIt, this will probably be ugly though (like let<s say moveIt sends 20 points for a 20 cm trajectory, the robot would stop much too often).

An alternative could be to pre-process in some way the list of trajectory points sent by MoveIt and resample it for 10 ms time increments, and then send appropriate velocity commands to the robot, and then maybe add a position commands with the end goal as an ultimate command to be sure the robot is as close as possible to the angular configuration target provided.  

Finally, a third possibility is to work with the positions outputted with MoveIt, but design some kind of PID loop or closed-loop control that would receive the commands from MoveIt as well as the joint position feedback from the robot, and would output appropriate velocity commands. This is actually the strategy that wads used in the package kinova-movo (https://github.com/Kinovarobotics/kinova-movo). 

path_tolerance and goal_tolerance are not set by default by Moveit, and we have not set them explicitly, therefore they are empty. 

j2s7s300_control.yaml defines the controller structure for the Gazebo model. It is involved with Gazebo, not MoveIt. You need to define a controller for the virtual robot in Gazebo as Gazebo replicates the real world (if you have a robot in the real world with no controller, it would just no move or collapse. Hnece you robot model needs controller). The real robot has its own controllers. j2s7s300_demo.launch launches MoveIt nodes needed to plan and insure that the plan can be sent to the real robot. Gazebo is not involved here, so this config file is not necessary. 

I hope this long e-mail has at least provided some answers to your questions, and explained a bit the robot's behavior. I will certainly continue to push for that issue to be solved as fast as possible. I'll let you know if we have new software development that can fix that issue. 

Thanks for your understanding
Martine

sjhansen3, 2018-02-08:
Hi Martine,

Thank you for your explanation. I wanted to clarify my understanding. 

Please respond if my understanding is correct :)

1. When I execute a trajectory using moveit the only data that is used to follow the trajectory is the velocity data. No positional data is used in the loop at all.

2. The velocity commands are fed into the robot at a rate of 100Hz (every 10ms) regardless of the timing data from moveit.

Given this data I am surprised we achieve even 1-4 cm of accuracy on the robot. It looks like my moveit plan has a waypoint ~500ms, if these waypoints are then sent as velocity only commands at 10ms intervals we are off by a factor of 50 for velocity commands. 

Looking forward to hearing your clarification. 

-Steve

martine1406, 2018-02-09:
Hello Steve

Okay I digged in a little more today and I have some more answers. It turns out I was not completely right yesterday. The key is in file kinova_joint_trajectory_controller.cpp, more specifically in function commandCB which includes a timer I had not seen (time_pub_joint_vel_). I am not the developer of this code, so sometimes I miss some information. Sorry about that. 

That said, this timer starts when the JointTrajectoryController receives the trajectory on topic /"+ robot_name + "_driver/trajectory_controller/command". Then, you will see in function pub_joint_vel that is called every 10 ms that there is this condition

```
if( (ros::Time::now() - time_pub_joint_vel_) >= traj_command_points_[traj_command_points_index_].time_from_start)
        {
            ROS_INFO_STREAM("Moved to point " << traj_command_points_index_++);
        }
```
Actually, we check the current time and we publish the same velocity command until the current time is bigger than the corresponding point's time_from_start. I must admit it makes way more sense than my initial explanation where each velocity vector was sent only once! So what it means is that you will send the velocity vector from point 1 until the current time elapsed from the moment the time_pub_joint_vel_ timer was started is bigger than point 1's time_from_start. Then, you will send the velocity vector from point 2. And so on. 

That explains better why the robot does go approximately to the right location, but of course, we are making an approximation that the velocity stays constant in between two points instead of making an interpolation. 

So to answer your questions more specifically:
1- correct, positional data isn't used. 
2- velocity commands are fed to the robot at 100 Hz, but considering my new explanation, some timing information from Moveit is considered. 

Sorry about the confusion here. Again, as I said, I am not the person who developed this code, so sometimes, I misunderstand some concepts. Nonetheless, I'll still check quickly if I can find a quick fix for this problem. 

Let me know if this is still unclear
Martine

sjhansen3, 2018-02-12:
Hi Martine,

I did some development over the weekend and got a trajectory controller to work. I have a few remaining questions, so I created a pull request here:
https://github.com/Kinovarobotics/kinova-ros/pull/116

If possible can you clarify my questions in the PR?

Thanks
-Steve

closed 2018-03-15 by martine1406