Kinovarobotics/kinova-ros #117: What is specific to kinova-ros to indigo? Can I upgrade to kinetic?

sjhansen3, 2018-02-14:
I am considering upgrading to kinetic to support more modern sensor drivers and I want to know if I will be able to use the kinova drivers or which ones will need modification.

-Steve

mpmendes, 2018-02-14:
Hi Steve,

I'm using j2n6s300 with Ubuntu 16.04 and Kinetic.
I don't know if it works for other robot versions, but if I'm not mistaken I only had to change the zero dimensional boxes in some files. Check this: [https://github.com/Kinovarobotics/kinova-ros/issues/103](url)

I had other minor issues with some moveit code, that I don't remember... 

Miguel

sjhansen3, 2018-02-14:
Thanks Miguel! Would you be willing to share your code? Perhaps create a PR on a Kinova branch so others can see the changes that were made. :)

-Stevr

mpmendes, 2018-02-14:
I have a pending PR, not exactly for these specific changes yet, because they would only work for the 16.04 users. Check #112, and then commit 0f6e8b3. If you see I accidently pushed all my changes, but then reverted the ones that would be of no use for the 14.04 users. I think you can get an idea :)

martine1406, 2018-02-20:
Hey guys

I do not know if this can help you, but we have a collaborator who has forked our repo and has made the required changes to make it work on Ubuntu 16.04/ROS Kinetic. Here is the link to the repo: https://github.com/CNURobotics/kinova-ros/tree/kinetic_devel

mpmendes, 2018-03-01:
If I'm not mistaken, the error code 1015 is when a device has not been found... if you check the header file KinovaAPICommLayerUbuntu.h you can see the error codes definition. I've never had that error,  so I don't know really know what is wrong. I've had the error 1010 but that is when the robot is not connected/on. 

@sjhansen3 I've used the USB connection in the past but right now I'm trying to develop a teleoperation system with this robot. That's why I've been working with the ethernet connection. I've had no issues with the USB prior to this, It's just that with that I need to be close to the robot.

martine1406, 2018-03-01:
Hi guys

In this case, it seems to be the GetGeneralInformation function that fails. Maybe the APIs are not up to date? You can find the latest version on the kinova website (supporting the 7 dof). Just a guess here. I have not tested the CNURobotics branch yet. If I have time today, I can try quickly to connect with a 7 dof at work.

sjhansen3, 2018-03-01:
I think that could be it - I'll try pulling from master on kinova-ros to incorporate those API changes.

sjhansen3, 2018-03-01:
So I was being dumb - I did not have the SDK installed. I am still having problems installing the SDK on ubuntu 16. 

Is this the correct driver? It just hangs when I run:
https://drive.google.com/file/d/1dDk3BoWXXT68u9_syI5O3J5PPg66lF7X/view

mpmendes, 2018-03-01:
I have that one installed. My firmware version is 6.1.9 I think...

sjhansen3, 2018-03-01:
@mpmendes could you zip up the installer for 6.1.9? The kinova-api at the link above is versioned 6.0.1

mpmendes, 2018-03-01:
I installed my SDK through that version though :/ . I'm almost certain my firmware version is 6.1.9, but I can only confirm that tomorrow morning when I get to the lab.

sjhansen3, 2018-03-01:
okay - perhaps @martine1406 has something to say about it too. really strange that that installer hangs. I tried a few others and those didnt hang, but since they crash, im guessing because they are for jaco1 not jaco2.

Thanks again for all your help. :)

mpmendes, 2018-03-01:
What do you mean by "hang"? You can't download?

sjhansen3, 2018-03-01:
Here is my folder:

`~/Downloads/kinova_downloads/Ubuntu/16_04/64 bits$ ls
installSDK64.sh  KinovaAPI-6.0.1-amd64.deb  Read me first  SDK_JACO2_1.4.0_install64  SDK.png`

Here is the command I run:
`~/Downloads/kinova_downloads/Ubuntu/16_04/64 bits$ ./installSDK64.sh 
KINOVA SDK INSTALLATION
This operation require root privilege.
(Reading database ... 323791 files and directories currently installed.)
Preparing to unpack KinovaAPI-6.0.1-amd64.deb ...
Unpacking kinova-api (6.0.1) over (6.0.1) ...
Setting up kinova-api (6.0.1) ...`

And that is where the installation "hangs" It does not move past setting up kinova-api. The JACO 2 SDK Installer setup shows 0% progress and the buttons are unresponsive.

mpmendes, 2018-03-01:
Hm.. I remember that I also had some problems installing through the terminal. Edit the SDK_JACO2_1.4.0_install64 and check the allow executing file as a program. Double click and try again... It should work now

sjhansen3, 2018-03-02:
alright, so the issue was sudo. That seems rather unintuitive and really should be fixed so that an error gets thrown if the command is called without super user access.

sjhansen3, 2018-03-02:
To conclude this issue here is what I did.

1. Clone [](https://github.com/CNURobotics/kinova-ros/tree/kinetic_devel) and merge with the upstream master (kinova-ros/master), handle all merge conflicts (mostly accepting upstream changes, a couple additional changes need to be made to xacro files)
2. Download the SDK and make sure to use sudo when calling the install script. Also make sure to chmod +x for all the executable files.
[](https://drive.google.com/file/d/1dDk3BoWXXT68u9_syI5O3J5PPg66lF7X/view)

sjhansen3, 2018-03-02:
Created PR https://github.com/Kinovarobotics/kinova-ros/pull/122 to add documentation and tie in accurate robot code into kinetic.

martine1406, 2018-03-02:
Hi guys

Miguel, thanks for your support here. Yes, this may be non-intuitive that you have to run the installer in sudo. I got mixed up myself when I did it the first time, but I was told by our software team that this is something common for Linux users. In any case, I agree with you that this should be documented. I'll check the pull request. thanks

patlican23, 2018-05-14:
Hi 

Btw, I recently made a PR for Ros Kinetic to work on Ubuntu 16.04.
I just noticed this comment https://github.com/Kinovarobotics/kinova-ros/issues/117#issuecomment-367131134 linking to a repo with this change.
Is there a reason why the change hasn't been merged?

sjhansen3, 2018-05-14:
I created a pull request for this a while back and it was merged into kinova-ros-beta. I think there is probably a need for a naming change, perhaps they can create a kinetic devel branch to make this more clear.

GianlucaCerilli, 2018-08-06:
Hi @martine1406 ,

I am on Ubuntu 16.04.5 and I am building the kinetic version of Kinova at branch kinetic_devel.
There are a few errors related to Eigen3 and MoveGroup that is deprecated.

Can you please update the repo? Because I am having some troubles with it.
Thank you very much!

martine1406, 2018-08-06:
Hi @GianlucaCerilli 

We do not maintain this branch. However, I can tell you that I used the kinova ros package on ROS Kinetic. What I did is pull the master branch, then switch to commit 94c25e19. I did not have any problem at compilation (catkin_make). 

Thank you
Martine

GianlucaCerilli, 2018-08-06:
HI @martine1406 ,

Thank you very much! So, you are saying that I can clone the repo from the master branch (so, no from kinetic_devel) and then ¨switch to commit 94c25e1¨? What do you mean with this last step?
And does it makes some difference if I use catkin build instead of catkin_make?

Thank you again and sorry, I am not an expert.
Gianluca

martine1406, 2018-08-06:
Hi @GianlucaCerilli

I am not an expert either unfortunately. On github, you can switch to a branch, or you can switch to a specific commit and when create yourself a branch or something. I am using catkin_make. I do not know if using catkin build makes a difference or not. I recommend you use catkin_make.

polthergeist, 2020-03-20:
THere is no more activity on this issue so I will close it but feel free to reopen it if you have any other question.

closed 2020-03-20 by polthergeist