Kinovarobotics/kinova-ros #118: Change the arm speed while position cartesian control

nehagarg, 2018-02-15:
Hello, I want to get some point clod feedback while doing the Cartesian position control. It currently takes 0.5 second to get that feedback and in that time arm has already stopped moving. Is it possible to slow the execution while doing Cartesian position control?

martine1406, 2018-02-20:
hi @nehagarg 

Yes it is possible. To do so, I would recommend you modify setCartesianPosition in kinova_comm and add the velocity limits you want. You can check the TrajectoryPoint structure in the Kinova SDK html documentation. You'll see that TrajectoryPoint has a field LimitationsActive and another field Limitations in which you can set your velocity limits in Cartesian and angular space. You need to set LimitationsActive to 1 so the limits are applied, and Limitations.speedParameter1 and Limitations.speedParameter2 to the limits you want. 

Finally, you will have to call kinova_api_.sendAdvanceTrajectory instead of kinova_api_.sendBasicTrajectory in the same setCartesianPosition in kinova_comm. 

Do not hesitate to look at the kinova SDK html documentation for more details on the kinova API. Thanks

Martine

nehagarg, 2018-02-21:
Thanks for the suggestions. Will try that.

closed 2018-02-21 by nehagarg