Kinovarobotics/kinova-ros #120: Modify speed of joint movements

kevincrawfordknight, 2018-02-26:
I'm trying to change the velocity of joint movements on m1n4s200.   I would like something like

```
rosrun kinova_demo joints_action_client.py -v -r m1n4s200 degree -- 0 0 0 20
```

to just move faster. 

I can't follow the instructions below.  It seems that /m1n4s200_driver is a subscriber (rostopic info /m1n4s200_driver/in/joint_velocity). But I don't know the "rostopic pub" and "rosrun" commands to make the arm go faster.  Thanks!

===

From the GitHub README file:

The user has access to both joint velocity and Cartesian velocity (angular velocity and linear velocity). The joint velocity control can be realized by publishing to topic /'${kinova_robotType}_driver'/in/joint_velocity. The following command can move the 4th joint of a mico robot at a rate of approximate 10 degree/second. Please be aware that the publishing rate does affect the speed of motion.

eg: rostopic pub -r 100 /m1n4s200_driver/in/joint_velocity kinova_msgs/JointVelocity "{joint1: 0.0, joint2: 0.0, joint3: 0.0, joint4: 10.0}"

nsaif-kinova, 2018-02-27:
Hi Kevin,
to change the velocity you can edit this file _kinova_bringup/launch/config/robot_parameters.yaml_ there is joint speed limit for joints 1,2,3 and joints 4,5,6 and you can set this to your desired value, the big actuators max speed is 36 deg/sec and the small ones 48 deg/sec

you can also set it sending the following command : rosparam set /m1n4s200_driver/jointSpeedLimitParameter1 (for joints 1,2,3)
and 
rosparam set /m1n4s200_driver/jointSpeedLimitParameter2 (for joints 4,5,6)

kevincrawfordknight, 2018-02-27:
Thanks very much.  Hmm, when I do this:
```
% rosparam set /m1n4s200_driver/jointSpeedLimitParameter1 30
% rosrun kinova_demo joints_action_client.py -v -r m1n4s200 degree -- -20 0 0 0
```

nsaif-kinova, 2018-02-28:
I edit the file before calling roslaunch kinova_bringup kinova_robot.launch (it worked)

To use rosaparam set, i put the joint speed limit as comment in robot_parameters.yaml
then I set the value using rosparam set before starting roslaunch kinova_bringup kinova_robot.launch (it worked)

kevincrawfordknight, 2018-03-02:
Got it.  That worked, thank you!

closed 2018-03-02 by martine1406