Kinovarobotics/kinova-ros #121: Finger positions different between Jaco v1 and v2

fredrikbaberg, 2018-03-01:
We have access to both version 1 and version 2, which now should be run as j2n6s300 according to the instructions. However, the finger positions are different. With version 2, the fingers are set to open and close positions. With the same code, but version 1, the fingers are completely closed for both positions.

martine1406, 2018-03-01:
Hi @fredrikbaberg 

What positions are you talking about? What are the specific instructions you run in ros and how does the behaviour differ from what you are expecting? Is this an issue with version 2 or version 1? I do not quite get you when you say " the fingers are completely closed for both positions." 

Please provide a little more details
Thanks
Martine

fredrikbaberg, 2018-03-01:
Sorry, here's a clarification of the issue:
If I launch kinova_bringup, and thereafter `rosrun actionlib axclient /j2n6s300_driver/fingers_action/finger_positions`, sending the same value gives different results for Jaco v1 and Jaco v2.
Sending a value of 6000 for each finger on v2 gives a small space between the finger tips (I believe 6800 is the maximum value), while for v1 the maximum value is roughly 57.

martine1406, 2018-03-01:
oh I see. Sorry, I am not too familiar with how the finger motion were programmed on the jaco v1. On Jaco2, the completely closed finger position is about 6000-7000, and 0 is completely open. The ros package is intended to be used with Jaco2. Jaco1 is not officially supported here unfortunately. 

Therefore, you can still use the ros package for Jaco1, but some adaptations are expected on the user's side. In addition, the urdf model in kinova_description for j2n6s300 will correspond to the jaco2 and will not be exact for the jaco1 robot since the wrist part angle have changed slightly between the jaco v1 and jaco v2. 

Thanks for your understanding
Martine

closed 2018-03-15 by martine1406