Kinovarobotics/kinova-ros #126: Jitter in gazebo ?

haudren, 2018-03-15:
I am trying to perform some Gazebo simulations with `j2n6s300`. Upon launching Gazebo, the robot goes to the initial position but instead of settling down, visibly jitters. Moreover, executing trajectories fails often because the first point is too far away from the current state (> 0.01).

I guess this is due to a misconfiguration of the ros_control module: I noticed that all controllers are initialized with 0 d gains and high p gains, wouldn't this be a problem ? I see that there is a d gain and even an i gain in `pid_controller.py` set to 60 / 0 / 20. What's the difference between those and the gazebo controls ? Degrees and radians ? Something else ?

martine1406, 2018-03-15:
Hi haudren

PID gains in j2n6s300_control.yaml may require some more tuning. Do not hesitate to modify the gains to fit your needs better. 
I searched for pid_controller.py but could not find it. I do not know which file you are referring to. Can you copy the url to that file?
Thanks

haudren, 2018-03-16:
Thanks for the answer: I tried quickly but so far no good results. Gazebo itself might also be to blame, I haven't tried to fiddle with its physics settings. Is any arm model known to work well within Gazebo? Which physics simulator do you use at Kinova?

I was talking about `kinova-ros/kinova_driver/src/scripts/pid_controller.py` but I realize that this one is doing velocity control.  Note that I am using kinetic and the `kinetic-ros-beta` branch, is that correct ?

martine1406, 2018-03-16:
Hello haudren

can you try with the master branch? j2n6s300 seems quite stable on my end. If it is still not working, can you share a video showing the robot jittering? you can send that to our suppot e-mail: support@kinova.ca

haudren, 2018-03-16:
Hi, I tried on the master branch, but as I am using Ubuntu 16.04, I only have kinetic and launching the gazebo sim crashes :cry: It is not awfully unstable, but the plan fails from time to time, I'll see if I can adapt the gains or gazebo.

martine1406, 2018-03-16:
Oh ok I get you. You can continue using kinova-ros-beta then. The PID gains in kinova_control/config/j2n6s300_control.yaml are the same in both branches. I imagine we use the default physics engine in Gazebo (ODE I think). I do not think we specify any instructions about the physics so I assume the default one is used. Again, if you send a video, I can compare with what I see on my computer and let you know if it is similar. 

Thanks
Martine

martine1406, 2018-03-29:
HI @haudren 

I wanted to follow-up with you on your issue. Were you able to solve it? Could you get a stable Jaco2 (j2n6s300) robot in gazebo? Thanks
Martine

haudren, 2018-03-30:
Hello,

Actually, as we have a ` j2s7s300` I focused my efforts on this robot. I did two things, not sure which one had most impact:
- Add some damping to the joints. In `kinova_common.xacro`
- Removed the `root` joint. Now I directly connect `world` to `j2s7s300_link_base` 

```diff
diff --git a/kinova_description/urdf/kinova_common.xacro b/kinova_description/urdf/kinova_common.xacro
index 3fa2d12..1ee5bd9 100644
--- a/kinova_description/urdf/kinova_common.xacro
+++ b/kinova_description/urdf/kinova_common.xacro
@@ -60,7 +60,7 @@
         <axis xyz="${joint_axis_xyz}"/>
         <limit effort="2000" velocity="1" lower="${joint_lower_limit}" upper="${joint_upper_limit}"/>
         <origin xyz="${joint_origin_xyz}" rpy="${joint_origin_rpy}"/>
-        <dynamics damping="0.0" friction="0.0"/>
+        <dynamics damping="0.1" friction="10.0"/>
      </joint>
    <xacro:unless value="${fixed}">
      <transmission name="${joint_name}_trans">
```

martine1406, 2018-03-30:
Okay that's good to know. I'll try to test it on my end as soon as I have time!