Kinovarobotics/kinova-ros #130: How to control 2 JACO2 using one ros master

makeecat, 2018-03-23:
`roslaunch kinova_bringup kinova_robot.launch kinova_robotType:=j2n6s300 kinova_robotName:=left kinova_robotSerial:=PJ00000001030703130
roslaunch kinova_bringup kinova_robot.launch kinova_robotType:=j2n6s300 kinova_robotName:=right kinova_robotSerial:=PJ00000001030703133`

I tried to use my serial number and "Could not find the specified arm". I have tried the method that leaving kinova_robotSerial as blank or as 0, and I can control one arm in this way.

But how to control two arms?

makeecat, 2018-03-25:
I solved this problem my update serial number by DevelopmentCenter. You need to modify robotName in control codes in this repository.

closed 2018-03-25 by makeecat

hasauino, 2018-11-04:
I have the same problem. Can you please elaborate?
I have the serial number of each robot, however they are not found!

makeecat, 2018-11-04:
Please flash the serial number of your robots in the DevelopmentCenter again. It is possible that these numbers are not flashed in!.

hasauino, 2018-11-05:
Yes correct. It worked after I did this.
Thanks.