Kinovarobotics/kinova-ros #132: MoveIt demo with Gazebo and j2s7s300

martine1406, 2018-04-30:
The following code runs moveit with the Gazebo model for robot j2s7s300 (7 dof robot with a spherical wrist) 

`roslaunch kinova_gazebo robot_launch.launch kinova_robotType:=j2s7s300`
then
`roslaunch j2s7s300_moveit_config j2s7s300_gazebo_demo.launch `

the planning in moveit fails every time with this warning:
[ WARN] [1525102790.308848354, 63.185000000]: Joint 'j2s7s300_joint_6' from the starting state is outside bounds by a significant margin: [ 7.4013 ] should be in the range [ 1.13446 ], [ 5.14872 ] but the error above the ~start_state_max_bounds_error parameter (currently set to 0.1)

Indeed, when we monitor the joint_states topic /j2s7s300/joint_states, you see that joint 6 is at 7.401301704590916, which is outside of its limits. I could not find much information on this error in gazebo, but it seems it might be related to an instability or something when the robot get's spawned. 

I first tried to modify the controllers of the 7 dof robot [here ](https://github.com/Kinovarobotics/kinova-ros/blob/master/kinova_control/config/j2s7s300_control.yaml), but nothing would do. Then I tried to change the [urdf file ](https://github.com/Kinovarobotics/kinova-ros/blob/master/kinova_description/urdf/j2s7s300.xacro) and change the angle limits on joint 6. I was very surprised to see that if I am more tolerant on the limits e.g. 
`    <property name="joint_6_lower_limit" value="${30/180*J_PI}" />
    <property name="joint_6_upper_limit" value="${330/180*J_PI}" />`
the robot was then spawned with angle 6 inside the limits. I do not get this exact behaviour....

The thing is, the real Kinova 7 dof robot with a spherical wrist has angle limits 65-295 degree on joint 6 for safety reasons. Changing the limits for this actuator in the urdf file means the gazebo model as well as the planning in moveit will not be really representative of the real robot's capabilities. Then, I tried to modify the frames orientation so the zero position on joint 6 in the urdf is when link5 and link6 form a straight line. This corresponds to 180 degree on joint 6 on the real robot. 

The changes in the urdf file were the following:

joint 6 now defined as:
`<property name="joint_6" value="joint_6" />
    <property name="joint_6_type" value="revolute" />
    <property name="joint_6_axis_xyz" value="0 0 1" />
    <property name="joint_6_origin_xyz" value="0 0 -0.10375" />
    <property name="joint_6_origin_rpy" value="${-J_PI/2} 0 0" />
    <property name="joint_6_lower_limit" value="${-2.78}" />
    <property name="joint_6_upper_limit" value="${2.78}" />`

instead of 
 <property name="joint_6" value="joint_6" />
    <property name="joint_6_type" value="revolute" />
    <property name="joint_6_axis_xyz" value="0 0 1" />
    <property name="joint_6_origin_xyz" value="0 0 -0.10375" />
    <property name="joint_6_origin_rpy" value="${J_PI/2} 0 ${J_PI}" />
    <property name="joint_6_lower_limit" value="${65/180*J_PI}" />
    <property name="joint_6_upper_limit" value="${295/180*J_PI}" />

I also modified the start up position in [move_robot.py](https://github.com/Kinovarobotics/kinova-ros/blob/master/kinova_control/src/move_robot.py) from moveJoint ([0.0,2.9,0.0,1.3,4.2,1.4,0.0],prefix,nbJoints) to moveJoint ([0.0,2.9,0.0,1.3,1.5,**-1.8**,2.0],prefix,nbJoints) to take into account joint 6's new "zero position"

This is a working solution, but it requires to change what is published by [kinova_arm.cpp](https://github.com/Kinovarobotics/kinova-ros/blob/4ebae0119eb9ae5636e4ef8718c18415a78ba9b2/kinova_driver/src/kinova_arm.cpp) on j2s7s300_driver/out/joint_angles for joint 6, and possibly [kinova_joint_trajectory_controller.cpp](https://github.com/Kinovarobotics/kinova-ros/blob/master/kinova_driver/src/kinova_joint_trajectory_controller.cpp).

I wondered if anyone has already experienced any issues with Gazebo not respecting joint limits given in the urdf at spawning? If so, let us know

Cheers

haudren, 2018-05-25:
We are using gazebo and j2s7s300 right now, and never experienced such behaviour... Strange.

martine1406, 2018-05-30:
Yeah... well maybe it is related to my hardware (just guessing here?). I now I had some trouble running heavy simulation in Gazebo on my laptop... I'll look up on the internet once more to see if I find something related. Anyone, let us know if you experience such a similar behaviour and if you found a solution for it obviously :-)
Thanks!

haoxuw, 2018-07-04:
I have also encountered this issue. In my case the value for j2s7s300_joint_6 could be as high as 17. 

When outside of limit, I've also noticed the gazebo controller might also demonstrate incorrect behavior. For example, when the corrent postion is at 17, and I issue a command for 16.5, the controller sometimes direct the arm towards positive direction and reach 18, etc.

martine1406, 2018-07-04:
Hi @haudren 

Good observation. I still do not know what causes this issue though. It seems to be a problem with the Gazebo controller. This controller seems to prefer symmetric angle limits (e.g. -X to +X rather than limits from 0 to 2X). I do not know why....

sahandrez, 2018-09-06:
Hi, 

I have encountered this strange behaviour a while ago. Apparently, there is a problem with the `joint_state_controller` for publishing the states (look [here](https://github.com/jacknlliu/development-issues/issues/27) and [here](https://github.com/jacknlliu/development-issues/issues/49) for more information). In other words, Gazebo is respecting the joint limits, and for some strange reason the `joint_state_controller` is somehow changing the joint positions.

I am not using MoveIt! currently, but I stumbled upon this bug when I was trying to reset the arm in Gazebo through `set_model_configuration` service. Once the model is reset in Gazebo, the `joint_state_controller` does not publishes the reset values. 

I ended up implementing my own [plugin](https://github.com/sahandrez/kinova-ros/tree/master/kinova_plugin) for getting the joint states. This plugin publishes joint position and velocity. It also provides a topic for applying torque to the robot joints (which I guess you wouldn't be needing it).

Hope this helps, 
Sahand

rafaelpossas, 2018-09-24:
Hi @sahandrez 

I've tried your solution but when I launch the Robot in gazebo with use_ros_control:=false the arm shows as if there was no controller (not static in home position but jiggling).

Am I doing something wrong? Also, would your plugin fix the inconsistent values on joint_5 and joint_6?

Thanks in advance

felixmaisonneuve, 2021-06-02:
This thread has being inactive for over 2 years, I am closing this issue. If there is still a problem, please create a new issue.

Thank You

ZZWang21, 2023-05-19:
is the problem solved? I am using UBUNTU 20.04. The same problem? Can anyone help?

felixmaisonneuve, 2023-05-25:
Hi @ZZWang21,
You have the same behaviour? You get errors like 
```
[ INFO] [1538058213.924470173, 102.585000000]: Planning attempt 1 of at most 1
[ WARN] [1538058213.924668692, 102.585000000]: Joint 'j2n6s300_joint_3' from the starting state is outside bounds by a significant margin: [ 12.898 ] should be in the range [ 0.331613 ], [ 5.95157 ] but the error above the ~start_state_max_bounds_error parameter (currently set to 0.1)
[ INFO] [1538058213.925556105, 102.586000000]: Planner configuration 'arm' will use planner 'geometric::RRTConnect'. Additional configuration parameters will be set when the planner is constructed.
[ WARN] [1538058213.925981734, 102.586000000]: RRTConnect: Skipping invalid start state (invalid bounds)
[ERROR] [1538058213.926134385, 102.587000000]: RRTConnect: Motion planning start tree could not be initialized!
```

Can you elaborate on what you changed? What error messages do you get? How often do you have this issue? Does it happen in a particular sequence? In a specific position?

Best,
Felix

ZZWang21, 2023-05-25:
Currently I did not change anything. I am using arm to do reinforcement learning tasks. Every time the scene resets, the joint is 2pi more or less than the value I set. I have no idea why.

felixmaisonneuve, 2023-05-25:
This is interesting. 
If you are on Ubuntu20.04, this means you are using the Noetic branch with Gazebo11. Can you confirm?

Can you detail what is your exact sequence of commands and the behaviour you have?

Best,
Felix

ZZWang21, 2023-05-25:
Yes. Noetic with Gazebo11.

The robot goes to initial position. And it learns to Move to target using RL. After Every step it comes back to initial position. But after a certain number of cycles, the robot comes back to initial pose, but cannot move any more. Because the joint position out of bound, some joints are 2pi bigger than the initial position. But you know, 2pi bigger means the position is the same, but since it is out of bound, it will not move again.

felixmaisonneuve, 2023-05-25:
This makes me think it might be a wrap around issue. 
For example your actutator moves 3/4 of a turn, than you send it home and it is shorter to do the missing 1/4 turn instead of going back 3/4, but somewhere the position is not wrapped correctly so, when you try to send it somewhere else it just fails because position is out of bounds

Can you check what movement the simulation is doing before it fails to see if it could be something like this?

felixmaisonneuve, 2023-06-08:
I think you have no choice but to send a command with the position of each joint. 
You can inspire yoursef from our [move_robot.py](https://github.com/Kinovarobotics/kinova-ros/blob/melodic-devel/kinova_control/src/move_robot.py) script (our use this script directly if you are able to)

For the home position, you can find the position of eah joint in the Gazebo robot_launch.launch file : 
https://github.com/Kinovarobotics/kinova-ros/blob/d8a6b5a1434bdaed9b16fa7a14965ca8a1b8a998/kinova_gazebo/launch/robot_launch.launch#L37-L45

Best,
Felix