Kinovarobotics/kinova-ros #133: kinova_driver different architecture armhf support.

DataCorrupted, 2018-05-02:
The dynamically linked library provided in kinova_driver only has two architecture support, i386, and x86. 

However, I am using a tinker board for compatibility, which runs on armhf architecture. I can't use any of kinova's driver.

Would you please release source code of kinova_driver? Or cross compile it to armhf architecture?

Much thanks.

martine1406, 2018-05-08:
Hi @DataCorrupted

Unfortunately, we do not have any compiled version for armhf architecture yet... and we cannot share with you the source code either!

Can you let me know what Kinova robots do you have? May I have their serial numbers please
Thank you

DataCorrupted, 2018-05-09:
PJ0090 0006 WO432764 - 0

martine1406, 2018-05-09:
Hi Peter

I am sorry, but we cannot rebuild the library for you. I have a version I can share with you for ARMv8 64-bit CPU (as in the Jetson Nvidia Tx2 platform), but already, this is a custom version because we normally only compile for x86/x64 architecture running Linux or Windows. Let me know if you would be interested by these custom libraries for ARMv8 64-bit architecture. 

Sorry about that.
Martine

DataCorrupted, 2018-05-09:
Hi Martine:

Thanks for your effort.

At last, we decided to change our tinker board and use something else instead. No worries on armhf anymore!
Worse case scenario, we can try qemu a little bit.

Thanks again.

closed 2018-06-04 by martine1406

flymaxty, 2018-08-14:
Hi martine,
Currently we are trying to use Kinova on TX2, could you please tell us how to get compiled version for TX2?

Thanks!

martine1406, 2018-08-14:
Hi @flymaxty 

Please follow this link: https://github.com/Kinovarobotics/Extra_code_examples_Jaco2arm/tree/master/sdk%20compiled%20for%20tx2

flymaxty, 2018-08-14:
@martine1406 Thank you for your prompt reply! We will try it on our board.