Kinovarobotics/kinova-ros #135: Error Code 1015: Could not get general information about the device

sahandrez, 2018-05-16:
Hi, 

I know this issue has already been posted [here](https://github.com/Kinovarobotics/kinova-ros/issues/104), but there wasn't any solution to it. 

I tried to bring up the robot but the following error occurred. I am using Ubuntu 14.04 and ROS Indigo. Now the interesting point is that I was able to connect to the robot just a week ago, but I had to reinstall my Ubuntu and now it does not connect.

Another point worth mentioning is that `connectrobot` command gives back **device connected** before performing the `roslaunch` command, but after `roslaunch` fails, `connectrobot` also gives back ***device not found***.

This is the output of my terminal:

`$ connectrobot`
`Device connected`
`$ roslaunch kinova_bringup kinova_robot.launch kinova_robotType:=j2n6s300`
````
... logging to /home/sahand/.ros/log/4a7a2dec-5951-11e8-aadc-14abc57e911d/roslaunch-sahand-PC-21901.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://sahand-PC:45156/

SUMMARY
========

PARAMETERS
 * /j2n6s300_driver/connection_type: USB
 * /j2n6s300_driver/ethernet/local_broadcast_port: 25025
 * /j2n6s300_driver/ethernet/local_cmd_port: 25000
 * /j2n6s300_driver/ethernet/local_machine_IP: 192.168.100.100
 * /j2n6s300_driver/ethernet/subnet_mask: 255.255.255.0
 * /j2n6s300_driver/jointSpeedLimitParameter1: 10
 * /j2n6s300_driver/jointSpeedLimitParameter2: 20
 * /j2n6s300_driver/robot_name: j2n6s300
 * /j2n6s300_driver/robot_type: j2n6s300
 * /j2n6s300_driver/serial_number: not_set
 * /j2n6s300_driver/torque_parameters/publish_torque_with_gravity_compensation: False
 * /j2n6s300_driver/torque_parameters/use_estimated_COM_parameters: False
 * /kinova_number_of_robots: 2
 * /kinova_robots: [{'serial': 'PJ00...
 * /robot_description: <?xml version="1....
 * /rosdistro: indigo
 * /rosversion: 1.11.21

NODES
  /
    j2n6s300_driver (kinova_driver/kinova_arm_driver)
    j2n6s300_state_publisher (robot_state_publisher/robot_state_publisher)

auto-starting new master
process[master]: started with pid [21916]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 4a7a2dec-5951-11e8-aadc-14abc57e911d
process[rosout-1]: started with pid [21929]
started core service [/rosout]
process[j2n6s300_driver-2]: started with pid [21944]
process[j2n6s300_state_publisher-3]: started with pid [21947]
[ INFO] [1526506632.124767484]: kinova_robotType is j2n6s300.
[ INFO] [1526506632.125463399]: kinova_robotName is j2n6s300.
[ INFO] [1526506632.132080405]: Initializing Kinova USB API (header version: 50300, library version: 5.2.0)
[ERROR] [1526506642.589784574]: KinovaCommException: Could not get general information about the device (return code: 1015)

[j2n6s300_driver-2] process has died [pid 21944, exit code -11, cmd /home/sahand/catkin_ws/devel/lib/kinova_driver/kinova_arm_driver j2n6s300 __name:=j2n6s300_driver __log:=/home/sahand/.ros/log/4a7a2dec-5951-11e8-aadc-14abc57e911d/j2n6s300_driver-2.log].
log file: /home/sahand/.ros/log/4a7a2dec-5951-11e8-aadc-14abc57e911d/j2n6s300_driver-2*.log
````
`$ connectrobot`
`Device not found`

 I have followed the instructions completely and have copied `10-kinova-arm.rules` to udev and have also tried changing its name to `60-kinova-arm.rules`.

Is something wrong with the drivers? Any help would be appreciated. 

Thanks in advance. 
Sahand

nsaif-kinova, 2018-05-17:
Hello Sachand,
When you turn the robot ON and you send it home using the joystick, do you have a stable green light always ON on the joystick?
Do you have problems connecting the robot to the Dev Center on Ubuntu?
Thank you.

nsaif-kinova, 2018-05-17:
Hello Sahand,
When you turn the robot ON and you send it home using the joystick, do you have a stable green light always ON on the joystick?
Do you have problems connecting the robot to the Dev Center on Ubuntu?
Thank you.

sahandrez, 2018-05-17:
Hello Nassif, 
Yes there was a stable green light. Actually, I have figured out the problem; tlp package was preventing the USB connection. I removed it, and everything is working fine now. 
Thanks!

nsaif-kinova, 2018-05-17:
Perfect thank you.

-- 

*Nassif Saif*

Innovation - Field Application Engineer

514-277-3777,4192

nsaif@kinova.ca

4333 Grande-Allée blvd., Boisbriand (QC)
<https://maps.google.com/?q=4333+Grande-All%C3%A9e+Boulevard,+Boisbriand+QC++J7H+1M7&entry=gmail&source=g>

<https://maps.google.com/?q=4333+boul.+de+la+Grande-All%C3%A9e,+Boisbriand+(QC)++%C2%A0+J7H+1M7&entry=gmail&source=g>J7H
1M7
<https://maps.google.com/?q=4333+Grande-All%C3%A9e+Boulevard,+Boisbriand+QC++J7H+1M7&entry=gmail&source=g>

kinovarobotics.com <http://www.kinovarobotics.com/>   Facebook
<http://facebook.com/kinovarobotics/>  Twitter
<http://twitter.com/kinovarobotics>  LinkedIn
<https://www.linkedin.com/company/kinova/>  YouTube
<https://www.youtube.com/user/Kinovarobotics>

The information contained in this email message is private and
confidential. It is intended for the exclusive use by the recipient named
above. If you are not the intended recipient, be advised that disclosing,
copying, distributing or taking any action in reliance on the contents of
this information is strictly prohibited. If you have been sent this email
by error, please destroy it and immediately inform us at
info@kinovarobotics.com.

JuliusSustarevas, 2018-05-25:
I'm having the same issue with the same arm. I  can get it to run if  i use use_urdf=false. Can you elaborate what the solution here was?

closed 2018-06-04 by martine1406

Hubert51, 2019-06-26:
I have the same problem. And my green LED flashing all the time. Do you know why I can not get stable green LED light?

dominikfrey, 2019-06-28:
I got the same problem. I could fix it with just giving Kinova little bit more time. The light flashes in the beginning while starting up and after a time it is constant. (I guess that's how it is supposed to be). When I check `lsusb` it shows the USB connection shortly after the lights went green. The earliest after this I start my ROS node. Then at least the `ERROR 1015` is gone.

But then I (sometimes) get the error:
> [ INFO] Kinova Init Communication completed
> [ INFO] devices_count: 1
> [ INFO] [1561728129.390236956]: Found 1 device(s), using device at index 0, serial number:  ?w<?)
> [ERROR] Could NOT find actuator with index0

Does this have something to do with each other?

KinovaCustomerSupport, 2019-07-02:
Hi @Hubert51,

I don't think you have the issue describe in this thread.
It looks like your arm has a firmware problem, could you fill a quick support request here : https://www.kinovarobotics.com/en/support

I will then send you the procedure to fix this.

Thanks,
Jean-Philippe Côté
Customer support

Hubert51, 2019-07-09:
I fixed my problem. And I write the solution in  https://github.com/Kinovarobotics/kinova-ros/issues/248 Thank you!

cygreat7, 2019-07-18:
> Hello Nassif,
> Yes there was a stable green light. Actually, I have figured out the problem; tlp package was preventing the USB connection. I removed it, and everything is working fine now.
> Thanks!

Which is the tlp package?

ferrolho, 2021-08-09:
I was also having this issue. Running the command below solved my problems:
```
sudo apt --purge remove 'tlp*'
```