Kinovarobotics/kinova-ros #140: Finger positions of kinova arm does not update in RViz

RajViky, 2018-06-01:
Hi,
I am using kinova-ros package installed from source (kinova-driver release 1.2.1). The finger positions of kinova arm does not update in Rviz.

But both joint_state and finger positions topics are published. And I am able to controller each fingers with the script "fingers_action_client.py".

rosrun kinova_demo fingers_action_client.py j2n6s300 -r -v percent -100 -100 100

Tf is intact.

snippet of topic /j2n6s300_driver/out/joint_state

name: ['j2n6s300_joint_1', 'j2n6s300_joint_2', 'j2n6s300_joint_3', 'j2n6s300_joint_4', 'j2n6s300_joint_5', 'j2n6s300_joint_6', 'j2n6s300_joint_finger_1', 'j2n6s300_joint_finger_2', 'j2n6s300_joint_finger_3']
position: [4.357342735441942, 2.9764899818103308, 0.8698737663328161, 3.9588342341413187, 1.6252170915642843, 1.3924464259631075, 0.012781970648700604, 0.012063306225569713, 0.012268638026209425]
velocity: [0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0]
effort: [-0.21822483837604523, -0.0, 0.003679371438920498, -0.0, 0.0, 0.0, 0.0, 0.0, 0.0]

snippet of topic /j2n6s300_driver/out/finger_position

finger1: 62.25
finger2: 58.75
finger3: 59.75
---

Arm model: j2n6s300
rosversion: Kinetic

RajViky, 2018-06-02:
Hi @martine1406 ,
I dint try it in indigo version. But I want to make it work in kinetic version or I need to change my other sensor nodes to indigo and 14.04 has more problem, I am not considering for switch. And all topics from this repo are published successfully, i am guessing this finger_position update should be a minor issue. But i couldn't debug it. Kindly help me!

martine1406, 2018-06-04:
Hi @RajViky

Can you confirm what branch are you using. Have you modified the code in any way?
Also, what are the exact commands you are sending? 
With roscore, 
then roslaunch kinova_bringup kinova_robot.launch kinova_robotType:=j2n6s300, 
then rosrun kinova_demo fingers_action_client.py j2n6s300 -r -v percent 100 100 100 or rosrun kinova_demo fingers_action_client.py j2n6s300 -r -v percent -100 -100 -100 (to close and open the fingers respectively), 
topics /j2n6s300_driver/out/joint_state and /j2n6s300_driver/out/finger_position get updated. 

I am on Indigo and the master branch. Are you running the exact same above code on Kinetic? 
Let's say you roslaunch kinova_bringup kinova_robot.launch kinova_robotType:=j2n6s300, then directly rosrun kinova_demo fingers_action_client.py j2n6s300 -r -v percent 100 100 100, do you see the fingers on the real hardware closing completely?

The fact that you get perfect zeros in the velocity and many of the effort fields in topic /j2n6s300_driver/out/joint_state seems weird to me. 

The joint_state message is filled out in this [function ](https://github.com/Kinovarobotics/kinova-ros/blob/c89b4ba71438b3bde4f007d470a30d7dfb579e38/kinova_driver/src/kinova_arm.cpp#L502). Maybe you can start your debugging from here?

Thank you
Martine

martine1406, 2018-06-04:
Hi Raj

My rosversion is 1.11.21.
out/finger_position and out/joint_state get updated with the same finger position data, coming from kinova_comm_.getFingerPositions(fingers). This function will read the position sensors of the real robot's gripper. 

Maybe I did not get your question correctly. You say that the topics ARE updated correctly (sorry, I thought you were saying that they weren't updated). The topics are updated correctly both on Indigo and Kinetic?  What you are saying is, if you open Rviz, the joint positions are updated correctly, but not the fingers'. In my case, it seems to be working on Indigo....

I do 
roscore
then roslaunch kinova_bringup kinova_robot.launch kinova_robotType:=j2n6s300,
then rosrun rviz rviz (I see the robot model appear)
then rosrun kinova_demo fingers_action_client.py j2n6s300 -r -v percent 100 100 100

and I see the fingers closing on my real robot and on the rviz model (with some minor delay). Do you see any difference at this level between Indigo and Kinetic?

thanks
Martine

RajViky, 2018-06-04:
Hi Martine,
I am using Kinetic and didn't test it in indigo yet. Yes, topics are published and I am able to control joint positions and finger positions. But only the finger poses are not updated in Rviz, that's my problem. (maybe issue with URDF)

If it works in Indigo, then the issue is with kinetic version. Is there any workaround for kinova-ros package in kinetic devel. It was discussed in #117, but does this solved completely?  I think its time to create a kinetic-devel branch separately and solve this issue.

My main problem is I cannot revert to indigo version as all other nodes of my system are intact with kinetic version.

BR,
Raj

martine1406, 2018-06-04:
Hi Raj

well you could try to test the code from this [repo](https://github.com/CNURobotics/kinova-ros/tree/kinetic_devel), which is discussed in [#117](https://github.com/Kinovarobotics/kinova-ros/issues/117)

I understand your concern about switching to indigo. @sjhansen3, @mpmendes, @dcconner since you guys are working on kinetic, do you have the same problem?

I'll try to find a 16.04 computer with kinetic to test on my end too. If you find the way to solve this issue, do not hesitate to submit a pull request for a kinetic-devel branch with the fix. 

thanks and let me know
Martine

dcconner, 2018-06-04:
IIRC, the base fingers update properly in our repo, but not the finger tips.  I believe they need to have a special kind of linked joint in the URDF that I haven't bothered to fix

martine1406, 2018-06-04:
@dcconner 

thank you for your answer. The real robot's fingers are under-actuated, but this under-actuation is not modeled in the URDF file. This is probably linked to the behaviour you are observing on your end.

mpmendes, 2018-06-04:
Hi,
Hm, I don't  recall checking that kind of data with RViz using kinetic, but the problem could exist (I suppose...?). Unfortunately, I wouldn't be able to test it too, as I'm not working with the robot anymore 
:-| .

closed 2018-07-03 by martine1406