Kinovarobotics/kinova-ros #141: Does the robot in Gazebo use the same controller as the real robot?

sahandrez, 2018-06-04:
Hi, 
This is more of a question rather than an issue. I have implemented a code which runs the robot via MoveIt planner. It can perfectly run on both the virtual and actual robot and everything is fine. The code is available [here](https://github.com/sahandrez/aml_jaco).

Now as far as I know, MoveIt! plans the trajectory and sends out the points along side the path. Ros_control then gets the joint angles (correct me if I am wrong) and guides the robot through them. Now what I need to know is that are both the real robot and the robot in Gazebo use the exact same controllers implemented in ros_control? If not, what are the differences of these two controllers? 

Could you also briefly explain the low-level controllers on the arm?

Thank you very much.
Sahand

martine1406, 2018-06-04:
Hi @sahandrez 

Thanks for sharing your code and your question. This is a good question actually. The answer is that Gazebo and the real robot each have their separate controller, which is not the same in both cases. The real robot's PID controllers are directly implemented in the actuators' firmware. The gains will be different. I guess you want to get more information and access the controllers for the real robot?

In the real arm, the actuators include a PID controller on the position (used when doing position control... which you do in Moveit) and a P controller for the torque (used when doing torque control). The default gains are P=2.0, I=0, D=0.05. You have an API function to change those gains: SetActuatorPID (unsigned int address, float P, float I, float D). In the ros package, you can access this function from [here ](https://github.com/Kinovarobotics/kinova-ros/blob/master/kinova_driver/src/kinova_api.cpp#L145).

I hope this answers your question. 
Martine

sahandrez, 2018-06-04:
Hi Martine,

Thank you very much for your elaborate response. From what you have explained, I suppose ros_control is only used in Gazebo then; and in the case of the real robot, the joint angles planned by MoveIt! are directly sent to the hardware. Is that right?

How can I configure the ros_control package used in Gazebo and try to make it as close as possible to the controller on the real robot? Because I suppose the controller on the real arm is not open source.

As the last question, are torque and velocity controls available in Gazebo?

Thank you again. 
Sahand

martine1406, 2018-06-04:
Hi @sahandrez 

Yes, ros_control is only used in Gazebo. And yes, in the case of the real robot, the joint angles planned by MoveIt! are directly sent to the hardware.
Honestly, I would not know how to make the ros_control package used in Gazebo as close as possible to the controller on the real robot :-O. Since the gains are very different, and I am not super familiar with ros_control controllers, it's hard to say. I guess it would involve tuning the gains of the ros_control joint controllers until you get similar responses to different input signals from both the real and the virtual actuators. But this will be quite a tedious process since you have to stimulate and monitor both the real and virtual hardware
Furthermore, we do not have a very elaborate model of the actuator in the URDF, so the real hardware and the simulation hardware behaviours could be different even if the controlllers would be the same. Our goal here was to get a virtual representation of the robot with functional controllers that give you a rough idea on the real robot's general behaviour, not to get a perfect replica of the real robot in Gazebo... 

I think torque and velocity controllers are supported in ros_control controller in general: position controller, velocity controller or effort controller.  See '2. Controllers' section on the [ros_control page](http://wiki.ros.org/ros_control). Now, those controllers are not implemented by default for kinova robot models....  if you want to use those type of control signal, you will need to describe the additional controllers on your end I believe (you can add them in the right yaml file in [kinova_control/config/](https://github.com/Kinovarobotics/kinova-ros/tree/master/kinova_control/config)). 

Sorry I cannot help you more
Martine

sahandrez, 2018-06-04:
Hi Martine,

Thank you so much for your reply. Your answer put me on the right track. Now I know where I should look into. 

Bests, 
Sahand

dcconner, 2018-06-04:
There is a ActionLib server provided as part of the kinova_driver package that provides the same interface as the Gazebo ros controller, but this does not work the same as it just forwards a command to the kinova driver.   This causes issues as you lose some of the safety checks and cannot easily terminate a trajectory or monitor performance.

It is on my todo wish list to re-write the kinova driver for our use, but it keeps getting pushed down the priority stack.

closed 2018-07-03 by martine1406