Kinovarobotics/kinova-ros #143: Hello! Problem of setting value of Kinova arm joint by moveit setJointPositions

HankerSia, 2018-06-22:
When we request a pose to moveit api, we can get the plan result, As following:
     `group_->setPoseTarget(start_pose_);
      my_plan= evaluate_plan(*group_);`
I modified the evaluate_plan function so that i can let plan result return. Then i print part information of the plan result, as following:

    int joint_name_size, plan_point_size;
    joint_name_size = my_plan.trajectory_.joint_trajectory.joint_names.size();   
    plan_point_size = my_plan.trajectory_.joint_trajectory.points.size();
    std::cout <<"joint_name size:"<< joint_name_size <<std::endl;
    std::cout <<"points  size:"<< plan_point_size <<std::endl;
    for(int i=0; i< joint_name_size;i++)
    {
        ROS_INFO("joint name:%s", my_plan.trajectory_.joint_trajectory.joint_names[i].c_str());
    }
    ROS_INFO_STREAM("Last point of Plan:"<<my_plan.trajectory_.joint_trajectory.points[plan_point_size-1]);
So, we can know the joint name and a points demo, as following:
joint_name size:6
points  size:30
[ INFO] [1529570085.394579180]: joint name:j2n6s300_joint_1
[ INFO] [1529570085.394606035]: joint name:j2n6s300_joint_2
[ INFO] [1529570085.394628844]: joint name:j2n6s300_joint_3
[ INFO] [1529570085.394651328]: joint name:j2n6s300_joint_4
[ INFO] [1529570085.394673488]: joint name:j2n6s300_joint_5
[ INFO] [1529570085.394695385]: joint name:j2n6s300_joint_6
[ INFO] [1529570085.394770602]: Last point of Plan:positions[]
  positions[0]: -2.19888
  positions[1]: 3.73149
  positions[2]: 1.84508
  positions[3]: -2.77105
  positions[4]: 0.407463
  positions[5]: 2.5911
velocities[]
  velocities[0]: 0
  velocities[1]: 0
  velocities[2]: 0
  velocities[3]: 0
  velocities[4]: 0
  velocities[5]: 0
accelerations[]
  accelerations[0]: 0.559416
  accelerations[1]: -0.627759
  accelerations[2]: -0.652831
  accelerations[3]: 0.54093
  accelerations[4]: 0.800661
  accelerations[5]: -0.988729
effort[]
time_from_start: 4.042044055
Then, i wrote a function used to solve forward kinematic model, as the following:
'
```
void PickPlace::getForK(std::vector<double> &joint_value, geometry_msgs::Pose &m)
{
    ROS_INFO("Forward Kinematic test");
    robot_state::RobotStatePtr kinematic_state(new robot_state::RobotState(robot_model_));
    kinematic_state->setToDefaultValues();
    const robot_state::JointModelGroup* joint_model_group = robot_model_->getJointModelGroup("arm");
    const std::vector<std::string> &joint_names = joint_model_group->getJointModelNames();  
    // for(std::size_t i = 0; i < joint_names.size(); i++)
    // {
    //     ROS_INFO("Write Size: %dJoint i:%d %s: %f", joint_names.size(), i, joint_names[i].c_str(), joint_value[i]);
    // }
    // kinematic_state->setToRandomPositions(joint_model_group);
    for(char i=0;i<joint_value.size();i++)
    {
        kinematic_state->setJointPositions(joint_names[i+1], &joint_value[i]);
    }
    //ROS_INFO("Joint i:%d %s: %f", i, joint_names[i].c_str(), joint_value[i]);
    // std::vector<double> joint_values_(7, 0.0);
    // 
    //j2n6s300_joint_1 joint_values_[0] ;
    //j2n6s300_joint_2 joint_values_[1] ;
    //j2n6s300_joint_3 joint_values_[2] ;
    //j2n6s300_joint_4 joint_values_[3] ;
    //j2n6s300_joint_5 joint_values_[4] ;
    //j2n6s300_joint_6 joint_values_[5] ;
    // kinematic_state->setJointGroupPositions(joint_model_group,joint_values_);   
    // kinematic_state->setJointPositions(joint_names[0],angle);
    // kinematic_state->setToRandomPositions(joint_model_group);
    std::vector<double> joint_values;
    kinematic_state->copyJointGroupPositions(joint_model_group, joint_values);
    for(std::size_t i = 0; i < joint_names.size(); i++)
    {
        ROS_INFO("Read Joint i:%d %s: %f", i, joint_names[i].c_str(), joint_values[i]);
    }  
    const Eigen::Affine3d &end_effector_state = kinematic_state->getGlobalLinkTransform("j2n6s300_end_effector");
    /* Print end-effector pose. Remember that this is in the model frame */
    //ROS_INFO_STREAM("Translation: " << end_effector_state.translation());
    //ROS_INFO_STREAM("Rotation: " << end_effector_state.rotation());   
    tf::poseEigenToMsg(end_effector_state,m);
    ROS_INFO_STREAM("Pose Computed:"<<m);
}
```
'
While, The first problem is the joint name read from joint_model_group->getJointModelNames() has 8 members which is j2n6s300_joint_base, j2n6s300_joint_1, ... , j2n6s300_joint_6, j2n6s300_joint_end_effector. Then, when i tried to set joint value according joint name, like the follow method(p.s. i set a std::vector<double> forKJointValue(6,0.0) as the getForK() input variable):
    for(char i=0;i<joint_value.size();i++)
    {
        kinematic_state->setJointPositions(joint_names[i+1], &joint_value[i]);
    }
Ideally, i would only set joint value of  j2n6s300_joint_1, ... , j2n6s300_joint_6. But, i got follow results
[ INFO] [1529633314.430620000]: Read Joint i:0 j2n6s300_joint_base: -1.478416
[ INFO] [1529633314.430661300]: Read Joint i:1 j2n6s300_joint_1: 2.924741
[ INFO] [1529633314.430705143]: Read Joint i:2 j2n6s300_joint_2: 1.001931
[ INFO] [1529633314.430739710]: Read Joint i:3 j2n6s300_joint_3: -2.080089
[ INFO] [1529633314.430767370]: Read Joint i:4 j2n6s300_joint_4: 1.445890
[ INFO] [1529633314.430795050]: Read Joint i:5 j2n6s300_joint_5: 1.323290
[ INFO] [1529633314.430823610]: Read Joint i:6 j2n6s300_joint_6: 0.000000
[ INFO] [1529633314.430857823]: Read Joint i:7 j2n6s300_joint_end_effector: 0.000000
Although i tried different methods, but i can not solve this problem. But when i ignore this problem, the position of distal joint seems right, because i displayed point computed by getForK() using rviz marker. While i am confused about all of this. Looking forward your advice! Thanks a lot! 
BTW, i solve the inverse kinematic problem. As following:
'
```
void PickPlace::getInvK(geometry_msgs::Pose &eef_posee, std::vector<double> &joint_values)
{
    ROS_INFO("Inverse Kinematic test");
    robot_state::RobotStatePtr kinematic_state(new robot_state::RobotState(robot_model_));
    kinematic_state->setToDefaultValues();
    const robot_state::JointModelGroup* joint_model_group = robot_model_->getJointModelGroup("arm");
    const std::vector<std::string> &joint_names = joint_model_group->getJointModelNames();   
    geometry_msgs::Pose target_position1=eef_posee; // Expressed in group->getPlanningFrame()
	// target_position1.position.x =  0.5;
	// target_position1.position.y = -0.5;
	// target_position1.position.z =  0.5;
    // tf::Quaternion q;
    // q = EulerZYZ_to_Quaternion(-M_PI/4, M_PI/2, M_PI);
    // target_position1.orientation.x = q.x();
    // target_position1.orientation.y = q.y();
    // target_position1.orientation.z = q.z();
    // target_position1.orientation.w = q.w();	
    // Compute IK
    Eigen::Affine3d target_eig_pose1;
    // const Eigen::Affine3d &end_effector_state = kinematic_state->getGlobalLinkTransform("j2n6s300_end_effector");
    std::string kinova_distalLink = "j2n6s300_end_effector";
    tf::poseMsgToEigen(target_position1, target_eig_pose1);	
    bool found_ik = kinematic_state->setFromIK(joint_model_group, target_eig_pose1, kinova_distalLink, 10, 0.1);
    if (found_ik)
    {
        kinematic_state->copyJointGroupPositions(joint_model_group, joint_values);
        for(std::size_t i=0; i < joint_names.size(); ++i)
        {
            ROS_INFO("InvK Joint %s: %f", joint_names[i].c_str(), joint_values[i]);
        }
    }
    else
    {
        ROS_INFO("Did not find IK solution");
    }
}
```
'
I wish it can help you!

HankerSia, 2018-06-27:
Hello, Is there anybody?

martine1406, 2018-06-27:
HI @HankerSia 

Sorry for the delay in answering you. If I get you correctly, your commanded angles update the 6 first joints that are returned by getJointModelNames(), a little bit as if you specified 

for(char i=0;i<joint_value.size();i++)
{
kinematic_state->setJointPositions(joint_names[i], &joint_value[i]);
}

although you specify 

for(char i=0;i<joint_value.size();i++)
{
kinematic_state->setJointPositions(joint_names[i+1], &joint_value[i]);
}

first of all, shouldn't i be declared as an int?
then, are you sure you should call setJointPositions(joint_names[i+1], &joint_value[i]), not setJointPositions(joint_names[i+1], joint_value[i])? I saw the setJointPositions used in this example http://docs.ros.org/indigo/api/moveit_tutorials/html/doc/pr2_tutorials/planning/src/doc/planning_pipeline_tutorial.html
Besides that, I do not really know why you get this weird behaviour, but again, if I get you correctly, although you get some weird readings, the frames in Rviz seem to be updated correctly?
Finally, have you tried posting this same question on a MoveIt forum/Q&A (e.g. https://answers.ros.org/questions/scope:all/sort:activity-desc/tags:moveit_core/page:1/)? I do not think this is specifically related to Kinova robots, but might be related to the kinematic_state->setJointPositions function

All the best,
Martine

HankerSia, 2018-06-28:
Hi@martine1406
Thanks for your advice!
I have asked the problem here [https://answers.ros.org/question/295568/problem-of-setting-value-of-arm-joint-by-moveit-setjointpositions/](url). But it seems there is nobody answer it.

HankerSia, 2018-06-28:
Thanks for your advice!
BTW, It seems that i can not upload attachments because net restriction of my region! If you can leave a email , it would be convenient for me to send you the main app file and the result picture of arm distal link.

martine1406, 2018-07-03:
HI @HankerSia

Did you find the solution to your problem?

HankerSia, 2018-07-04:
@martine1406 Hi!
In a word, the problem is the joint name read from joint_model_group->getJointModelNames() has 8 members which is j2n6s300_joint_base, j2n6s300_joint_1, ... , j2n6s300_joint_6, j2n6s300_joint_end_effector. Then, when i tried to set joint value according joint name, like the follow method(p.s. i set a std::vector forKJointValue(6,0.0) as the getForK() input variable):
for(char i=0;i<joint_value.size();i++)
{
kinematic_state->setJointPositions(joint_names[i+1], &joint_value[i]);
}
Ideally, i would only set joint value of j2n6s300_joint_1, ... , j2n6s300_joint_6. But, i got follow results
[ INFO] [1529633314.430620000]: Read Joint i:0 j2n6s300_joint_base: -1.478416
[ INFO] [1529633314.430661300]: Read Joint i:1 j2n6s300_joint_1: 2.924741
[ INFO] [1529633314.430705143]: Read Joint i:2 j2n6s300_joint_2: 1.001931
[ INFO] [1529633314.430739710]: Read Joint i:3 j2n6s300_joint_3: -2.080089
[ INFO] [1529633314.430767370]: Read Joint i:4 j2n6s300_joint_4: 1.445890
[ INFO] [1529633314.430795050]: Read Joint i:5 j2n6s300_joint_5: 1.323290
[ INFO] [1529633314.430823610]: Read Joint i:6 j2n6s300_joint_6: 0.000000
[ INFO] [1529633314.430857823]: Read Joint i:7 j2n6s300_joint_end_effector: 0.000000
it means that i did not set joint value rightly. A case in point is j2n6s300_joint_base has been set as the value of j2n6s300_joint_1. Generally, the value of target joint has a shift.

HankerSia, 2018-07-04:
And when we print the plan result of  my_plan.trajectory_.joint_trajectory.joint_names[i], it only showed follow joint names:
[ INFO] [1529570085.394579180]: joint name:j2n6s300_joint_1
[ INFO] [1529570085.394606035]: joint name:j2n6s300_joint_2
[ INFO] [1529570085.394628844]: joint name:j2n6s300_joint_3
[ INFO] [1529570085.394651328]: joint name:j2n6s300_joint_4
[ INFO] [1529570085.394673488]: joint name:j2n6s300_joint_5
[ INFO] [1529570085.394695385]: joint name:j2n6s300_joint_6
There is not  j2n6s300_joint_base.

HankerSia, 2018-07-04:
So i do not test my control demo on real jaco arm, i am afraid that it won't work...

martine1406, 2018-07-04:
HI @HankerSia 

I guess that you do not see j2n6s300_joint_base when you print my_plan.trajectory_.joint_trajectory.joint_names[i] because j2n6s300_joint_base is a fixed joint. 

Can you print my_plan.trajectory_.joint_trajectory.points and see if those make sense?

If you do not want to test on the real robot, you can always test on the Gazebo model. 

Have you tried calling
kinematic_state->setJointPositions(joint_names, table_of_joint_value)
with table_of_joint_value = [0.0, your 6 desired joint position, 0.0] instead?

I get your problem more clearly now, but what I am telling you is that it is not related to Kinova robots specifically. It is related to moveit packages and how you use them. Consequently, I will need to close that issue for now. Please open another issue if you have other questions related to Kinova robots. 

Thank you and good luck with that

closed 2018-07-04 by martine1406

HankerSia, 2018-07-06:
While i think problem comes from the definition of  jaco moveit controller, Do not you think so?

martine1406, 2018-07-06:
Hi @HankerSia 

I am not too sure I get what you are trying to tell me with the code. Does this code give you something different than before?
What makes you think it would be related to the definition of jaco moveit controller? 
Thank you
Martine

HankerSia, 2018-07-10:
@martine1406 i guess it, because i had tried all kinds of method, none of them succeed, so it won't be problem of moveit... but it's just guess, cuz i am beginer level programer of moveit