Kinovarobotics/kinova-ros #145: Running kinova_bringup kinova_robot.launch hangs, and `rostopic list` shows the related topics were not loaded

haoxuw, 2018-06-27:
Hello,

Thanks for reading my question : )

The kinova_robot.launch under kinova_bringup hangs on my Ubuntu machine:
$ uname -a
Linux haoxus 4.4.0-124-generic #148~14.04.1-Ubuntu SMP Thu May 3 07:26:53 UTC 2018 x86_64 x86_64 x86_64 GNU/Linux

The workspace is also a fresh copy:
$ git log
commit 6ea73bd5acb1b3505ecfc79c6d25048193518204
Author: Nassif <35464719+nsaif-kinova@users.noreply.github.com>
Date:   Tue Jun 26 09:38:43 2018 -0400

Here is the console log:
----------------------------------------------------------------------------------

$ roslaunch kinova_bringup kinova_robot.launch kinova_robotType:=j2n6s300
... logging to /home/haoxuw/.ros/log/1487ca58-799c-11e8-be1a-00e18cc7b109/roslaunch-haoxus-14486.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://haoxus:40105/

SUMMARY
========

PARAMETERS
 * /j2n6s300_driver/connection_type: USB
 * /j2n6s300_driver/ethernet/local_broadcast_port: 25025
 * /j2n6s300_driver/ethernet/local_cmd_port: 25000
 * /j2n6s300_driver/ethernet/local_machine_IP: 192.168.100.100
 * /j2n6s300_driver/ethernet/subnet_mask: 255.255.255.0
 * /j2n6s300_driver/jointSpeedLimitParameter1: 10
 * /j2n6s300_driver/jointSpeedLimitParameter2: 20
 * /j2n6s300_driver/robot_name: j2n6s300
 * /j2n6s300_driver/robot_type: j2n6s300
 * /j2n6s300_driver/serial_number: not_set
 * /j2n6s300_driver/torque_parameters/publish_torque_with_gravity_compensation: False
 * /j2n6s300_driver/torque_parameters/use_estimated_COM_parameters: False
 * /kinova_number_of_robots: 2
 * /kinova_robots: [{'serial': 'PJ00...
 * /robot_description: <?xml version="1....
 * /rosdistro: indigo
 * /rosversion: 1.11.21

NODES
  /
    j2n6s300_driver (kinova_driver/kinova_arm_driver)
    j2n6s300_state_publisher (robot_state_publisher/robot_state_publisher)

ROS_MASTER_URI=http://localhost:11311

core service [/rosout] found
process[j2n6s300_driver-1]: started with pid [14507]
process[j2n6s300_state_publisher-2]: started with pid [14512]
[ INFO] [1530117262.824434591]: kinova_robotType is j2n6s300.
[ INFO] [1530117262.824953395]: kinova_robotName is j2n6s300.
[ INFO] [1530117262.829258144]: Initializing Kinova USB API (header version: 50300, library version: 5.2.0)
[ INFO] [1530117262.985682571]: Found 1 device(s), using device at index 0 (model: Jaco V2 Service, serial number: PJ00900006163520001, code version: 328960, code revision: 5)
------------------------------------------

During hanging, rostopic list does not return the related topics.

During driver's hanging pose_action_client also hangs:
$ rosrun kinova_demo pose_action_client.py -v -r j2n6s300 mdeg -- 0.01 0.02 0.1 0.1 1 1`
`control-c` Traceback : hangs at:
    rospy.wait_for_message(topic_address, kinova_msgs.msg.KinovaPose)

I suspect some other ubuntu package is interferencing with Kinova driver. Please let me know if anyone has encountered this issue. 

Thanks!

martine1406, 2018-06-27:
Hi @haoxuw 

Just to be sure, you do not have the SDK/Development Center and the kinova-ros driver running at the same time, right?

Do you have other packages in your catkin workspace?

I do not have any problem on my side. 
uname -a  gives me
Linux KINOVA-LPT-75-U 4.4.25-040425-generic #201610161231 SMP Sun Oct 16 16:33:39 UTC 2016 x86_64 x86_64 x86_64 GNU/Linux

 I do not know if the Linux kernel version can make a difference? yours seems more recent than mine...
Let me know
Martine

martine1406, 2018-07-03:
Any news @haoxuw?

haoxuw, 2018-07-04:
Hi Martine,

I am using a different arm now, which also hanged for me before. For unknown reasons, it works now. My best guess is that the setup.sh of one of my experimental workspace contaminated the environment.

Now showing:

[ INFO] [1530117262.985682571]: Found 1 device(s), using device at index 0 (model: Jaco V2 Service, serial number: PJ00900006163520001, code version: 328960, code revision: 5)
[ INFO] [1530721088.439046865]: Initializing fingers...this will take a few seconds and the fingers should open completely
[ INFO] [1530721088.597100192]: The arm is ready to use.
j2n6s300_joint_1 j2n6s300_joint_2 j2n6s300_joint_3 j2n6s300_joint_4 j2n6s300_joint_5 j2n6s300_joint_6 

(For my curiosity, I also tried to run DevelopmentCenter alone with bringup, they would crash on my machine, however with a different message rather than hanging.)

Thanks for looking into it. I will close this thread.

My email is haoxuw@gmail.com , please ping me for any questions.

closed 2018-07-04 by haoxuw