Kinovarobotics/kinova-ros #147: Kinova arm moving very slow

Achllle, 2018-07-06:
When publishing velocity commands to the kinova mico 

i.e.
`
rostopipub -r 100 /m1n6s300_driver/in/joint_velocity kinova_msgs/JointVelocity "{joint1: 0.0, joint2: 100.0, joint3: 2.3, joint4: 0.0, joint5: 0.0, joint6: 100.0, joint7: 0.0}" 
`

the kinova mico moves very very slowly. On the order 0.25 deg per second with a command of 100 sent at a rate of 100 hz.

1. Can you clarify what the units are of the joint velocities?
2. What could be causing this?
3. Does the publish rate effect the velocity of the joint?

This feels like a potential kinova comms issue, please let me know if there are any logs or debug steps I can take to help you reproduce this error. It is intermittent and quite frustrating. 

Any and all help is appreciated.

nsaif-kinova, 2018-07-06:
Hi,
The joint velocity is degree/s

Please try to reinstall your Kinova ROS package, and update your Mico Firmware
Test your robot velocity using the Dev Center Virtual joystick to make sure all your actuators are moving properly.

The following command can move the 4th joint of a mico robot at a rate of approximate 10 degree/second. Please be aware that the publishing rate does affect the speed of motion.
`rostopic pub -r 100 /m1n6s300_driver/in/joint_velocity kinova_msgs/JointVelocity "{joint1: 0.0, joint2: 0.0, joint3: 0.0, joint4: 10.0, joint5: 0.0, joint6: 0.0}"`

** Note on publish frequency ** The joint velocity is set to publish at a frequency of 100Hz, due to the DSP inside the robot which loops each 10ms. Higher frequency will not have any influence on the speed. However, it will fill up a buffer (size of 2000) and the robot may continue to move a bit even after it stops receiving velocity topics. For a frequency lower than 100Hz, the robot will not able to achieve the requested velocity.
Therefore, the publishing rate at 100Hz is not an optional argument, but a requirement.

Also Please note you have a speed limit on the joints, you can show these limits using the following commands : 
`rosparam get /m1n6s300_driver/jointSpeedLimitParameter1 `(for joint 1,2,3)
`rosparam get /m1n6s300_driver/jointSpeedLimitParameter2` (for joints 4,5,6)

to change the velocity you can edit this file kinova_bringup/launch/config/robot_parameters.yaml there is joint speed limit for joints 1,2,3 (big actuators) and joints 4,5,6(small actuators) and you can set this to your desired value, the big actuators max speed is 36 deg/sec and the small ones 48 deg/sec

you can also set it sending the following command : 
`rosparam set /m1n6s300_driver/jointSpeedLimitParameter1` (for joints 1,2,3)
and
`rosparam set /m1n4s200_driver/jointSpeedLimitParameter2 `(for joints 4,5,6)

Achllle, 2018-07-10:
Thanks for your help. It seemed to only happen on one machine and did not occur again. Closing for now.

Achllle, 2018-07-10:
Thanks for your reply. This seemed to only happen on one machine and didn't happen again. Closing for now.

closed 2018-07-10 by Achllle