Kinovarobotics/kinova-ros #148: What are the homing requirements for the Kinova mico?

sjhansen3, 2018-07-06:
When and why do I need to home the mico?

Do I need to home the mico before I launch bringup? 

How do I know if it is properly homed?

nsaif-kinova, 2018-07-06:
The home position is a predefined known valid Cartesian position. Technically, you could switch to Cartesian control mode from any valid Cartesian position. If you are already in Cartesian control mode, you are guaranteed that your robot will stay in a valid Cartesian position
Every time you reboot your arm you have to send it Home again to be able to use the Cartesian control mode.

You can use the joystick to send it Home (By keeping your finger on the home button until the robot stops moving) or you can use a command that will send the robot Home (It will be home once it stops moving)

Users can home the robot by the command below. It takes no argument and brings the robot to pre-defined home position. The command supports customized home position that users can define by using the SDK or JacoSoft as well. /'${kinova_robotType}_driver'/in/home_arm

martine1406, 2018-07-06:
Hi @sjhansen3 

you do not need to home the arm before launching bringup, but it will be convenient to home to arm if you want to control the arm in Cartesian mode. 
You know the arm is homed when the angles you can monitor = the home angles. You can turn on the arm, press on the joystick home button us

closed 2018-07-06 by martine1406