Kinovarobotics/kinova-ros #152: Torque controller not working

sahandrez, 2018-07-19:
Hi, 

I would like to use the torque controller on the Jaco 2 arm. The robot is successfully set to the torque control mode by calling the service, but when I publish joint torque commands the robot does not move. Am I missing something? 
Do I need to update the firmware in order to use the torque controller? These are the versions of the firmware:
- Main firmware: 5.5.0.0
- Actuators: 2.1.1
- Fingers: 1.1.4
- Comm interface: 1.0.7

Thank you very much. 
Sahand

martine1406, 2018-07-20:
HI Sahand

I would recommend updating your base firmware with the latest version available. You can download the Kinova SDK from our website (https://www.kinovarobotics.com/en/knowledge-hub/all-kinova-products). Take the Kinova SDK Jaco2. 

When you say The robot is successfully set to the torque control mode by calling the service, it means that you are able to move the robot by hand after calling the service? What kind of joint torque commands are you trying to publish?

Thanks

sahandrez, 2018-07-20:
Hi Martine, 

Thank you for your message. 

Yes I am able to move the robot by hand after calling the service, but as soon as I send a torque command to the robot, it locks down and does not move. 
I tried several torque commands, including the example on the [ReadMe](https://github.com/Kinovarobotics/kinova-ros#torque-control) file. 

Do I still need to update the firmware given that the robot can successfully switch to the torque control mode?

Thanks, 
Sahand

martine1406, 2018-07-20:
Hi Sahand

Try to set the safety factor to 1.0. You have an api function setTorqueSafetyFactor https://github.com/Kinovarobotics/kinova-ros/blob/master/kinova_driver/src/kinova_api.cpp#L217. You can try without updating your firmware.

closed 2018-08-13 by nsaif-kinova