Kinovarobotics/kinova-ros #154: Velocity control in cartesian space

haoxuw, 2018-07-26:
Hello,

If I understand correctly, the reference frame for linear Cartesian velocity is the root of the arm, same as world frame. But the reference frame for angular velocity is the gripper itself. Is that correct?

Is there a good way to reliably move the gripper towards a fix orientation? Or should I keep trying to get the latest orientation, and keep calculating relative twist_angle in the while loop?

If there is a better way, please point me to the right direction.

Thanks a lot.

Cheers,
Haoxu

martine1406, 2018-07-26:
Hi @haoxuw 

Yes this is correct. Translations are expressed in the base frame and rotations, in the effector frame. 
You can provide an orientation command as a CARTESIAN_POSITION command, yes. Or you can proceed with velocity commands in a while lopp.

Thanks
Martine

closed 2018-07-26 by martine1406

haoxuw, 2018-07-31:
Hi Martine, @martine1406 

Thanks for helping out! I tried to implement the second solution (by publishing PoseVelocity message to /j2s7s300_driver/in/cartesian_velocity), but couldn't control orientation correctly.

Say, I hope to use cartesian velocity control to move the arm, while maintain orientation towards **goal_twist_angular**. Should I get the current orientation quaternion from /j2s7s300_driver/out/tool_pose? In order to fill in twist_angular commands, what is the formular to transform **goal_twist_angular** from the arm's base frame to the effector frame?

Thank you so much!

Regards,
Haoxu

martine1406, 2018-07-31:
HI Haoxu

/j2s7s300_driver/in/cartesian_velocity expects to receive the rotation commands in the end-effector frame and the translation commands in the base frame. 

/j2s7s300_driver/out/tool_pos will provide you the orientation in the base frame. You can use tf to find the transformation between the two frames. 

Thank you
Martine

break-seven, 2022-11-28:
I am sorry, I'm curious  how to implement speed control for Cartesian space end-effectors. At present, the only thing I can understand in the code is to add the specified speed to the FIFO. However, it has not been found how to turn the Cartesian speed command of the end-effectors into the joint speed or  position command. Or can the robotic arm itself handle Cartesian space velocity commands?

felixmaisonneuve, 2022-12-01:
Hi @break-seven,
You can publish on the `in/cartesian_velocity` topic to send cartesian velocity commands. This is mentionned in the main [ReadMe](https://github.com/Kinovarobotics/kinova-ros#velocity-control-for-joint-space-and-cartesian-space). You would have to do something like this : 
```
stopic pub -r 100 /j2n6s300_driver/in/cartesian_velocity kinova_msgs/PoseVelocity "{twist_linear_x: 0.0, twist_linear_y: 0.0, twist_linear_z: 0.0, twist_angular_x: 0.0, twist_angular_y: 0.0, twist_angular_z: 10.0}"
```

This calls the `sendAdvancedTrajectory` API function
https://github.com/Kinovarobotics/kinova-ros/blob/c13ad4367ffc0449dbbfd0fd2277f9795a207430/kinova_driver/src/kinova_comm.cpp#L1092-L1119

break-seven, 2022-12-01:
> Hi @break-seven, You can publish on the `in/cartesian_velocity` topic to send cartesian velocity commands. This is mentionned in the main [...]

Thank you very much for your patient reply, it seems that how to convert velocity commands in Cartesian space to joint group velocity command is encapsulated in the API, or implemented on the manufacturer's robotic arm controller. (I don't know if this statement is correct)

break-seven, 2022-12-01:
Hi @felixmaisonneuve 
Thank you very much for your patient reply.
It seems that how to convert velocity commands in Cartesian space to joint group velocity command is encapsulated in the API, or implemented on the manufacturer's robotic arm controller. Am I correct in this understanding?
Thank you again for your time on this!