Kinovarobotics/kinova-ros #155: moveit problem

marooncn, 2018-08-01:
Hi, I want to use j2s7s300 to grasp objects and the grasp model can be found [here](https://github.com/marooncn/robot_grasp). Now I use moveit  to plan a multi-segment continuous track to pick the object in one basket and then place into the other.  I plan the track in the Cartesian space and I define a number of points which kinova effector goes through point by point. It's OK in rviz but when I execute in gazebo, the robot always moves to the state of vertical which is defined in moveit setup-assistance before moves to the target pose.  The pose which kinova reaches is always wrong and terminal shows a problem:
>  ABORTED: Solution found but controller failed during execution

I just wonder why it moves to the state of vertical firstly every time. In the code, I make the current state as initial planning state. Why it happened?

martine1406, 2018-08-01:
Hi @marooncn

When you start Gazebo via [kinova_gazebo robot_launch. launch](https://github.com/Kinovarobotics/kinova-ros/blob/master/kinova_gazebo/launch/robot_launch.launch), the [kinova_control kinova_control.launch](https://github.com/Kinovarobotics/kinova-ros/blob/master/kinova_control/launch/kinova_control.launch) file gets launched.  In this last file, the script [move_robot.py](https://github.com/Kinovarobotics/kinova-ros/blob/master/kinova_control/src/move_robot.py). The initial position the robot tries to reach is defined in this file. 

Maybe you could try another position if this one does not suit you well. 

Otherwise, do you see any error in the gazebo terminal? 
If so, it could be linked to the Gazebo controllers. https://github.com/Kinovarobotics/kinova-ros/issues/132

I hope this helps
Martine

marooncn, 2018-08-02:
Hi @martine1406. 
Maybe you misunderstand what I mean. I just want the robot to execute the plan trajectory from the current pose rather than initial pose (the state of vertical defined in moveit setup-assistance)  to the target pose which is defined in the [file](https://github.com/marooncn/robot_grasp/blob/master/Kinova_pick_place.cpp) . But actually  it always go back to the initial pose then execute the trajectory I defined so the terminal pose is wrong correspondingly. I just wonder why it is and how can I solve it to make sure it start to move from the current state directly to the target state as I defined.  Thank you.

martine1406, 2018-08-02:
Hi @marooncn

Have you run the default kinova [pick place demo ](https://github.com/Kinovarobotics/kinova-ros/blob/c89b4ba71438b3bde4f007d470a30d7dfb579e38/kinova_moveit/kinova_arm_moveit_demo/src/pick_place.cpp)? I do not get your problem because when running this demo on my end, the robot does not go back to some initial pose in between moving to various targets...

FransicsXu, 2018-08-03:
Hi @martine1406 
I also meet the same problem as marooncn. When I use moveit in cartesian space, just the same as pick_place kinova pick place demo, I define various targets pose. I hope kinova can move to the targets successively,but kinova always moves to the vertical,then moves to the target. It causes me a lot of trouble. So I record a video, it can be found [here](https://pan.baidu.com/s/1bLU9YRJGiDjgqP9IY2AfNg). How can I solve it? Thank you.

martine1406, 2018-08-03:
Hi guys

From the video @FransicsXu, I can see two things

1) the loop animation option is toggled on (that's why the plan is shown multiple times)
[image]

2) the robot never executes the plan when you press 'e'.

What are the specific commands you are launching for this demo?
Do you launch gazebo first? this is what you have to do, unless you launch the real robot's driver node. 

From what I see, it seems that you are launching either 
'roslaunch j2s7s300_moveit_config j2s7s300_gazebo_demo.launch'
or
'roslaunch j2s7s300_moveit_config j2s7s300_demo.launch'

but without launching gazebo or the real robot's driver (respectively) first.
Let me know the specific commands you give in ROS

FransicsXu, 2018-08-04:
Hi @martine1406 
Thx, According to the j2s7s300_gazebo_demo.launch file,  I solve the problem.

closed 2018-08-04 by marooncn