Kinovarobotics/kinova-ros #157: kinova finger problem

marooncn, 2018-08-06:
Hi, When I use j2s7s300 to grasp objects. There are some problems about the robot finger.
1. The [move_robot.py](https://github.com/Kinovarobotics/kinova-ros/blob/master/kinova_control/src/move_robot.py) should be called multi times before the robot arm reaches the setting pose including the finger position. 

2. The finger controller becomes invalid and cannot be recovered after hitting other objects. Just like the below:
[image]

3. There is little grip force after the hand is closed, and the friction is not enough. Is there friction of finger?
[image]

4. I tried to tune the PID parameters of the robot finger. When I set D to some non-zero value, the model changes fast and strange:
[image]
And there are errors:
[image]

I think the robot hand controller is not very stable and robust. And I'm curious about finger friction because I think the grasp force is so small. I want to grasp the object in the basket want should I do to make finger controller more robust  and how can I increase the finger force to grasp the object?

Thank you sincerely.

martine1406, 2018-08-06:
Hi @marooncn 

We are not Gazebo experts unfortunately. I think you would have more chances to find the answers you are looking in Gazebo forums. 

For the friction, no friction model was defined between the fingers and the object. You may need to define this part on your own: http://answers.gazebosim.org/question/14887/object-slips-out-of-gripper-gazebo-7-ur5-robotiq-3-finger/, http://answers.gazebosim.org/question/9371/finger-contact-erp-cfm-kp-and-kd/

For the controller stability, you may need to tune the robot and fingers PIDs. You may also decide to implement a different type of ros-controller than an effort controller. You need to look at what will meet your simulation requirements... ROS offers many tools and those we have used may not fit your exact needs.

We provide a basic Gazebo model of the robot to start off, but yes there are limitations, and we hope the community can help resolve them and improve the model as a whole. 

Thank you for your understanding and sorry for this situation.

marooncn, 2018-08-07:
@martine1406 Thank you for answering and I'll try to solve this problem. If I have some progress I'll reopen this issue and share my experience.

closed 2018-08-07 by marooncn

marooncn, 2018-08-23:
The problem has been solved. The model can be found at [robot_grasp ](https://github.com/marooncn/robot_grasp) and the source code to grasp can be found [here](https://github.com/alxer/JCAR-Competition). We 
1. provide a stable simulation environment for grasping task.
2. solve the problem that the robot hand can't grasp 
3. provide a function that can control the robot hand continuously.
4. provide a demo that the robot grasps the objects from one basket to the other.

martine1406, 2018-08-23:
Hi @marooncn 

Thank you very much for sharing your solution here. It will be very useful for the community
Martine