Kinovarobotics/kinova-ros #158: Moveit do not show the interactive marker

jackietom, 2018-08-10:
I am new to Moveit!. I did as the wiki says in ubuntu 14.04, ros-indigo in a docker. Everything seems to be OK except for there is no interactive marker.

nsaif-kinova, 2018-08-10:
Hello,
please follow the steps on this page [](https://github.com/Kinovarobotics/kinova-ros/wiki/MoveIt)

With actual robot connected run this command : 
roslaunch kinova_bringup kinova_robot.launch kinova_robotType:=*robot_type*

open a new terminal and run this command :
roslaunch robot_name_moveit_config robot_name_demo.launch

kinova_robotType specifies which robot type is used. For better supporting wider range of robot configurations, robot type is defined by a char[8], in the format of: [{j|m|r|c}{1|2}{s|n}{4|6|7}{s|a}{2|3}{0}{0}].

Robot category {j|m|r|c} refers to jaco, mico, roco and customized
version is {1|2} for now
wrist type {s|n} can be spherical or non-spherical
Degree of Freedom is possible to be {4|6|7}
robot mode {s|a} can be in service or assistive
robot hand {2|3} may equipped with 2 fingers or 3 fingers gripper.
The last two positions are undefined and reserved for further features.

eg: j2n6s300  refers to jaco v2 6DOF service 3 fingers.

1. roslaunch kinova_bringup kinova_robot.launch kinova_robotType:=j2n6s300
2. roslaunch j2n6s300_moveit_config j2n6s300_demo.launch

3. Drag the interactive marker to the pose that you would like to set for the robot

4. Next, click on plan and MoveIt will visualize its planned path to move from the start state to the goal state. If the visualized plan is acceptable you can command the robot to execute the plan by clicking execute.

jackietom, 2018-08-10:
Thank you for your immediate response! I will try it and tell you the result.

jackietom, 2018-08-10:
It turns out that I forgot to install ros-indigo-trac-ik package. It works now. Thank you very much and I will close this issue.

closed 2018-08-10 by jackietom