Kinovarobotics/kinova-ros #160: Is IKFast Broken? What are recommended fixes if it is?

sjhansen3, 2018-08-27:
From the documentation:
"IKFast works well too, unfortunately, it often gives the elbow down solution which is undesireble. This will be improved with future releases."

Can you provide more detail? I am currently implementing decartes for the Mico - this requires all IK solutions for each pose. I am concerned that IKFast might not be returning all valid poses - in some cases I cannot find solutions where I believe I should.

1. Are there any tools that Kinova uses internally to identify whether a given pose is reachable? This would be super helpful to know if the IK solver is working or not.
2. Is the IKFast plugin known to be broken? If so, how do we know?
3. Was the IK Fast plugin recreated in the past few months? The IKFast plugin was last updated for the m1n6s300 on Dec 21, 2017 for commit 40e923013d3f304d18499ccfe9426c42f74436ec. Whereas a bug was found in the kinova URDFs and fixed in commit on Feb 12, 2018 with commit 69359bf895516fc252d8335fcffd38e4a6553e50. I have a running theory that the IKFast plug-in is not working because it was ran using the old URDF, but I have yet to validate this theory.

Thanks
Steve

martine1406, 2018-08-27:
Hi @sjhansen3 

we are not experts in this field unfortunately. The comment that you see in the documentation warns the users that the IkFast IK solver often finds an elbow-down solution, which we do not find optimal. We did not mean that the IKFast was broken or anything. 
There are various IK solvers available in ROS. Among them, you have KDL, IKFast and TracIK. 
I think IKFast will return you valid solutions, only maybe not the most elegant ones. 
I do not think the IKFast plugin was updated recently... you may need to regenerate it. 
I hope this helps
Martine

sjhansen3, 2018-08-27:
I believe IKFast is in fact not working. Here is the test I ran to show this:

1. Use moveit to find random valid joint positions which are not in self collision
2. Use moveit's forward kinematics to compute the pose of the end effector
3. Use IKFast to compute the joint positions

Since the pose was generated from valid joints I know it must have a valid solution. Since IKFast computes all analytical solutions it should be able to solve for the joints which produced the pose.

Currently, IKFast is failing to solve for the joint positions about 1/3 of the time: Here are the results.

#Fail 1
Joint positions: [ 0.264571, 4.78346, 1.48337, 1.79758, 1.29986, 0.935139 ]

FK transformation result: 
 0.977443 0.0145463 -0.210699 -0.353013
 0.185894 -0.532773  0.825588  0.224054
-0.100245 -0.846133 -0.523459 -0.151771
        0         0         0         1

#Fail 2
joint positions [ 2.50052, 4.69368, 5.49493, 2.70871, -2.73724, 0.945767,  ]

FK result: 
  0.644361  -0.700995  -0.305622 -0.0615348
 -0.064643  -0.448151   0.891618   0.277219
 -0.761984  -0.554767  -0.334086   0.394621
         0          0          0          1

#Fail 3
joint positions [ -2.53011, 4.61211, 1.0902, 2.12166, 2.84611, -1.5577,  ]

FK result: 
 0.236212  0.305103   0.92256  0.410172
 0.298165  0.880867 -0.367656 -0.171665
-0.924825   0.36192    0.1171 0.0329529
        0         0         0         1

#Fail 4
joint positions [ 0.591168, 1.46163, 3.90504, -0.168106, 2.51934, -1.24554,  ]

FK result: 
 0.753025 -0.286398  0.592393  0.558866
0.0025756   0.90158  0.432604 -0.130659
-0.657987 -0.324236   0.67965  0.165373
        0         0         0         1

sjhansen3, 2018-08-28:
I see the IKFast was generated by user Soli: https://github.com/soli7?tab=overview&from=2016-12-01&to=2016-12-31

Is this an employee at Kinova or Clearpath? Is there any way we could talk directly to them or someone else who has knowledge of IKFast plugins?

martine1406, 2018-08-28:
Hi @sjhansen3 

Sid was an employee at Kinova but is no longer working at Kinova. I do not have his personal contact. I am sure you can find people who know the IKFast plugins way better on ros/moveit forums. Have you tried to regenerate the IKFast plugin from the urdf? Honnestly, I would not be able to tell you why this plugin fails 1/3 of the time. Maybe the curved wrist is hard to deal with for this plugin. Is it possible for you to test other IK solver? 

Sorry about this situation
Martine

martine1406, 2018-08-28:
Hi @sjhansen3

Have you tried looking a little bit online and see why IKFast might fail to find a solution sometimes? I wonder if it is necessarily linked to an error in the robot's urdf/in this specific package, or if this is something related to IKFast. 

In all cases, I do not think anyone at Kinova is currently an expert on IKFast. The IKFast maintainers would be in a way better position to be able to comment on their tool's performances. Nonetheless, I can try to open a support ticket in our engineering department. For that, I would need your robot's serial number please. 

Thank you
Martine

sjhansen3, 2018-08-28:
I found this marking on the robot - if it isnt the serial number please advise on how to find it. :)

Thanks for opening up a ticket Martine! I will also look for other places to post an issue with Moveit or OpenRAVE perhaps, but I doubt Ill get much from OpenRAVE - they currently have 220 open issues going back to 2012...

[image]

tgn3000, 2018-08-29:
Hi, I am interested in IK of this robot. Can you please send me the robot file? Should be some *.dae or *.xml file. Thanks!

Cheers,
Gary

martine1406, 2018-08-29:
Hi @tgn3000 

Hmm not sure what you are talking about. The IK algorithm that the Kinova robots runs internally is not available as a standalone code. You will find the robot's DH parameters in the user guide https://www.kinovarobotics.com/sites/default/files/ULWS-RA-MIC-UG-INT-EN%20201804-1.0%20%28KINOVA%20MICO%E2%84%A2%20Robotic%20arm%20user%20guide%29_0.pdf

The urdf of the robot used by the ROS IK solvers is already present in this package. 
Thank you

tgn3000, 2018-09-11:
Thanks Martine! I also found this document you put here. As I look at the kinematics matrices defined by its DH parameters, this six-joint problem seems not to fall into the difficult category of IK. That is, one can somehow separate certain variables from the others, and solve smaller problems. That said, I believe Dr Diankov's IKFast in OpenRAVE should be able to codegen a C++ solver successfully, unless this repo did not call IKFast properly, or did not use the latest version of IKFast, perhaps. In fact I would be interested to know how they call IKFast in their package here. Thank you very much!

tgn3000, 2018-09-24:
I exchanged some emails with Kinova engineers. From my perspective, they seem to mix their classic DH parameters notation with the modified DH. This can be seen in Table 20 on page 55 in the PDF you referenced. So the initial IK problem I got was wrong; it is not an easy problem where one can separate variables.

Indeed after some analysis on the correct kinematics matrices, it seems a very difficult problem.

One engineer also sent me another PDF, which seems a more up-to-date document, saying Mico has two types of arms: curved (in your PDF) and spherical.

https://www.kinovarobotics.com/sites/default/files/ULWS-RA-JAC-UG-INT-EN%20201804-1.0%20%28KINOVA%E2%84%A2%20Ultra%20lightweight%20robotic%20arm%20user%20guide%29_0.pdf

Also, yes, I was wondering how IKFast was called in this ROS and which version they were using. Was it the same IKFast from Dr Diankov's OpenRAVE, or this repo's devs' modified/improved version? Would be very interesting to know.

Thanks
Gary

martine1406, 2018-10-05:
Hi all

Some clarification here: 
Mico does NOT have two types of wrist: curved (in your PDF) and spherical
Jaco (now rebaptised Ultra lightweight robotic arm) does have those two types of wrist available

Mico is only available in curved wrist version. 

That said, the DH tables indeed indicated alpha(i-1) and a(i-1) parameters in the title of the columns, which suggest the modified DH convention, but reported parameters from the classical convention. I can confirm that we are using the Classical DH convention, so alpha(i-1) (only the column title, not the contente of the column) should be replaced by alpha(i) and the same applies for a(i). This willl be corrected in the next version of the documentation. Sorry for the confusion. 

I cannot say for sure because I am not the person who generated the IKFast packages for the Kinova robots in this repo, but I doubt that we would have used a very custom version of IKFast.
If this is something one of your guys is very familiar with, could you try to regenerate the IKFast package from the urdf files provided (you can do it for MICO only if you want since it seems to be the problematic one)?

Thank you
Martine

sjhansen3, 2018-10-05:
I am not very familiar but I did run the IKFast plugin a few times with
twiddled settings.

I used these instructions to run it in a Docker container.
https://answers.ros.org/question/263925/generating-an-ikfast-solution-for-4-dof-arm/

When IKFast runs it runs a series of tests after creating the plugin and
returns an accuracy, I'm guessing these tests are similar to the one I
wrote which runs FK then IK. Both my tests and  the IK tests return IK
success rates below 60%. This means 40% of the time the solver is not
finding a solution to a pose when it should. These accuracy values did not
improve when I tried adjusting the accuracy.

After running these tests I can confidently say the current IKFast plugin
is broken. It also seems plausible that the curved wrist is not able to be
solved for by IKFast.

My company is working with PickNik consulting  and they have experience
designing analytical IK solvers. They also have tons of experience with
MoveIt. I would recommend working with them to clean up the Kinova-ROS code
base. Their great.

They also put a proposal together for development of an analytical IK
solver. These Analytical IK solvers can return all possible configurations
to achieve a particular pose. This is important for planning applications.
If you guys plan on supporting this kind of work I would highly recommend
working with PickNik to develop a custom IK Solver for the Mico.

Martine - feel free to send me an email if you want to chat more about this
offline :)

martine1406, 2018-10-09:
Hi sjhansen3 

Thanks for the tip. I did not know of PickNik. It is good to know that they have experience with analytical IK solvers and with MoveIt as well. A custom, robust analytical IK solver could indeed be a good addition to our ROS package. As for anything, it will be a question for priority and budget for Kinova. In all cases, we have the information now, so that's a good start. 

Thanks again for your feedback

felixmaisonneuve, 2021-06-02:
This thread has being inactive for over 2 years, I am closing this issue. If there is still a problem, please create a new issue.

Thank You

closed 2021-06-02 by felixmaisonneuve