Kinovarobotics/kinova-ros #161: Add a proper license for the `kinova_bringup` package

alimuldal, 2018-09-04:
Please consider giving the [`kinova_bringup` package](https://github.com/Kinovarobotics/kinova-ros/blob/master/kinova_bringup/package.xml#L19) an open-source license. Without a license, users won't be able to reproduce, modify, or redistribute this code without your explicit permission.

alimuldal, 2018-09-04:
On a related note, please could you include the full license text for packages marked as "BSD"-licensed (e.g. https://github.com/Kinovarobotics/kinova-ros/issues/93). "BSD" isn't specific enough, since there are multiple different BSD licenses with slightly different terms.

martine1406, 2018-09-04:
Hi

Please consider the following [license file](https://github.com/Kinovarobotics/kinova-ros/blob/master/LICENSE) that was added to the repository. The repo now features clearly a BSD 3-Clause license. Thank you

closed 2018-09-04 by martine1406

alimuldal, 2018-09-05:
Thanks, much appreciated! Would you mind also updating the `package.xml` files that still contain `<license>TODO</license>`?

martine1406, 2018-09-05:
I did it on the master branch. Thank you for letting us know