Kinovarobotics/kinova-ros #162: Is there any software emergency stop in Kinova API

sahandrez, 2018-09-04:
Hi, 

We are looking for a way to manually stop/kill the movement of the robot (an emergency stop in case things go wrong, since we are doing torque control) through software. Is there any function/method that has the same functionality as of the physical key on the base of the robot? If not, could you please point out some possible ways of implementing a software emergency stop?

Thank you very much.

Bests, 
Sahand

martine1406, 2018-09-05:
Hi @sahandrez 

My best way to stop the motion in torque mode would be to switch the robot in trajectory mode (position control of the actuator) using the API function SwitchTrajectoryTorque. The robot would "lock" at its current position at the moment it receives the switch command to position mode. It does not have the same effect as the power button though, which disables power in the actuator. There is no software function that will disable the power in the arm's actuators. 

What I would do is monitor the robot's state and call SwitchTrajectoryTorque(0) to switch to position control when necessary. You can also define yourself a node that you call from the terminal whenever you feel a situation is dangerous, and that node would call SwitchTrajectoryTorque(0)

I hope this helps
Martine

sahandrez, 2018-09-05:
Hi Martine, 

Thanks a lot for your answer. I will take this approach then. Do you think I will get the same responsiveness by calling the `set_torque_control_mode` service or is it safer to call the API function directly? 

Bests, 
Sahand

martine1406, 2018-09-05:
Hi @sahandrez 

I think the responsiveness would be comparable since there is not so much wrapping around the API function. Calling the API function directly could be slightly faster, but I do not think the difference would be perceptible by a human.
Cheers
Martine

sahandrez, 2018-09-05:
Hi Martine, 

OK sounds good. Thank you very much. 

Bests, 
Sahand

pirobot, 2019-01-08:
We are also interested in disabling the arm when we issue an emergency stop signal.  The solution from @martine1406 locks the arm in the current configuration so it does not prevent the arm from breaking itself or some other object if the arm is attached to a mobile robot and the robot keeps moving.

Is there a programmatic way to set all the torques to very small values so that the arm becomes compliant--similar to the way it feels when it is powered off but without actually turning it off?

martine1406, 2019-01-08:
Hello. Sorry I will not be able to answer very often as I am a new mother and my newborn keeps me very busy. 

If I understand correctly, you are controlling the arm in torque mode. So you are sending it torque commands? If you send torque commands of zero, the arm is supposed to only compensate its own weight, so it will be compliant as you say. 

The arm expects to receive a torque command every 10 ms, so if you stop sending commands, it will just be in zero force mode, i.e. just compensate its own weight. 

I guess you could have a function/node that runs your torque control loop as long as a certain flag is true or so and a function/node that updates the flag to false when you want to stop the torque control in emergency, in which case no torque commands would be sent to the arm and it would just be in zero gravity mode?

If I misunderstood what you want to do, please clarify and it might help my colleagues help you better. 
Cheers
Martine

pirobot, 2019-01-09:
@martine1406 Congratulations on your newborn! :)  I will pass along your message to the person working with our arm to see if that might work for us.  Thanks!

felixmaisonneuve, 2021-06-02:
This thread has being inactive for over 2 years, I am closing this issue. If there is still a problem, please create a new issue.

Thank You

closed 2021-06-02 by felixmaisonneuve