Kinovarobotics/kinova-ros #164: Kinova sideways mount

sjhansen3, 2018-09-14:
Can the mico be mounted sideways with the base plate orthogonal to the floor? In trying this it seems the payload might be effected, or joint torque limits are reached. Is it possible to mount the robot this way? Do you have any documentation on this?

nsaif-kinova, 2018-09-14:
Hello Stephen,
The robot can be mounted in any direction, but it is very important to update your gravity vector accordingly, otherwise the robot will not compensate the gravity correctly.

sjhansen3, 2018-09-15:
Could you repost the second link it looks like you linked a file on your computer which I do not have access to.

nsaif-kinova, 2018-09-17:
Hello Stephen,
On Ubuntu the link should be this one [file:///opt/JACO2SDK/GUI/html/_kinova_8_a_p_i_8_usb_command_layer_ubuntu_8h_a62322ebef1fd5c52c265a7b4681c65bc.html#a62322ebef1fd5c52c265a7b4681c65bc](url)
It is an offline link to i`nt SetGravityVector` in the API.

martine1406, 2018-10-04:
Hi @sjhansen3 

Just to comment on this answer, the payload should theoretically not be affected by a side mount since actuator 1 and 2 have the same torque production capability. In a vertical mounting, joint 2 is the one limiting the payload. In a horizontal mounting, both joints 1 and 2 will be the bottleneck depending on the robot's configuration, but the maximum moment of arm is not bigger in a horizontal configuration. 

That said, the arms have been tested less in a horizontal mounting than in a vertical mount. There might be a little more flexion on some links, but this has not been characterized officially. 

As I said, in theory, the robot will behave similarly in a vertical and horizontal mounting configuration, and the only thing you need to insure is that you change the gravity vector. 

In practice, the behaviour, especially in admittance or torque mode, might be a bit different because joint 1 experiences a lot of cross-torque and torque sensors are partly sensitive to this cross-torque (although they would not be in an ideal situation). But I would still say it is totally feasible to mount the arm in an horizontal configuration and work with it just fine (arms are actually mounted horizontally on our Movo platform). 

I hope this makes things clearer
Let me know if I can close that issue
Martine

closed 2018-10-30 by martine1406