Kinovarobotics/kinova-ros #165: Is it possible to change the home position? if so how?

sjhansen3, 2018-09-16:
We mounted our robot sideways and the home position doesn't really make sense and is often awkward to reach. Can we change the home position for the robot? if so, how? Is this a bad idea - are there many hard coded things that depend on a very specific home position?

nsaif-kinova, 2018-09-17:
Hello Stephen,
Yes you can change the home position of the robot, you can do it via the Dev Center check this document page 13 [https://www.kinovarobotics.com/sites/default/files/SDK-UG-INT-EN%20201804-1.0%20%28KINOVA%E2%84%A2%20Software%20development%20kit%20user%20guide%29_2.pdf](url) insert two new position and click activate to set your new Home and Retracted psotion.

Make sure your Robot is in a safe position and not colliding with other objects, and that it is a valid Cartesian position.

Sincerely,

martine1406, 2018-10-04:
Hi @sjhansen3 

Let add a few more details on Nassif's answer. The purpose of the home position is only to give the robot a valid Cartesian position reference at start-up. With this in mind, no problem in changing the home position.

The procedure Nassif suggested is indeed useful. The advance retract option will indeed change the ready and retract position from the robot which are used when you send the robot home or retract **with the joystick**. If this is what you want, then awesome. 

However, I must say the MoveHome API function will still send you to the default home position. If you want to send the robot to a home position programatically, then I would suggest you define yourself a new <move home> function that sets the control to Angular mode (SetAngularControl), sends the robot an angular position command that corresponds to a valid Cartesian position from the robot's workspace, and finally sets the control to Cartesian (SetCartesianControl).

I hope this makes things clearer
let me know if I can close this issue
Martine

closed 2018-10-30 by martine1406

sjhansen3, 2019-03-07:
@nsaif-kinova That link you posted is broken

https://www.kinovarobotics.com/sites/default/files/SDK-UG-INT-EN%20201804-1.0%20%28KINOVA%E2%84%A2%20Software%20development%20kit%20user%20guide%29_2.pdf

Can you repost?

nsaif-kinova, 2019-03-07:
I tried it, it is working, can you please try again?

ruturajsambhusvt, 2025-03-24:
> Hi [@sjhansen3](https://github.com/sjhansen3)
> 
> Let add a few more details on Nassif's answer. The purpose of the home position is only to give the robot a valid Cartesian position reference at start-up. With this in mind, no problem in changing the home position.
> 
> The procedure Nassif suggested is indeed useful. The advance retract option will indeed change the ready and retract position from the robot which are used when you send the robot home or retract **with the joystick**. If this is what you want, then awesome.
> 
> However, I must say the MoveHome API function will still send you to the default home position. If you want to send the robot to a home position programatically, then I would suggest you define yourself a new function that sets the control to Angular mode (SetAngularControl), sends the robot an angular position command that corresponds to a valid Cartesian position from the robot's workspace, and finally sets the control to Cartesian (SetCartesianControl).
> 
> I hope this makes things clearer let me know if I can close this issue Martine

Hello, we are trying to do this with the Kinova API (not ROS). We define the target angular position, set PID gains, and use `SendBasicTrajectory` to send it. Once it reaches the position, we switch it to torque control with zero commanded torque. We are having two issues-
1) The trajectory is very aggressive, we want to slow it down ( similar to `MoveHome`). Upon reducing PD gains to 0.5,0.0125, the robot stops in the middle with red lights flashing on the remote and then goes into torque control mode with zero torques commanded (as expected). If the gains are 0.1, 0.025, the trajectory is still aggressive
2) When it reaches the desired pose and switches to torque control mode, it does not go to zero torque commands but seems locked in position control mode.

It would be great if you could provide any guidance to solve this issue.