Kinovarobotics/kinova-ros #167: Joint Values Inconsistency on j2n6s300

rafaelpossas, 2018-09-21:
Hi All,

I've been trying to do some joint planning using my Jaco j2n6s300 and I came through the problem below.

I am using the **kinova-ros package** along with **ROS Kinetic and Ubuntu 16.04**

Whenever I start my simulation with: **roslaunch kinova_gazebo robot_launch.launch**

and I read the joint states topic (**/j2n6s300/joint_states**), there seem to be inconsistency on the last 2 joints of the arm (j2n6s300_joint_5 and j2n6s300_joint_6). The problem is twofold:

1) The joints are not even within the lower and upper limits specified within the URDF

2) Everytime I restart Gazebo they seem to get a random values for a same initial pose, See below an example

Starting Gazebo for the first time, look at **j2n6s300_joint_5 and j2n6s300_joint_6**

name: [j2n6s300_joint_1, j2n6s300_joint_2, j2n6s300_joint_3, j2n6s300_joint_4, j2n6s300_joint_5,
  j2n6s300_joint_6, j2n6s300_joint_finger_1, j2n6s300_joint_finger_2, j2n6s300_joint_finger_3]

position: [-3.433988137491184e-05, 2.9003741053466783, 1.2992024657877943, 4.202535734888536, -11.165148707463949, 37.699077393946325, 1.000724306808693, 0.9994006487356604, 0.9993573993723155]

If I restart Gazebo and Run the same command again:

name: [j2n6s300_joint_1, j2n6s300_joint_2, j2n6s300_joint_3, j2n6s300_joint_4, j2n6s300_joint_5,
  j2n6s300_joint_6, j2n6s300_joint_finger_1, j2n6s300_joint_finger_2, j2n6s300_joint_finger_3]

position: [-0.0006671249350160835, 2.8976716669402833, 1.3000769288977914, 4.197928347574803, -4.878021615876596, 43.9824150056194, 0.9989748324374386, 1.001878501799836, 1.0015273938192504]

These values are not only pretty high (Its all in radians) but they are different for a same robot pose.

Am I missing something in here or is this a bug?

Thanks in advance.

Rafael

nsaif-kinova, 2018-09-21:
Hello Rafael,
Please check this link and let me know it this can help you.
[https://github.com/Kinovarobotics/kinova-ros/issues/132](url)
Sincerely,

rafaelpossas, 2018-09-24:
Hi Thanks for the Response.

I have checked and tried the solutions proposed, but it still doesnt work. I can't do any kind of planning with such an inconsistency on retrieving joint values.

What is actually causing this?

martine1406, 2018-10-04:
hi @rafaelpossas 

Unfortunately, we do not know the source of this issue. I would say it is most likely linked with Gazebo... but I am not an expert in Gazebo. Have you tried posted on Gazebo-ros related forums?

martine1406, 2018-10-30:
Hi @rafaelpossas 

Have you found something related to this issue?

felixmaisonneuve, 2021-06-02:
This thread has being inactive for over 2 years, I am closing this issue. If there is still a problem, please create a new issue.

Thank You

closed 2021-06-02 by felixmaisonneuve